espressif/esp-idf #14202: [ESP-WIFI-MESH] Mesh devices stop working when Root Node is in mesh sleep mode.

andy-danieal, 2024-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The problem seems to be with the **Root Node's sleep mode**. If I turn it on, then this issue occurs, but **if I turn it off, the devices work properly.** Can anyone please help me understand what might be happening?

**Note:-** Only added sleep mode in Root Node.

**Old code**

ESP_ERROR_CHECK(esp_mesh_disable_ps());

**New code**

ESP_ERROR_CHECK(esp_mesh_enable_ps());

/**  After Mesh esp_mesh_start() **/
ESP_ERROR_CHECK(esp_mesh_set_active_duty_cycle(50, MESH_PS_DEVICE_DUTY_REQUEST));
ESP_ERROR_CHECK(esp_mesh_set_network_duty_cycle(50, -1, MESH_PS_NETWORK_DUTY_APPLIED_ENTIRE));

zhangyanjiaoesp, 2024-07-18:
@andy-danieal 
Do you enable PS on root device only? Or both root and node devices?
what do you mean by saying `stop working` ? Can you provide a log to illustrate the problem ?

andy-danieal, 2024-07-18:
@zhangyanjiaoesp 

>Do you enable PS on root device only? Or both root and node devices?

Yes, Enable PS **ONLY on Root device**.

>what do you mean by saying stop working ? Can you provide a log to illustrate the problem ?

Not connected to Mesh network. 

We have attached child node logs with Enable PS mode and Disable PS Mode
[Disable-PS-Child-Mode.log](https://github.com/user-attachments/files/16280417/Disable-PS-Child-Mode.log)
[Enable-PS-Child-Mode.log](https://github.com/user-attachments/files/16280418/Enable-PS-Child-Mode.log)

zhangyanjiaoesp, 2024-07-23:
@andy-danieal When you enable the mesh PS on the root device, you should also enable the mesh PS on the node device, otherwise the node device can't find the root.

andy-danieal, 2024-07-23:
> @zhangyanjiaoesp mentioned in the comments that enabling mesh PS on the root device also requires enabling mesh PS on the child node device. 
**This is not mentioned in the [ESP-WIFI-MESH documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp-wifi-mesh.html).**

Could you please explain Why is this happening?

zhangyanjiaoesp, 2024-07-24:
@andy-danieal When the node device receive the mesh AP's beacon, it will check the mesh PS state, only when both the mesh AP and myself support mesh PS, or neither supports mesh PS, a mesh network can be formed.
we will update the mesh documentation.

closed 2024-07-24 by andy-danieal