espressif/esp-idf #14203: 我想在usb作为网卡时启用dhcp，我应该怎么做

yunfusheng33, 2024-07-17:
### Describe the solution you'd like.

我已经将usb模拟为网卡设备初始化后，将pc的网络共享给该设备，从而通过tcp client连接外部网络，但是我的网络IP是静态的，我应该怎么更改为dhcp自动获取IP？

yunfusheng33, 2024-07-18:
@leeebo 我现在分配给esp的是一个静态的192.168.137.x 的 IP 地址，然后我在pc使用了一个DHCP Server软件，将esp的虚拟网卡做HTTP(Web Server)，我应该如何使用dhcp client来获取192.168.137.2-192.168.137.255之间的一个ip？

下面是我的HTTP(Web Server)
[image]

leeebo, 2024-07-18:
@yunfusheng33 以太网默认配置下 DHCP Client 就是开启的，不用额外配置，你可以直接跑示例试试看 https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic

yunfusheng33, 2024-07-18:
@leeebo 这个例程我有看过，但我是想通过usb模拟为虚拟网卡使用，类似于esp-iot-solution里的usb-dongle

leeebo, 2024-07-18:
@yunfusheng33 明白了。不过一般 usb-dongle 方案，ESP32 端不再跑上层网络协议栈，只做接口间转发。你的需求可以参考这个方案 https://github.com/espressif/esp-iot-bridge

yunfusheng33, 2024-07-18:
@leeebo 好的，我看下，而且我在跑上面这个以太网例程的时候，串口并没有输出获取到dhcp的信息，我的板子是esp32s3devkitC，是板子或者我的dhcps软件有问题吗？

leeebo, 2025-05-08:
@yunfusheng33 以太网例程需要使用带以太网接口的开发板，并且需要连接网线到路由器或 PC 的共享以太网接口

closed 2025-05-08 by Alvin1Zhang