espressif/esp-idf #14210: IDF freeRtos task contex switch DO NOT release the spinlock

xmusco, 2024-07-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3 DEV KIT

### Power Supply used.

USB

### What is the expected behavior?

idf freetros  task create suscess and work ok

### What is the actual behavior?

这一版本的IDF，task可以创建成功，但是。在创建task后（包括xtaskcreate和pinnedtocore）访问各种资源，包括printf，LOGI, I2C, 还有take信号量等，都会报同一个错误，spinlock error。
idf5.1.4没有这个bug。

### Steps to reproduce.

**this task code run with a error with idf 5.2.2 but ok with idf5.1.x and 4.4.x**
#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "esp_log.h"
#include "esp_err.h"

const char* TAG="LAND";

static SemaphoreHandle_t sync_task;
TaskHandle_t task1_handle;
TaskHandle_t task2_handle;

void task1(void *arg)
{
    ESP_LOGI(TAG, "task1 start\n");
    xSemaphoreTake(sync_task,pdMS_TO_TICKS(5000));
    ESP_LOGI(TAG, "task1 take ok\n");
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 1 is running\n");
    }
}

void task2(void *arg)
{
    ESP_LOGI(TAG, "task2 start\n");
    vTaskDelay(pdMS_TO_TICKS(2000));
    ESP_LOGI(TAG, "task2 give\n");
    xSemaphoreGive(sync_task);
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 2 is running\n");
    }
}

void app_main(void)
{
    ESP_LOGI(TAG, "main start\n");
    sync_task=xSemaphoreCreateBinary();
    esp_err_t res = xTaskCreate(task1,"task1",4096,NULL,3,&task1_handle);
    esp_err_t res2 = xTaskCreate(task2,"task2",4096,NULL,3,&task2_handle);
    ESP_LOGI(TAG, "res=%d\n",res);
    if (res==pdPASS)
    {
        ESP_LOGI(TAG, "task1 create ok\n");
    }else{
        ESP_LOGI(TAG, "task1 create fail\n");
    }
    if (res2==pdPASS)
    {
        ESP_LOGI(TAG, "task2 create ok\n");
    }else{
        ESP_LOGI(TAG, "task2 create fail\n");
    }
    
    ESP_LOGI(TAG, "main finish\n");
}

KonstantinKondrashov, 2024-07-18:
Hi @xmusco!
I run your example on the v5.2.2 and see no issues (here is the log).
Seems like the configuration that you made caused the issue. Please provide your sdkconfig. Thanks.

```
I (174) cpu_start: Multicore app
I (185) cpu_start: Pro cpu start user code
I (185) cpu_start: cpu freq: 160000000 Hz
I (185) cpu_start: Application information:
I (188) cpu_start: Project name:     hello_world
I (193) cpu_start: App version:      v5.2.2-dirty
I (198) cpu_start: Compile time:     Jul 18 2024 09:58:52
I (205) cpu_start: ELF file SHA256:  d036771d0...
I (210) cpu_start: ESP-IDF:          v5.2.2-dirty
I (215) cpu_start: Min chip rev:     v0.0
I (220) cpu_start: Max chip rev:     v0.99 
I (225) cpu_start: Chip rev:         v0.3
I (230) heap_init: Initializing. RAM available for dynamic allocation:
I (237) heap_init: At 3FC95980 len 00053D90 (335 KiB): RAM
I (243) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (249) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (255) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (263) spi_flash: detected chip: generic
I (266) spi_flash: flash io: dio
W (270) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (283) sleep: Configure to isolate all GPIO pins in sleep state
I (290) sleep: Enable automatic switching of GPIO sleep configuration
I (297) main_task: Started on CPU0
I (307) main_task: Calling app_main()
I (307) LAND: main start
I (307) LAND: task1 start
I (307) LAND: task2 start
I (317) LAND: res=1
I (317) LAND: task1 create ok
I (327) LAND: task2 create ok
I (327) LAND: main finish
I (327) main_task: Returned from app_main()
I (2317) LAND: task2 give
I (2317) LAND: task1 take ok
I (3317) LAND: task 2 is running
I (3317) LAND: task 1 is running
I (4317) LAND: task 2 is running
I (4317) LAND: task 1 is running
```

xmusco, 2024-07-18:
Hi,
I delete my exist 5.2.2 fold and replaced with a new download version.
It seems that this bug had been fix in the new released idf5.2.2
Thanks

closed 2024-07-18 by KonstantinKondrashov

xmusco, 2024-07-21:
Hello,
I foud that the bug is still happen in some case.
please run the follow demo and will get a fail.
assert failed: xTaskRemoveFromEventList tasks.c:3860 (pxUnblockedTCB)

#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "esp_log.h"

const char* TAG = "LAND";
// static portMUX_TYPE my_spinlock = portMUX_INITIALIZER_UNLOCKED;

#define TASK1_PRIO      3                   /* 任务优先级 */
#define TASK1_STK_SIZE  1024                 /* 任务堆栈大小 */
TaskHandle_t            Task1Task_Handler;  /* 任务句柄 */

#define TASK2_PRIO      3                   /* 任务优先级 */
#define TASK2_STK_SIZE  1024                 /* 任务堆栈大小 */
TaskHandle_t            Task2Task_Handler;  /* 任务句柄 */

SemaphoreHandle_t BinarySemaphore;          /* 二值信号量句柄 */

void task1(void *pvParameters)
{
    ESP_LOGI(TAG,"give");
    vTaskDelay(pdMS_TO_TICKS(2000));
    xSemaphoreGive(BinarySemaphore);                    /* 释放二值信号量 */
    ESP_LOGI(TAG,"give ok");
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 1 is running\n");
    }
}

void task2(void *pvParameters)
{
    ESP_LOGI(TAG,"take");
    vTaskDelay(pdMS_TO_TICKS(5000));
    xSemaphoreTake(BinarySemaphore, portMAX_DELAY);
    ESP_LOGI(TAG,"take ok");
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 2 is running\n");
    }
}

void app_main(void)
{
    ESP_LOGI(TAG,"main start");
    // taskENTER_CRITICAL(&my_spinlock);           /* 进入临界区 */
    /* 创建二值信号量 */
    BinarySemaphore = xSemaphoreCreateBinary();
    /* 创建任务1 */
    xTaskCreate((TaskFunction_t )task1,
                (const char*    )"task1",
                (uint16_t       )TASK1_STK_SIZE,
                (void*          )NULL,
                (UBaseType_t    )TASK1_PRIO,
                (TaskHandle_t*  )&Task1Task_Handler
                );
    /* 创建任务2 */
    xTaskCreate((TaskFunction_t )task2,
                (const char*    )"task2",
                (uint16_t       )TASK2_STK_SIZE,
                (void*          )NULL,
                (UBaseType_t    )TASK2_PRIO,
                (TaskHandle_t*  )&Task2Task_Handler
                );

    // taskEXIT_CRITICAL(&my_spinlock);            /* 退出临界区 */
    ESP_LOGI(TAG,"main finish");
}

Closed #14210<https://github.com/espressif/esp-idf/issues/14210> as completed.

xmusco, 2024-07-21:
Hello,
I foud that the bug is still happen in some case.
please run the follow demo and will get a fail.
assert failed: xTaskRemoveFromEventList tasks.c:3860 (pxUnblockedTCB)

`#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "esp_log.h"

const char* TAG = "LAND";
// static portMUX_TYPE my_spinlock = portMUX_INITIALIZER_UNLOCKED;

#define TASK1_PRIO      3                   /* 任务优先级 */
#define TASK1_STK_SIZE  1024                 /* 任务堆栈大小 */
TaskHandle_t            Task1Task_Handler;  /* 任务句柄 */

#define TASK2_PRIO      3                   /* 任务优先级 */
#define TASK2_STK_SIZE  1024                 /* 任务堆栈大小 */
TaskHandle_t            Task2Task_Handler;  /* 任务句柄 */

SemaphoreHandle_t BinarySemaphore;          /* 二值信号量句柄 */

void task1(void *pvParameters)
{
    ESP_LOGI(TAG,"give");
    vTaskDelay(pdMS_TO_TICKS(2000));
    xSemaphoreGive(BinarySemaphore);                    /* 释放二值信号量 */
    ESP_LOGI(TAG,"give ok");
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 1 is running\n");
    }
}

void task2(void *pvParameters)
{
    ESP_LOGI(TAG,"take");
    vTaskDelay(pdMS_TO_TICKS(5000));
    xSemaphoreTake(BinarySemaphore, portMAX_DELAY);
    ESP_LOGI(TAG,"take ok");
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 2 is running\n");
    }
}

void app_main(void)
{
    ESP_LOGI(TAG,"main start");
    // taskENTER_CRITICAL(&my_spinlock);           /* 进入临界区 */
    /* 创建二值信号量 */
    BinarySemaphore = xSemaphoreCreateBinary();
    /* 创建任务1 */
    xTaskCreate((TaskFunction_t )task1,
                (const char*    )"task1",
                (uint16_t       )TASK1_STK_SIZE,
                (void*          )NULL,
                (UBaseType_t    )TASK1_PRIO,
                (TaskHandle_t*  )&Task1Task_Handler
                );
    /* 创建任务2 */
    xTaskCreate((TaskFunction_t )task2,
                (const char*    )"task2",
                (uint16_t       )TASK2_STK_SIZE,
                (void*          )NULL,
                (UBaseType_t    )TASK2_PRIO,
                (TaskHandle_t*  )&Task2Task_Handler
                );
    
    // taskEXIT_CRITICAL(&my_spinlock);            /* 退出临界区 */
    ESP_LOGI(TAG,"main finish");
}

`

KonstantinKondrashov, 2024-07-22:
printf (and related log functions) requires a lot of stack, so 1024 is not sufficient.