espressif/esp-idf #14211: @sjWang 此问题已在 master 上修复，请验证。如有问题，请 reopen 一下。

JasonYan324, 2024-07-18:
@sjWang 此问题已在 master 上修复，请验证。如有问题，请 reopen 一下。

_Originally posted by @ustccw in https://github.com/espressif/esp-at/issues/786#issuecomment-1867164747_
            
您好，请问这个问题在IDF5.2.2，这个分支上https://github.com/FreeRTOS/iot-reference-esp32c3也修复了么？现在我测试发现使用Python工具制作并烧录AWS证书的ESP32-C3-DevKit-RUST-1开发板上运行提示：
E (392) esp_secure_cert: Metadata magic word does not match
E (392) esp_secure_cert: Error in reading the metadata
E (402) main: Error in getting device certificate. Error: ESP_FAIL
E (412) main: Error in getting CA certificate. Error: ESP_FAIL

然后屏蔽该校验可以正确读出CA cert 和 certificate cert，但是 private key 读出来是乱码，而且private key的长度与实际文件长度也不相符
[image]

ustccw, 2024-07-18:
@JasonYan324 您好！ 
https://github.com/espressif/esp-at/issues/786 问题的修复未包含在 [IDF v5.2.2](https://github.com/espressif/esp-idf/releases/tag/v5.2.2) 上，而是在之后的 [commit](https://github.com/espressif/esp-idf/commit/fd6676e7)。您可能要更新一下 ESP-IDF 版本到 release/v5.2 最新的 commit，再验证看看。

JasonYan324, 2024-07-28:
> @JasonYan324 您好！ [espressif/esp-at#786](https://github.com/espressif/esp-at/issues/786) 问题的修复未包含在 [IDF v5.2.2](https://github.com/espressif/esp-idf/releases/tag/v5.2.2) 上，而是在之后的 [commit](https://github.com/espressif/esp-idf/commit/fd6676e7)。您可能要更新一下 ESP-IDF 版本到 release/v5.2 最新的 commit，再验证看看。

不是很理解，你是叫我从IDF5.2.2退回到5.2版本测试么？现在都更新到IDF5.3了哦

ustccw, 2024-07-29:
@JasonYan324 请参考 [更新至一个发布分支](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/versions.html#id16) 文档。

JasonYan324, 2024-07-30:
> @JasonYan324 请参考 [更新至一个发布分支](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/versions.html#id16) 文档。

看不懂。我退回IDF5.2版本测试烧录证书的确不存在这个问题了，但是跟糟糕的是IDF5.2版本编译同样的代码无法运行。而IDF5.2.2/IDF5.3编译可以运行，但是烧录证书却出现校验问题

ustccw, 2024-07-31:
@JasonYan324 可否按照 [模板](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=Type%3A+Bug&projects=&template=02_runtime_bug.yml) 重新 report 一下 issue。坦率的说，现在还不清楚您的工程用的是哪个 SDK（esp-idf 还是 esp-at），哪个版本 (`git describe --tags` 查看下)，具体什么硬件，怎么样的复现步骤。

JasonYan324, 2024-08-02:
> @JasonYan324 可否按照 [模板](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=Type%3A+Bug&projects=&template=02_runtime_bug.yml) 重新 report 一下 issue。坦率的说，现在还不清楚您的工程用的是哪个 SDK（esp-idf 还是 esp-at），哪个版本 (`git describe --tags` 查看下)，具体什么硬件，怎么样的复现步骤。

开发环境：
1、IDE：VS code 1.92.0
2、SDK：IDF5.2、IDF5.2.2、IDF5.3
3、测试代码：git clone https://github.com/FreeRTOS/iot-reference-esp32c3.git --recurse-submodules

测试情况：
1、使用 IDF5.2.2、IDF5.3 烧录证书，运行是报如下错误：
E (392) esp_secure_cert: Metadata magic word does not match
E (392) esp_secure_cert: Error in reading the metadata
E (402) main: Error in getting device certificate. Error: ESP_FAIL
E (412) main: Error in getting CA certificate. Error: ESP_FAIL

2、使用 IDF5.2 烧录证书没有报上述错误，但是编译同样的代码不能运行，从log上看直接无法boot完成就马上重启了

3、目前的做法是使用 IDF5.2 烧录证书，然后使用IDF5.2.2编译运行

AdityaHPatwardhan, 2024-08-06:
This issue seems identical to https://github.com/espressif/esp-idf/issues/14300, and can be followed there itself.
Closing this issue.

closed 2024-08-06 by AdityaHPatwardhan