espressif/esp-idf #14212: Likeliness of 368b7eb also impacting ESP32-S3?

JVKran, 2024-07-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How likely would it be that the fixes listed over here also apply to the ESP32-S3? We're currently having issues where we think the Wi-Fi stack fails to deinitialize (due to some reason) which is blocking the ESP32-S3 from entering deep-sleep. We've now updated to V5.1.4 (i.c.w. ESP-ADF V2.6) and are going to test if this doesn't happen anymore.

esp-wzh, 2024-07-22:
This fix is ​​not for esp32s3, could you describe the detailed phenomenon when the error occurs?

JVKran, 2024-07-22:
We had an issue with application logic which didn't manage the case when there wasn't a wifi connection, but we also hadn't fired up the AP yet. This resulted in the Wi-Fi stack not being deinitialized when entering deep-sleep. Sometimes, deep-sleep wasn't entered but the main-loop appeared like it was no longer running. Some other tasks in the background were still active. The Wi-Fi connection sometimes was also still active.

esp-wzh, 2024-07-22:
> Some other tasks in the background were still active. The Wi-Fi connection sometimes was also still active.

This situation should not happen. The software will ensure that the process of entering sleep is single-threaded. The core requesting sleep will enter the [critical section](https://github.com/espressif/esp-idf/blob/v5.1.4/components/esp_hw_support/sleep_modes.c#L976) and [stop other cores](https://github.com/espressif/esp-idf/blob/v5.1.4/components/esp_hw_support/sleep_modes.c#L977), and then prepare for sleep. Even if entering sleep fails, `esp_deep_sleep_start` function [will not return](https://github.com/espressif/esp-idf/blob/v5.1.4/components/esp_hw_support/sleep_modes.c#L1036) but trigger the watchdog.

closed 2024-11-13 by Sherry616