espressif/esp-idf #14214: ESP32 S2 SOLO-2 Wifi driver is not working

Abhilashtp74, 2024-07-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Development Kit: ESP32-S2
Module or chip used: ESP32-S2-2
IDF Version-4.4.6
VSCODE-Yes

I am Running ESP-IDF Example code for wifi_station, If wifi related code is not used then rest of the code is works fine, when i include the wifi initialization the error was got mention below. I have run the same code on ESP32-S2-SOLO-2 development board its works fine.

This the log i get after flashing the code to it
I (22) boot: ESP-IDF v4.4.6-98-g5f257494c5-dirty 2nd stage bootloader
I (22) boot: compile time 13:00:48
I (23) boot: chip revision: v1.0
I (27) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : DIO
I (36) boot.esp32s2: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1f0b0h (127152) map
I (118) esp_image: segment 1: paddr=0002f0d8 vaddr=3ffc6aa0 size=00f40h (  3904) load
I (119) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=997f8h (628728) map
I (249) esp_image: segment 3: paddr=000c9820 vaddr=3ffc79e0 size=02374h (  9076) load
I (251) esp_image: segment 4: paddr=000cbb9c vaddr=40022000 size=14aa0h ( 84640) load
I (285) boot: Loaded app from partition at offset 0x10000
I (285) boot: Disabling RNG early entropy source...
I (297) cpu_start: Unicore app
I (297) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (299) cpu_start: Pro cpu up.
I (315) cpu_start: Pro cpu start user code
I (315) cpu_start: cpu freq: 160000000
I (315) cpu_start: Application information:
I (319) cpu_start: Project name:     terra-m-firmware
I (325) cpu_start: App version:      v4.4.5-dirty
I (330) cpu_start: Compile time:     Jun  7 2024 13:00:06
I (337) cpu_start: ELF file SHA256:  4ec3cb20e0f806e6...
I (343) cpu_start: ESP-IDF:          v4.4.6-98-g5f257494c5-dirty
I (349) cpu_start: Min chip rev:     v0.0
I (354) cpu_start: Max chip rev:     v1.99
I (359) cpu_start: Chip rev:         v1.0
I (364) heap_init: Initializing. RAM available for dynamic allocation:
I (371) heap_init: At 3FFCF310 len 0002CCF0 (179 KiB): DRAM
I (377) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (383) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (390) spi_flash: detected chip: generic
I (394) spi_flash: flash io: dio
I (399) cpu_start: Starting scheduler on PRO CPU.
I (404) uart: queue free spaces: 30
E (1414) AP_wifi_webserver: Error (ESP_ERR_NVS_NOT_FOUND) reading SSID from NVS!
E (1414) AP_wifi_webserver: Error (ESP_ERR_NVS_NOT_FOUND) reading password from NVS!
I (1424) wifi:wifi driver task: 3ffd8afc, prio:23, stack:6656, core=0
I (1424) system_api: Base MAC address is not set
I (1424) system_api: read default base MAC address from EFUSE
I (1444) wifi:wifi firmware version: d9959e6
I (1444) wifi:wifi certification version: v7.0
I (1444) wifi:config NVS flash: enabled
I (1444) wifi:config nano formating: disabled
I (1454) wifi:Init data frame dynamic rx buffer num: 32
I (1454) wifi:Init management frame dynamic rx buffer num: 32
I (1464) wifi:Init management short buffer num: 32
I (1464) wifi:Init dynamic tx buffer num: 32
I (1464) wifi:Init static rx buffer size: 1600
I (1474) wifi:Init static rx buffer num: 10
I (1474) wifi:Init dynamic rx buffer num: 32
I (1484) wifi_init: rx ba win: 6
I (1484) wifi_init: tcpip mbox: 32
I (1484) wifi_init: udp mbox: 6
I (1494) wifi_init: tcp mbox: 6
I (1494) wifi_init: tcp tx win: 5744
I (1504) wifi_init: tcp rx win: 5744
I (1504) wifi_init: tcp mss: 1440
I (1514) wifi_init: WiFi IRAM OP enabled
I (1514) wifi_init: WiFi RX IRAM OP enabled
I (1524) phy_init: phy_version 2401,e5f86f8,Jul 22 2023,14:39:14
W (1524) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1544) phy: pll_cap_ext 10
I (1544) phy: pll_cap_ext 10
I (1544) phy: pll_cap_ext 10
I (1544) phy: pll_cap_ext 10
I (1554) phy: pll_cap_ext 10
I (1554) phy: pll_cap_ext 10
I (1554) phy: pll_cap_ext 10
I (1564) phy: pll_cap_ext 10
I (1564) phy: pll_cap_ext 10
I (1574) phy: pll_cap_ext 10
I (1574) phy: pll_cap_ext 10
I (1574) phy: pll_cap_ext 10
I (1584) phy: pll_cap_ext 10
I (1594) phy: pll_cap_ext 10
I (1594) phy: pll_cap_ext 10
I (1594) phy: pll_cap_ext 10
I (1594) phy: pll_cap_ext 10
I (1594) phy: pll_cap_ext 10
I (1604) phy: pll_cap_ext 10
I (1614) phy: pll_cap_ext 10
I (1624) phy: pll_cap_ext 10
I (1624) phy: pll_cap_ext 10
I (1624) phy: pll_cap_ext 10
I (1624) phy: pll_cap_ext 10
I (1634) phy: pll_cap_ext 10
I (1634) phy: pll_cap_ext 10
I (1634) phy: pll_cap_ext 10
I (1634) phy: pll_cap_ext 10
I (1644) phy: pll_cap_ext 10
I (1644) phy: pll_cap_ext 10
I (1644) phy: pll_cap_ext 10
I (1654) phy: pll_cap_ext 10
I (1654) phy: pll_cap_ext 10
I (1654) phy: pll_cap_ext 10

abort() was called at PC 0x40024dfb on core 0
Setting breakpoint at 0x40023909 and returning...

MaxwellAlan, 2024-07-19:
Hi @Abhilashtp74 

> I (1544) phy: pll_cap_ext 10
I (1544) phy: pll_cap_ext 10

These log means the PHY calibration fail, could you try with the IDF v4.4.8 or the latest master ?

Abhilashtp74, 2024-07-19:
Hi @MaxwellAlan 
I have tried with IDF v4.4.8 but still getting same error mentioned above

Abhilashtp74, 2024-07-19:
Hi @MaxwellAlan 
When I debugged using JTAG programmer I get this log

Thread 9 "wifi" received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1073564652]
0x4002380d in panic_abort (details=0x3ffd444b "abort() was called at PC 0x40024c6b on core 0") at C:/Users/Abhilash/esp/v4.4.8/esp-idf/components/esp_system/panic.c:408
408	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets

MaxwellAlan, 2024-07-31:
Hi @Abhilashtp74 

Sorry for reply late.

The cause of this issue might be related to the internal PHY failing calibration, potentially due to unstable power supply to the board. Based on our previous experience, possible issues include: 1. Power supply to the board; 2. Some GPIO pins having input voltages that exceed 3.3V; 3. Potential hardware damage on the board.

Is this issue an isolated case or a more common occurrence on your end?

Abhilashtp74, 2024-07-31:
Issue was resolved, the reason was tx pin was pulled up by 5v. Thank You for the Update

closed 2024-07-31 by Abhilashtp74