espressif/esp-idf #14219: ULP-RISCV GPIO-interrupt example question

JVKran, 2024-07-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The example states the following;

/* Wakeup interrupt is a level interrupt, wait 1 sec to
       allow user to release button to avoid waking up the ULP multiple times */

I however wonder why that would pose a problem. ulp_riscv_wakeup_main_procesor will be called everytime, but the main processor will already be running after the first one.

KonstantinKondrashov, 2024-07-19:
Hi @JVKran!
Once the button is pressed [the main ULP function ](https://github.com/espressif/esp-idf/blob/v5.1.4/examples/system/ulp_riscv/gpio_interrupt/main/ulp/main.c#L20) is called. It wakes up the HP core and makes the ULP core ready to handle the button again. If there is no wait function, the main ULP function will be running multiple times because there is a contact bounce after pressing the button, leading to multiple waking-ups for ULP and HP core. (Running HP core is power-consuming, which we want to reduce.). HP core after waking up goes to sleep again.

When the main ULP function finishes its execution then [the ulp_riscv_halt()](https://github.com/espressif/esp-idf/blob/v5.1.4/examples/system/ulp_riscv/gpio_interrupt/main/ulp/main.c#L29C8-L29C22) is called. It means ULP is ready for the next wake-up.
```c
/**
 * @brief Halts the program currently running on the ULP-RISC-V
 *
 * @note  Program will restart at the next ULP timer trigger if timer is still running.
 *        If you want to stop the ULP from waking up then call ulp_riscv_timer_stop() first.
 */
void ulp_riscv_halt(void);
```

KonstantinKondrashov, 2024-07-25:
I close this issue, feel free to reopen it. Thanks.

closed 2024-07-25 by KonstantinKondrashov