espressif/esp-idf #14220: MESH: Root looses IP when a new node connects to the network

luisyague, 2024-07-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi there, I am having a problem while connecting a new node to my mesh network: 

- **IDF Version**: release 5.2 (commit cc649ea869b8c24d19ba2d94c4d3f1146cb2947b)
- **Board**: custom
- **Setup**: mesh network with one root and one node already installed, trying to connect a second node (although it looses ip also while connecting the first one
- **Expected Behaviour**: the root should accept a new node without losing connection to the router
- **Actual behaviour**: the root looses IP when the new node attemps to connect to the network
- **sdkconfig**:  [sdkconfig.txt](https://github.com/user-attachments/files/16310770/sdkconfig.txt)
- **log**:  [log.txt](https://github.com/user-attachments/files/16310765/log.txt)

Is there something in the sdkconfig i should change in order the prevent this issue?

Thank you very much in advance!

zhangyanjiaoesp, 2024-08-12:
@luisyague 
1. which ESP device do you use?
2. Can you provide a demo for us to test?

luisyague, 2024-08-12:
Hello, @zhangyanjiaoesp I am using a custom-made board with ESP32 32E, unfortunately as this is production firmware so I can't give you a demo, but I am using a pretty straightforward approach: 
1. I am using both cores (core 0 for stack threads and wifi-mesh and core 1 for my app thread (it is just a heater app which uses adc, spi some digital outputs)
2. I am using secure-boot + flash encryption (it also happens without flash-encryption)
3. Then my mesh application just calls some initilizations from esp-mdf with the default values, I know it is no longer supported but it is just a wrapper for esp-mesh so I dont think it is related.

zhangyanjiaoesp, 2024-08-12:
@luisyague 
It should related to the application code. Because our official mesh example can works well. And according to the existing log, I can't determine what the problem is.

luisyague, 2024-10-08:
Sorry for the delay, okei i'll take a look at the application code then, and will make a post with the solution if i find something. Thanks a lot for the support!!!

luisyague, 2024-10-15:
Found it, pretty weird... I just changed th defaut channel from 1 to 0 and started working without problems.  Hope this helps anyone with the same problem.

closed 2024-10-15 by luisyague