espressif/esp-idf #14221: A stack overflow in task ble_hidd_events has been detected

mike-2020, 2024-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3, with 8M PSRAM

### Power Supply used.

USB

### What is the expected behavior?

No crash

### What is the actual behavior?

A stack overflow in task ble_hidd_events has been detected

### Steps to reproduce.

Crash at booting time

### Debug Logs.

```plain
I (21204) BLE_INIT: BT controller compile version [c23ab4c]
I (21214) BLE_INIT: Bluetooth MAC: ec:da:3b:95:5a:d2

-----------listening-----------
I (21224) HID_DEV_DEMO: setting ble device
I (21234) HID_DEV_DEMO: BLE Host Task Started
I (21244) NimBLE: GAP procedure initiated: stop advertising.

I (21254) HID_DEV_BLE: START

RROR*** A stack overflow in task ble_hidd_events has been detected.

Backtrace: 0x40379f92:0x3fcecf50 0x40388ec9:0x3fcecf70 0x40389f3e:0x3fcecf90 0x4038b713:0x3fced010 0x4038a004:0x3fced030 0x40389ffa:0x00000000 |<-CORRUPTED
0x40379f92: panic_abort at /home/mike/esp-idf-v5.2.2/components/esp_system/panic.c:466
0x40388ec9: esp_system_abort at /home/mike/esp-idf-v5.2.2/components/esp_system/port/esp_system_chip.c:93
0x40389f3e: vApplicationStackOverflowHook at /home/mike/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:553
0x4038b713: vTaskSwitchContext at /home/mike/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/tasks.c:3664 (discriminator 7)
0x4038a004: _frxt_dispatch at /home/mike/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x40389ffa: _frxt_int_exit at /home/mike/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246
```

### More Information.

This is not a new issue. It has been reported in the past, and I observed the stack size has been changed from 2048 to 4096 in IDF 5.2.2. Howerver, it still reports the stack overflow issue.

I am using NimBLE.

rahult-github, 2024-08-06:
Hi @mike-2020 , 

Looks like the operations in the context of the thread is exceeding the size. May i request to increase the size in this example and check once ?

rahult-github, 2024-09-10:
Hi @mike-2020  , did increasing stack size help resolve the issue for you ?

mike-2020, 2024-09-11:
Yes, it works, thanks.

closed 2024-09-11 by mike-2020