espressif/esp-idf #14223: twai_driver_install_v2 multiple instance error

lomal788, 2024-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2, but master branch does the same

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

twai driver install multiple instance.q

### What is the actual behavior?

first instance was successfully install, started.
but second instance is not started. ( error : ESP_ERR_INVALID_ARG ( 0x102 ) )

### Steps to reproduce.

same as this code.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/twai.html?highlight=twai#install-multiple-twai-instances

### More Information.

only 1 instance was successfully work.

I found theres code in twai_driver_install_v2 function
theres   **"TWAI_CHECK(g_config->controller_id < SOC_TWAI_CONTROLLER_NUM, ESP_ERR_INVALID_ARG);"**
SOC_TWAI_CONTROLLER_NUM defined 1UL which only controller_id = 0 is work.

if I make controller_id as 0 as the second instance it successfully pass the TWAI_CHECK area and change error code : ESP_ERR_INVALID_STATE - > which already client is using.

SOC_TWAI_CONTROLLER_NUM should be editable by user or it should over a 2

atanisoft, 2024-07-21:
The ESP32 (base variant) only has a single TWAI/CAN controller in the hardware. You can not install a second instance.

The only ESP32 variants with additional controllers are ESP32-P4 (3), ESP32-C5 (2) and ESP32-C6 (2).

atanisoft, 2024-07-22:
@lomal788 that should work since the MCP2515 is an SPI device and wouldn't depend on the TWAI driver in any way, you would need to use a compatible driver for that IC with it's API.

wanckl, 2024-10-09:
@lomal788  Hi, dose issue resolved

Alvin1Zhang, 2025-02-17:
Thanks for reporting, will close due to short of updates, feel free to reopen with more updates.

closed 2025-02-17 by Alvin1Zhang