espressif/esp-idf #14224: stack overflow error when running blehr example

Zyuangeorge, 2024-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

Expect to see the ble notify behaviour, with the value shown on the ble monitor (Lightblue or nRF connect)
I (21611) BLE_GAP_SUBSCRIBE_EVENT: conn_handle from subscribe=0
GATT procedure initiated: notify; att_handle=3
GATT procedure initiated: notify; att_handle=3
GATT procedure initiated: notify; att_handle=3
GATT procedure initiated: notify; att_handle=3
GATT procedure initiated: notify; att_handle=3
GATT procedure initiated: notify; att_handle=3
GATT procedure initiated: notify; att_handle=3

### What is the actual behavior?

Saw an error: ***ERROR*** A stack overflow in task Tmr Svc has been detected at the serial monitor output. And the ble connection is terminated automatically.

### Steps to reproduce.

1. File->New->Espressif IDF Project->create a project using one of the templates->nimble/blehr
2. Build the project
3. Run the project on ESP32-S3

### Debug Logs.

```plain
I (16069) NimBLE: connection established; status=0

I (17189) NimBLE: subscribe event; cur_notify=1
 value handle; val_handle=16

I (17189) BLE_GAP_SUBSCRIBE_EVENT: conn_handle from subscribe=1
I (18189) NimBLE: GATT procedure initiated: notify; 
I (18189) NimBLE: att_handle=16

***ERROR*** A stack overflow in task Tmr Svc has been detected.

Backtrace: 0x40375a4e:0x3fca3850 0x4037f691:0x3fca3870 0x403805c2:0x3fca3890 0x40381bef:0x3fca3910 0x40380688:0x3fca3930 0x4038067e:0x00000000 |<-CORRUPTED
0x40375a4e: panic_abort at D:/ESP_IDF/components/esp_system/panic.c:466
0x4037f691: esp_system_abort at D:/ESP_IDF/components/esp_system/port/esp_system_chip.c:93
0x403805c2: vApplicationStackOverflowHook at D:/ESP_IDF/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:553
0x40381bef: vTaskSwitchContext at D:/ESP_IDF/components/freertos/FreeRTOS-Kernel/tasks.c:3664 (discriminator 7)
0x40380688: _frxt_dispatch at D:/ESP_IDF/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x4038067e: _frxt_int_exit at D:/ESP_IDF/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246

ELF file SHA256: 13869ba09

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759c4
0x403759c4: esp_restart_noos at D:/ESP_IDF/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2d9c
entry 0x403c9914
I (31) boot: ESP-IDF v5.2.2 2nd stage bootloader
I (31) boot: compile time Jul 20 2024 16:05:19
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.1
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=19310h (103184) map
I (122) esp_image: segment 1: paddr=00029338 vaddr=3fc98400 size=04074h ( 16500) load
I (126) esp_image: segment 2: paddr=0002d3b4 vaddr=40374000 size=02c64h ( 11364) load
I (131) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=52154h (336212) map
I (197) esp_image: segment 4: paddr=0008217c vaddr=40376c64 size=116cch ( 71372) load
I (221) boot: Loaded app from partition at offset 0x10000
I (221) boot: Disabling RNG early entropy source...
I (232) cpu_start: Multicore app
I (242) cpu_start: Pro cpu start user code
I (242) cpu_start: cpu freq: 160000000 Hz
I (242) cpu_start: Application information:
I (245) cpu_start: Project name:     blehr
I (250) cpu_start: App version:      1
I (254) cpu_start: Compile time:     Jul 20 2024 16:07:59
I (260) cpu_start: ELF file SHA256:  13869ba09...
I (266) cpu_start: ESP-IDF:          v5.2.2
I (271) cpu_start: Min chip rev:     v0.0
I (275) cpu_start: Max chip rev:     v0.99 
I (280) cpu_start: Chip rev:         v0.1
I (285) heap_init: Initializing. RAM available for dynamic allocation:
I (292) heap_init: At 3FC9E7C8 len 0004AF48 (299 KiB): RAM
I (298) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (304) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (311) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (318) spi_flash: detected chip: generic
I (321) spi_flash: flash io: dio
W (325) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (339) sleep: Configure to isolate all GPIO pins in sleep state
I (345) sleep: Enable automatic switching of GPIO sleep configuration
I (352) coexist: coex firmware version: d96c1e51f
I (358) coexist: coexist rom version e7ae62f
I (363) main_task: Started on CPU0
I (373) main_task: Calling app_main()
I (383) BLE_INIT: BT controller compile version [c23ab4c]
I (393) BLE_INIT: Bluetooth MAC: f4:12:fa:40:3f:86
I (393) phy_init: phy_version 670,b7bc9b9,Apr 30 2024,10:54:13
I (433) NimBLE_BLE_HeartRate: BLE Host Task Started
I (443) NimBLE: GAP procedure initiated: stop advertising.

I (443) NimBLE: Failed to restore IRKs from store; status=8

I (443) NimBLE: Device Address: 
I (443) NimBLE: f4:12:fa:40:3f:86
I (453) NimBLE: 

I (453) NimBLE: GAP procedure initiated: advertise; 
I (463) NimBLE: disc_mode=2
I (463) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (473) NimBLE: 

I (473) main_task: Returned from app_main()
```

rahult-github, 2024-07-21:
Freertos timer task for s3 is falling short in size . The error is from tvc service .

FREERTOS_TIMER_TASK_STACK_DEPTH .. increase this , may be make it double , say 4000 and check once

Zyuangeorge, 2024-07-21:
> Freertos timer task for s3 is falling short in size . The error is from tvc service .
> 
> FREERTOS_TIMER_TASK_STACK_DEPTH .. increase this , may be make it double , say 4000 and check once

Thanks it works for me

closed 2024-07-21 by Zyuangeorge