espressif/esp-idf #14225: IDF freeRtos task contex switch DO NOT release the spinlock

xmusco, 2024-07-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 DEV KIT

### Power Supply used.

USB

### What is the expected behavior?

Hello,
The previous bug still happen in some case.
https://github.com/espressif/esp-idf/issues/14210

### What is the actual behavior?

please run the follow demo will get a fail.

### Steps to reproduce.

// code

#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "esp_log.h"

const char* TAG = "LAND";
// static portMUX_TYPE my_spinlock = portMUX_INITIALIZER_UNLOCKED;

#define TASK1_PRIO      3                   /* 任务优先级 */
#define TASK1_STK_SIZE  1024                 /* 任务堆栈大小 */
TaskHandle_t            Task1Task_Handler;  /* 任务句柄 */

#define TASK2_PRIO      3                   /* 任务优先级 */
#define TASK2_STK_SIZE  1024                 /* 任务堆栈大小 */
TaskHandle_t            Task2Task_Handler;  /* 任务句柄 */

SemaphoreHandle_t BinarySemaphore;          /* 二值信号量句柄 */

void task1(void *pvParameters)
{
    ESP_LOGI(TAG,"give");
    vTaskDelay(pdMS_TO_TICKS(2000));
    xSemaphoreGive(BinarySemaphore);                    /* 释放二值信号量 */
    ESP_LOGI(TAG,"give ok");
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 1 is running\n");
    }
}

void task2(void *pvParameters)
{
    ESP_LOGI(TAG,"take");
    vTaskDelay(pdMS_TO_TICKS(5000));
    xSemaphoreTake(BinarySemaphore, portMAX_DELAY);
    ESP_LOGI(TAG,"take ok");
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
        ESP_LOGI(TAG, "task 2 is running\n");
    }
}

void app_main(void)
{
    ESP_LOGI(TAG,"main start");
    // taskENTER_CRITICAL(&my_spinlock);           /* 进入临界区 */
    /* 创建二值信号量 */
    BinarySemaphore = xSemaphoreCreateBinary();
    /* 创建任务1 */
    xTaskCreate((TaskFunction_t )task1,
                (const char*    )"task1",
                (uint16_t       )TASK1_STK_SIZE,
                (void*          )NULL,
                (UBaseType_t    )TASK1_PRIO,
                (TaskHandle_t*  )&Task1Task_Handler
                );
    /* 创建任务2 */
    xTaskCreate((TaskFunction_t )task2,
                (const char*    )"task2",
                (uint16_t       )TASK2_STK_SIZE,
                (void*          )NULL,
                (UBaseType_t    )TASK2_PRIO,
                (TaskHandle_t*  )&Task2Task_Handler
                );
    
    // taskEXIT_CRITICAL(&my_spinlock);            /* 退出临界区 */
    ESP_LOGI(TAG,"main finish");
}

### Debug Logs.

```plain
assert failed: xTaskRemoveFromEventList tasks.c:3860 (pxUnblockedTCB)

Backtrace: 0x40375906:0x3fc9a6d0 0x4037a1f5:0x3fc9a6f0 0x403805f9:0x3fc9a710 0x4037c06a:0x3fc9a830 0x4037a8dc:0x3fc9a850 0x4037a98e:0x3fc9a890 0x4037688d:0x3fc9a8b0 0x4037696d:0x3fc9a8e0 0x42008153:0x3fc9a900 0x42006cc1:0x3fc9a920 0x42006832:0x3fc9a940 0x4200ef12:0x3fc9a960 0x4200e77d:0x3fc9a980 0x4200e7da:0x3fc9a9a0 0x4200ee19:0x3fc9a9d0 0x42012e2b:0x3fc9aa00 0x420127ee:0x3fc9aa20 0x4200ef89:0x3fc9ad40 0x4201c4ad:0x3fc9ad70 0x4038036d:0x3fc9ada0 0x420081d9:0x3fc9adf0 0x4037aa59:0x3fc9ae10
```

KonstantinKondrashov, 2024-07-21:
Hi @xmusco!
1k stack size is too small especially if you are doing printing from a task. If you increase it up to 2k do you still see the issue?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html?highlight=minimal%20stack%20size#determining-stack-size

KonstantinKondrashov, 2024-07-22:
printf (and related log functions) requires a lot of stack, so 1024 is not sufficient.

closed 2024-07-22 by KonstantinKondrashov