espressif/esp-idf #14226: Support running esp_http_client applications on linux host

0xjmux, 2024-07-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I need to develop an ESP-IDF component to integrate with a currently unsupported API, and would much prefer being able to do the development legwork of interacting with the API from a desktop environment with full debugging tools. According to [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_client.html#overview), esp_http_client doesn't support mocking, and my attempt to compile it for native was unsuccessful. 

Being able to build it for linux natively like you can with the tcp_client examples would make developing API wrappers for ESP-IDF much easier and more straightforward.

mahavirj, 2024-07-23:
@0xjmux 

HTTP client example already has "Linux" under supported targets list:
https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_http_client

Are you facing any issues building this example under Linux host? If yes, could you please share the detailed build log?

0xjmux, 2024-07-24:
Oh, I didn't see it was supported in that example - I saw that it was listed as unsupported in the docs, and my attempt with another example wasn't successful, so I filed the issue. Thanks for pointing me towards that; I was able to get it to compile successfully.

closed 2024-07-24 by 0xjmux