espressif/esp-idf #14227: Heap leak in esp_http_client

supersunil, 2024-07-21:
Hello there,

While working with the [esp_http_client](https://github.com/espressif/esp-idf/tree/v5.2.1/examples/protocols/esp_http_client) I've came across a constant increase in the heap memory after each http actions and eventually stops performing the http actions (POST or GET).

Attached the snippet used for heap monitoring.

```c
static size_t preHeap = 0;
static size_t curHeap = 0;
..
..

static void sys_monitor(void *pvParameters)
{
while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(1000));
       curHeap  =  heap_caps_get_free_size(MALLOC_CAP_DEFAULT);
        printf("Diff : %d\n",(preHeap -curHeap));
    }
}

static void http_test_task(void *pvParameters)
{    
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(10000));       
        http_native_request();
    }
    vTaskDelete(NULL);
}
void app_main(void)
{
...
...
curHeap  =  heap_caps_get_free_size(MALLOC_CAP_DEFAULT);
preHeap  =  heap_caps_get_free_size(MALLOC_CAP_DEFAULT);
xTaskCreate(&sys_monitor, "monitor", 2048, NULL, 15, NULL);
xTaskCreate(&http_test_task, "http_test_task", 8192, NULL, 5, NULL);
....
}
```

What could be the potential issue in here?

Note: I've a device running with a log, will post it here soon, if that helps.

hrushikesh430, 2024-07-24:
Hello supersonic,
Can I get all your source code, including your http_native_request function. It will help me to debug and check what's happening.

supersunil, 2024-07-24:
Hi @hrushikesh430 ,

Here I am attaching the main.c file and the result screenshot,

```c
/* ESP HTTP Client Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include <sys/param.h>
#include <stdlib.h>
#include <ctype.h>
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_event.h"
#include "esp_netif.h"
#include "protocol_examples_common.h"
#include "protocol_examples_utils.h"
#include "esp_tls.h"
#if CONFIG_MBEDTLS_CERTIFICATE_BUNDLE
#include "esp_crt_bundle.h"
#endif

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"

#include "esp_http_client.h"

#include "esp_heap_trace.h"
#define NUM_RECORDS 1000
static heap_trace_record_t trace_record[NUM_RECORDS];

#define MAX_HTTP_RECV_BUFFER 512
#define MAX_HTTP_OUTPUT_BUFFER 2048
static const char *TAG = "HTTP_CLIENT";

/* Root cert for howsmyssl.com, taken from howsmyssl_com_root_cert.pem

   The PEM file was extracted from the output of this command:
   openssl s_client -showcerts -connect www.howsmyssl.com:443 </dev/null

   The CA root cert is the last cert given in the chain of certs.

   To embed it in the app binary, the PEM file is named
   in the component.mk COMPONENT_EMBED_TXTFILES variable.
*/
extern const char howsmyssl_com_root_cert_pem_start[] asm("_binary_howsmyssl_com_root_cert_pem_start");
extern const char howsmyssl_com_root_cert_pem_end[]   asm("_binary_howsmyssl_com_root_cert_pem_end");

extern const char postman_root_cert_pem_start[] asm("_binary_postman_root_cert_pem_start");
extern const char postman_root_cert_pem_end[]   asm("_binary_postman_root_cert_pem_end");

static char memBuffer[128];
static size_t preHeap = 0;
static size_t curHeap = 0;

void SystemBuffer()
{
	curHeap  =  heap_caps_get_free_size(MALLOC_CAP_DEFAULT);
	sprintf(memBuffer,
		"   Biggest /     Free /    Total\n"
		"\t  DRAM : [%8d / %8d / %8d]",
		heap_caps_get_largest_free_block(MALLOC_CAP_DEFAULT), heap_caps_get_free_size(MALLOC_CAP_DEFAULT),
[... 301 lines cut ...]
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());
    ESP_LOGI(TAG, "Connected to AP, begin http example");
    curHeap  =  heap_caps_get_free_size(MALLOC_CAP_DEFAULT);
	preHeap  =  heap_caps_get_free_size(MALLOC_CAP_DEFAULT);
#if CONFIG_IDF_TARGET_LINUX
    http_test_task(NULL);
#else
    xTaskCreate(&sys_monitor, "monitor", 2048, NULL, 15, NULL);
    xTaskCreate(&http_test_task, "http_test_task", 8192, NULL, 5, NULL);
#endif
}
```
[image]
[image]

hrushikesh430, 2024-07-26:
Hello @supersunil 
I tried to reproduce your issue. But I see no heap memory leak. After an interval of time, free heap size remains constant. But yes the reason why free heap size is getting reduced after particular interval of time is due to systems background
processes. 

for reference you can check [this](https://github.com/espressif/esp-idf/issues/11892)

and you want to debug your code for heap memory, you can refer [this](https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32c6/api-reference/system/heap_debug.html?highlight=heap%20debugging)

hrushikesh430, 2024-08-13:
Closing this issue due to lack of response. Please feel free to reopen if your needed.

closed 2024-08-13 by hrushikesh430