espressif/esp-idf #14228: How do I add my own service to gatt_security_server?

Ycc33, 2024-07-22:
### Is your feature request related to a problem?

How do I add my own service to gatt_security_server?Currently I have successfully added the service in gatt_server, but I can't connect with the phone, only using gatt_security_server I can connect with the phone properly, so I would like to ask how to add my own service in that routine, looking forward to your reply, thank you very much!

esp-zhp, 2024-07-29:
In the esp-idf/examples/bluetooth/bluedroid/ble/gatt_server example, the SMP (Security Manager Protocol) feature is disabled. As a result, some phones may not respond to pairing requests and may disconnect shortly after connecting.

The esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server example has the SMP feature enabled. If you intend to use this with a phone, I recommend using the esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server example.

You can refer to the service creation process in this example to create your own services.

[image]

Ycc33, 2024-08-07:
@esp-zhp Thank you very much for your reply!

esp-zhp, 2024-08-07:
If there are no further issues, please close this issue.

closed 2024-08-07 by Ycc33