espressif/esp-idf #14233: Monitor requires standard input to be attached to TTY

EvgeniyVol, 2024-07-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

https://github.com/espressif/esp-idf/releases/tag/v5.2.2

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The COM port will be read

### What is the actual behavior?

An error is coming out

### Steps to reproduce.

1. ESPPORT COM5
2. flash build
2. monitor build

### Build or installation Logs.

```plain
====================[ Build | monitor | Debug ]=================================
"D:\CLion 2024.1.4\bin\cmake\win\x64\bin\cmake.exe" --build D:\ESP32EDF\ESP32Project\NewProject\hello_world\cmake-build-debug --target monitor -j 14
[0/1] C:\WINDOWS\system32\cmd.exe /C "cd /D D:\ESP32EDF\ESP32CLINE\esp-idf-v5.2.2\components\esptool_py && "D:\CLion 2024.1.4\bin\cmake\win\x64\bin\cmake.exe" -D IDF_PATH=D:/ESP32EDF/ESP32CLINE/esp-idf-v5.2.2 -D SERIAL_TOOL=python;-m;esp_idf_monitor -D SERIAL_TOOL_ARGS=--target;esp32;--revision;0;D:/ESP32EDF/ESP32Project/NewProject/hello_world/cmake-build-debug/hello_world.elf -D WORKING_DIRECTORY=D:/ESP32EDF/ESP32Project/NewProject/hello_world/cmake-build-debug -P run_serial_tool.cmake"
error: Monitor requires standard input to be attached to TTY
CMake Error at run_serial_tool.cmake:66 (message):
  python;-m;esp_idf_monitor failed.

FAILED: CMakeFiles/monitor D:/ESP32EDF/ESP32Project/NewProject/hello_world/cmake-build-debug/CMakeFiles/monitor 
C:\WINDOWS\system32\cmd.exe /C "cd /D D:\ESP32EDF\ESP32CLINE\esp-idf-v5.2.2\components\esptool_py && "D:\CLion 2024.1.4\bin\cmake\win\x64\bin\cmake.exe" -D IDF_PATH=D:/ESP32EDF/ESP32CLINE/esp-idf-v5.2.2 -D SERIAL_TOOL=python;-m;esp_idf_monitor -D SERIAL_TOOL_ARGS=--target;esp32;--revision;0;D:/ESP32EDF/ESP32Project/NewProject/hello_world/cmake-build-debug/hello_world.elf -D WORKING_DIRECTORY=D:/ESP32EDF/ESP32Project/NewProject/hello_world/cmake-build-debug -P run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
```

peterdragun, 2024-07-23:
Hi @EvgeniyVol, can you please provide some more details? How are you executing the monitor? Are you using some IDE shortcut? Or are you executing the command in the integrated terminal of IDE?

If you are using some button in the IDE to execute the monitor, please make sure that you have followed the installation guide on the Clion website: https://www.jetbrains.com/help/clion/esp-idf.html

Please ensure you can execute the monitor using the `idf.py monitor` or `python -m esp_idf_monitor` command. Note that both commands need to be run in the ESP-IDF environment. Both the command line and PowerShell should work in this case.

EvgeniyVol, 2024-07-23:
1. IDE CLion 2024.1.4
2. I use the configuration in the upper right corner and the build button
3. Everything works in IDE terminal
5. I did as it says it doesn't work https://www.jetbrains.com/help/clion/esp-idf.html
6. In command line everything works

peterdragun, 2024-07-24:
If everything works in the terminal, I am afraid that we cannot do anything about that from the ESP-IDF side. Please report the bug to CLion.

dobairoland, 2024-07-24:
It is interesting that the linked page has a video tutorial and this issue is not happening there. IDF Monitor requires STDIN to be connected to it from the very beginning because it is an interactive tool and is not just receiving but also sending characters to the ESP. So I guess something must have changed on the Clion side. I agree that this should be address on their side.

However, we might think about to add a non-interactive mode for IDF Monitor in the future when it won't send anything and those related features would be disabled.

EvgeniyVol, 2024-07-24:
Maybe it will help someone, https://youtrack.jetbrains.com/issue/CPP-34647/ESP-IDF-Monitor-error-Monitor-requires-standard-input-to-be-attached-to-TTY. I have checked this method works

closed 2024-07-30 by EvgeniyVol