espressif/esp-idf #14234: Wifi connection loss is not recognised

christian94M, 2024-07-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v1.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

NMCU-ESP32 (berry base)

### Power Supply used.

USB

### What is the expected behavior?

This ESP-IDF program initializes the ESP32 to connect to a Wi-Fi network as a station (STA) and, if it fails to connect after a specified number of retries, it switches to SoftAP (Access Point) mode.

### What is the actual behavior?

no event handle or message after connection loss 

### Steps to reproduce.

1. Step ESP32 to connect to a Wi-Fi network as a station (STA) 
2. Step shutdown the Wifi (Router, etc)
3. Step it is not recognised that the connection is lost and no AP is created

### Debug Logs.

```plain
I (4184) main_task: Returned from app_main()
I (40784) wifi:state: run -> init (8a0)
I (40784) wifi:pm stop, total sleep time: 32368745 us / 37621739 us

I (40784) wifi:<ba-del>idx:0, tid:6
I (40794) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1

 *  Task wird ausgeführt: C:\Users\ok99m\.espressif\tools\cmake\3.24.0\bin\cmake.exe -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=c:\Users\ok99m\VSC\ESP32\test\build -S=c:\Users\ok99m\VSC\ESP32\test 

-- Found Git: C:/Users/ok99m/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/ok99m/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/ok99m/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 332 lines cut ...]
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "ESP32_AP",
            .ssid_len = strlen("ESP32_AP"),
            .password = "esp32_password",
            .max_connection = 4,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };
    if (strlen("esp32_password") == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "wifi_init_softap finished. SSID:%s password:%s",
             "ESP32_AP", "esp32_password");
}

void app_main(void) {
    ESP_ERROR_CHECK(nvs_flash_init());
    wifi_init_sta();
}

Alvin1Zhang, 2024-07-23:
Thanks for reporting.

hansw123, 2025-01-24:
@christian94M 
sorry for late reply, you should put your change wifi mode to ap releated code to WIFI_EVENT_STA_DISCONNECTED event handle func ,otherwise when wifi_init_sta() func finished , no func will process the WIFI_EVENT_STA_DISCONNECTED actuly what you want do

closed 2025-02-25 by MaxwellAlan