espressif/esp-idf #14235: examples/mesh/ip_internal_network: misconfiguration of driver_ifconfig.driver_free_rx_buffer

vjgriswold, 2024-07-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In [mesh_netif.c](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/mesh/ip_internal_network/main/mesh_netif.c),
functions **mesh_driver_start_root_ap()** and **mesh_driver_start_node_sta()**, the callback function **driver_free_rx_buffer** is initialized to the function **mesh_free()** which, in turn, forwards all calls to the heap-based **free()**.

The actual ESP-WIFI-MESH **esp_netif** initialization configures the **driver_free_rx_buffer** callback to the function **wifi_free()** in [wifi_netif.c](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_wifi/src/wifi_netif.c).  This correct function forwards all calls to **esp_wifi_internal_free_rx_buffer()**.

It therefore appears highly likely that buffer-free operations may release Rx buffers onto the wrong free-list, corrupting the WiFi buffer-management facility.

zhangyanjiaoesp, 2024-08-12:
@vjgriswold Thanks for report, we will check it ASAP.

zhangyanjiaoesp, 2024-12-24:
@vjgriswold 
If the netif uses the standard wifi driver, when receiving a packet, the path is this:
```
sta_input(eb) 
-> esp_netif_receive(netif, buffer, len, eb) 
-> esp_netif->lwip_input_fn(netif, buffer, len, eb) 
-> wlanif_input(netif, buffer, len, l2_buff) 
-> esp_netif_free_rx_buffer(netif, l2_buff) 
-> wifi_free(l2_buff) 
-> esp_wifi_internal_free_rx_buffer(l2_buff)
```
the l2_buff is the eb used in the wifi driver, so free it in the wifi layer is correct.

If the netif uses the mesh link driver, when receiving a packet, the path is this: 
```
sta_input(eb) 
-> mesh_task, convert the wifi eb into a mesh-type packet and release the eb 
-> receive_task() 
-> esp_netif_receive(netif, data.data, data.len, NULL) 
->esp_netif->lwip_input_fn(netif, buffer, len, l2_buff) 
-> wlanif_input(netif, buffer, len, l2_buff) 
-> esp_netif_free_rx_buffer(netif, l2_buff) 
-> mesh_free(l2_buff)
-> free(l2_buff)
```
the l2_buff is NULL, so free it using heap-based free() is correct.

closed 2025-01-07 by Alvin1Zhang