espressif/esp-idf #14237: examples/mesh/ip_internal_network: Use of 'if_desc' to distinguish between esp_netif configurations

vjgriswold, 2024-07-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In [mesh_netif.c](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/mesh/ip_internal_network/main/mesh_netif.cl), the logic within several functions distinguishes between different configurations of **netif_sta** through comparison of its **if_desc** property.  This property can only be set, however, through the **esp_netif_config_t** structure at the time of calling **esp_netif_new()**, and is immutable afterwards.

If one is to create a valid driver control block for the Mesh netif, though, _one does not have access_ to **if_desc** through public APIs until after **esp_netif_new()** has already been called within the ESP-WIFI-MESH library routines.

Either the logic within **mesh_netif.c** must be updated to not distinguish **netif_sta** via **if_desc**, or the corresponding Mesh library routines must be publicly-accessible, as they are today with WiFi.

zhangyanjiaoesp, 2024-08-12:
@vjgriswold Thanks for report, we will check it ASAP.

zhangyanjiaoesp, 2024-08-23:
@vjgriswold I don't think there is an issue with the netif operation in the example.

vjgriswold, 2024-08-29:
The use of **if_desc** in this example is valid only if a correct device control block can be created at the same time.  Given that the layout of the Mesh device control block is opaque, that is not possible.  Thus, there is insufficient visibility into the prerequisite data structures to use **if_desc** for this purpose.

zhangyanjiaoesp, 2024-12-24:
@vjgriswold 
The `mesh_netif.c` is only used in the `ip_internal_network`, the underlying logic is still based on the WiFi netif, the example only modifies some driver functions.

closed 2025-01-07 by Alvin1Zhang