espressif/esp-idf #14238: examples/mesh/ip_internal_network: Failure to re-acquire DHCP address after Root re-associates to AP

vjgriswold, 2024-07-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.7

### Espressif SoC revision.

ESP32-WROVER-E

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

Mesh forms, Root is elected, Root acquires a DHCP address from the AP to which it is associated, and all nodes are able to send UDP data to that AP.

The AP is rebooted.

A new Root (perhaps the same as the old Root) is elected.  This new Root acquires a (potentially) new DHCP address from the AP which it is associated, and all nodes are again able to send UDP data to that AP.

### What is the actual behavior?

Mesh forms, Root is elected, Root acquires a DHCP address from the AP to which it is associated, and all nodes are able to send UDP data to that AP.

The AP is rebooted.

A new Root (perhaps the same as the old Root) is elected.  In what appears to roughly 80% of the cases, the Root fails to acquire a DHCP address from the AP to which it is associated.  No nodes are able to communicate, because the Root is unable to communicate.

### Steps to reproduce.

Allow the mesh to successfully form, and the Root to acquire a DHCP address from its AP.
Cycle the radio on the AP through "**ifconfig wlan0 down**", wait 15 seconds, "**ifconfig wlan0 up**".
Observe that the Root fails to acquire a DHCP address.

### More Information.

Power-cycling the Root device clears the problem.  Use of **esp_restart()** _does not remedy the issue_!
Packet traces have shown that the Root is issuing DHCP Requests and that the AP is responding with DHCP offers.
LWIP trace diagnostics also show that DHCP Requests are being issued, but LWIP traces do not show the DHCP offers being received by LWIP.

zhangyanjiaoesp, 2024-08-12:
@vjgriswold Thanks for report, we will check it ASAP.

zhangyanjiaoesp, 2024-08-23:
@vjgriswold 
I have tested locally, but I didn't reproduce your issue. Here is my log on the root node:
```
I (127236) wifi:bcn_timeout,ap_probe_send_start
I (128956) mesh_main: Tried to publish layer:1 IP:192.168.4.2
I (128956) mesh_mqtt: sent publish returned msg_id=31148
I (128956) mesh_main: Sending routing table to [0] 64:b7:08:29:2b:a4: sent with err code: 0
I (128956) mesh_main: Received Routing table [0] 64:b7:08:29:2b:a4
I (128966) mesh_main: Sending routing table to [1] 64:b7:08:29:31:f4: sent with err code: 0
I (128966) mesh_main: Received Routing table [1] 64:b7:08:29:31:f4
I (129736) wifi:ap_probe_send over, resett wifi status to disassoc
I (129736) wifi:state: run -> init (0xc800)
I (129746) wifi:pm stop, total sleep time: 0 us / 119049802 us

I (129746) wifi:<ba-del>idx:0, tid:0
I (129746) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (129756) mesh: [wifi]disconnected reason:200(beacon timeout), continuous:1/max:12, root, vote(,stopped)<><>
I (129756) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:200
I (130866) mesh_mqtt: Other event id:7
I (130976) mesh_main: Tried to publish layer:1 IP:192.168.4.2
I (132616) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, root, vote(,stopped)<><>
I (132616) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (135456) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, root, vote(,stopped)<><>
I (135456) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (135476) wifi:primary chan differ, old=1, new=1, start CSA timer
I (135476) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (135476) wifi:state: init -> auth (0xb0)
I (137246) wifi:state: auth -> assoc (0x0)
I (137256) wifi:state: assoc -> run (0x10)
I (137286) wifi:connected with ttt, aid = 1, channel 1, 40U, bssid = 24:0a:c4:01:4e:05
I (137286) wifi:security: WPA3-SAE, phy: bgn, rssi: -25
I (137286) wifi:pm start, type: 0

I (137286) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (137306) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
I (137306) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:24:0a:c4:01:4e:05<ROOT>, ID:77:77:77:77:77:76
I (137306) mesh: <flush>root
I (137316) mesh_netif: Already wifi station, no need to do anything
I (137316) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
I (137326) mesh_main: <MESH_EVENT_TODS_REACHABLE>state:0
I (137336) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
I (137346) mesh_main: <MESH_EVENT_ROOT_ADDRESS>root address:64:b7:08:29:2b:a5
I (137366) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (137366) wifi:AP's beacon interval = 102400 us, DTIM period = 2
E (137866) esp-tls: couldn't get hostname for :mqtt.eclipseprojects.io: getaddrinfo() returns 202, addrinfo=0x0
E (137866) transport_base: Failed to open a new connection: 32769
E (137866) mqtt_client: Error transport connect
I (137876) mesh_mqtt: MQTT_EVENT_ERROR
I (137876) mesh_mqtt: MQTT_EVENT_DISCONNECTED
I (137886) mesh_mqtt: sent publish returned msg_id=63587
I (137886) mesh_main: Sending routing table to [0] 64:b7:08:29:2b:a4: sent with err code: 0
I (137886) mesh_main: Received Routing table [0] 64:b7:08:29:2b:a4
I (137896) mesh_main: Sending routing table to [1] 64:b7:08:29:31:f4: sent with err code: 0
I (137906) mesh_main: Received Routing table [1] 64:b7:08:29:31:f4
I (138306) mesh_main: <IP_EVENT_STA_GOT_IP>IP:192.168.4.2
I (138306) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 10.0.0.1
I (138306) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1

```

Can you provide a log from your side?

zhangyanjiaoesp, 2024-12-24:
@vjgriswold Do you still have this issue? If not, can we close it?

closed 2025-01-07 by Alvin1Zhang