espressif/esp-idf #14239: ESP32 (IDF v5.1) SSL Handshake Failure After Server Certificate Renewal

samuelclay, 2024-07-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I'm encountering an SSL handshake failure with my ESP32 devices (using ESP-IDF v5.1 with mbedTLS 3.3.0) when connecting to my server. This issue started occurring after the server's Let's Encrypt SSL certificate was renewed. When I upgrade to ESP-IDF v5.1.4, which uses mbedTLS 3.5.2, everything works as expected. But the ESP-IDF v5.1 devices are deployed in the field, so I can't easily update their firmware. 

### What is the actual behavior?

The ESP32 devices were able to connect successfully until a recent renewal of the SSL certificate on the server. The renewed certificate now causes the ESP32 devices to fail the connection with the following error:

```
E (80654) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (80660) esp-tls-mbedtls: (FFFFD900): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
E (80671) esp-tls: Failed to open new connection
E (80677) transport_base: Failed to open a new connection
E (80688) HTTP_CLIENT: Connection failed, sock < 0
E (80688) Sol: HTTP_EVENT_ERROR
```

The SSL certificates have an A rating on [Qualys SSL Server Test](https://www.ssllabs.com/ssltest/analyze.html?d=prod5.sindarin.com&hideResults=on) and work fine on web browsers and IDF v5.1.4, just not on IDF v5.1. The server is serving a TLS 1.2 certificate with many ciphers, most of which IDF 5.1 should be able to handle. 

I've confirmed that the ESP32 devices are successfully connecting to WiFi. And I verified that upgrading to a later version of mbedTLS on a test device resolves the issue (not feasible for deployed devices). I also reviewed Caddy configuration and SSL settings and switched to nginx with similar configurations, just to rule out the web server. Both Caddy and nginx provide A ratings for the SSL certificate, and work in a browser and on IDF 5.1.4.

### Steps to reproduce.

1. Set up an ESP32 device with ESP-IDF v5.1 and mbedTLS 3.3.0.
2. Configure the device to connect to a server using HTTPS.
3. Set up the server with Caddy as a reverse proxy and obtain an SSL certificate from Let's Encrypt.
4. Attempt to connect to the server from the ESP32 device.

### Debug Logs.

```plain
ESP32 Log:

I (16171) Sol: Connected, IP address: 192.168.1.70
D (16183) Sol: Free heap: 56k/290k Free PSRAM: 7971k/8668k 
D (16195) esp-tls: handshake in progress...
[... 206 lines cut ...]
E (18184) Sol: HTTP_EVENT_ERROR
D (18188) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:0 posted to loop 0x3fcddb28
E (18199) Sol: esp_http_client_open err=ESP_ERR_HTTP_CONNECT
```

Server Log (Caddy):
```
2024/07/23 13:53:58.495	DEBUG	events	event	{"name": "tls_get_certificate", "id": "4c4ca4fc-5227-468e-809d-19ae48f54687", "origin": "tls", "data": {"client_hello":{"CipherSuites":[49325,49162,49172,49327,49195,49199,49324,49187,49191,49161,49171,49326,49244,49248,49224,49228,49309,61,53,49167,49157,49313,156,49308,60,47,49201,49193,49166,49197,49189,49156,49312,49246,49250,49232,49226,49230,49212,255],"ServerName":"prod5.sindarin.com","SupportedCurves":[29,23,24,25,26,27,28],"SupportedPoints":"AA==","SignatureSchemes":[1027,1025],"SupportedProtos":null,"SupportedVersions":[771,770,769],"RemoteAddr":{"IP":"130.51.142.53","Port":62167,"Zone":""},"LocalAddr":{"IP":"64.225.15.171","Port":443,"Zone":""}}}}
2024/07/23 13:53:58.495	DEBUG	tls.handshake	choosing certificate	{"identifier": "prod5.sindarin.com", "num_choices": 1}
2024/07/23 13:53:58.495	DEBUG	tls.handshake	default certificate selection results	{"identifier": "prod5.sindarin.com", "subjects": ["prod5.sindarin.com"], "managed": true, "issuer_key": "acme-v02.api.letsencrypt.org-directory", "hash": "eb22767b8203303d93f0bc1727b42a77200fec476d4d8690ceb9c4897f4832d3"}
2024/07/23 13:53:58.495	DEBUG	tls.handshake	matched certificate in cache	{"remote_ip": "130.51.142.53", "remote_port": "62167", "subjects": ["prod5.sindarin.com"], "managed": true, "expiration": "2024/10/21 01:37:02.000", "hash": "eb22767b8203303d93f0bc1727b42a77200fec476d4d8690ceb9c4897f4832d3"}
2024/07/23 13:53:59.184	DEBUG	http.stdlib	http: TLS handshake error from 130.51.142.53:62167: remote error: tls: bad certificate
```
```

### More Information.

I switched from Caddy to Nginx and configured the server to use both the "ISRG Root X1" and "ISRG Root X2" certificates. The X1 is the current valid root certificate for Let's Encrypt, but the X2 should also work. 

1. Has anyone encountered similar issues with ESP32 and SSL certificate renewals?
2. Are there known compatibility issues between mbedTLS 3.3.0 and recent Let's Encrypt certificates?
3. What server-side changes can I make to resolve this issue without updating the ESP32 firmware?
4. Are there any specific ESP-IDF or mbedTLS configurations that might help mitigate this problem?

Any insights or suggestions would be greatly appreciated. Thank you in advance for your help!

KaeLL, 2024-07-23:
I had this issue too back in March. Ended up switching the esp crt bundle from minimal to full. The emergency update server bring-up was done by a coworker, and crisis was averted.

samuelclay, 2024-07-23:
We're already using the full default certificate bundle, but what was the emergency update server bring-up? Does that mean you added certificates to the chain on your server? Which ones and how did you know which ones?

KaeLL, 2024-07-23:
@windol

samuelclay, 2024-07-25:
Solved the problem. I'd like to pay it forward here to anybody else who may come across this issue in the future. Here's what I did.

I built a minimal test case that (included below):
1. Connected to wifi
2. Fetched an https endpoint and return the status code
3. Scanning for wifi networks if wifi failed

That allowed me to narrow down the issue. At first it worked. Then I imported my project's `sdkconfig.defaults` and adjusted a few of the configuration options so that it compiled (mainly about partition maps needing to change to address the larger flash size). And then I was able to reproduce the issue.

There were about 250 differences between the two `sdkconfig` settings. I used a binary search to narrow it down to a single setting: [CONFIG_MBEDTLS_SHA512_C](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-mbedtls-sha512-c): `Enable the SHA-384 and SHA-512 cryptographic hash algorithms`

We had that turned off in our shipping code, while it's turned on by default. Switching it on got everything working again. Obviously I can't switch it on because these devices are out in the field and I can't change their code. But that led me down the path of figuring out what changes were needed in the SSL certificate.

I learned that our certificate was using a signature algorithm of `ECDSA SHA-384`. By creating a RSA 2048-bit certificate signing request (CSR) and using that CSR when applying for a SSL certificate, I could generate a certification that used `RSA SHA-256`, which would be supported by my devices using `# CONFIG_MBEDTLS_SHA512_C is not set`.

closed 2024-07-25 by samuelclay

samuelclay, 2024-07-25:
Here's `esp32_https_test.c`, my minimal test case:

```
#include "esp_crt_bundle.h"
#include "esp_event.h"
#include "esp_http_client.h"
#include "esp_log.h"
#include "esp_netif.h"
#include "esp_system.h"
#include "esp_tls.h"
#include "esp_wifi.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include <stdio.h>

static const char *TAG = "HTTPS_TEST";
#define WIFI_POWER_19_5dBm 78
#define DEFAULT_SCAN_LIST_SIZE 10

#define WIFI_SSID "Hits of Sunshine"
#define WIFI_PASS "brittanyjanis"

esp_err_t _http_event_handler(esp_http_client_event_t *evt) {
  switch (evt->event_id) {
  case HTTP_EVENT_ERROR:
    ESP_LOGI(TAG, "HTTP_EVENT_ERROR");
    break;
  case HTTP_EVENT_ON_CONNECTED:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_CONNECTED");
    break;
  case HTTP_EVENT_HEADER_SENT:
    ESP_LOGI(TAG, "HTTP_EVENT_HEADER_SENT");
    break;
  case HTTP_EVENT_ON_HEADER:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key,
             evt->header_value);
    break;
  case HTTP_EVENT_ON_DATA:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
    if (!esp_http_client_is_chunked_response(evt->client)) {
      printf("%.*s", evt->data_len, (char *)evt->data);
    }
    break;
  case HTTP_EVENT_ON_FINISH:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_FINISH");
    break;
  case HTTP_EVENT_DISCONNECTED:
    ESP_LOGI(TAG, "HTTP_EVENT_DISCONNECTED");
    break;
  default:
    ESP_LOGI(TAG, "HTTP_EVENT_UNKNOWN, event=%d", evt->event_id);
    break;
  }
  return ESP_OK;
}

void create_client_and_test_website() {
  esp_http_client_config_t config = {
      .url = "https://prod5.sindarin.com",
      .event_handler = _http_event_handler,
      .method = HTTP_METHOD_GET,
      .timeout_ms = 30000,
      .cert_pem = NULL,
      .crt_bundle_attach = esp_crt_bundle_attach,
  };

  esp_http_client_handle_t client = esp_http_client_init(&config);

  // Perform the HTTP GET request
[... 156 lines cut ...]
  ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G |
                                         WIFI_PROTOCOL_11N);

  ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
  vTaskDelay(1000 / portTICK_PERIOD_MS);
  ESP_ERROR_CHECK(esp_wifi_start());
  vTaskDelay(1000 / portTICK_PERIOD_MS);
  ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
}

void app_main(void) {
  esp_log_level_set("wifi", ESP_LOG_VERBOSE);
  esp_log_level_set("wifi_init", ESP_LOG_VERBOSE);

  esp_err_t ret = nvs_flash_init();
  if (ret == ESP_ERR_NVS_NO_FREE_PAGES ||
      ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
    ESP_ERROR_CHECK(nvs_flash_erase());
    ret = nvs_flash_init();
  }
  ESP_ERROR_CHECK(ret);
  wifi_init();
}
```

samuelclay, 2024-07-25:
Here's `esp32_https_test.c`, my minimal test case:

```
#include "esp_crt_bundle.h"
#include "esp_event.h"
#include "esp_http_client.h"
#include "esp_log.h"
#include "esp_netif.h"
#include "esp_system.h"
#include "esp_tls.h"
#include "esp_wifi.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include <stdio.h>

static const char *TAG = "HTTPS_TEST";
#define WIFI_POWER_19_5dBm 78
#define DEFAULT_SCAN_LIST_SIZE 10

#define WIFI_SSID "<snip>"
#define WIFI_PASS "<snip>"

esp_err_t _http_event_handler(esp_http_client_event_t *evt) {
  switch (evt->event_id) {
  case HTTP_EVENT_ERROR:
    ESP_LOGI(TAG, "HTTP_EVENT_ERROR");
    break;
  case HTTP_EVENT_ON_CONNECTED:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_CONNECTED");
    break;
  case HTTP_EVENT_HEADER_SENT:
    ESP_LOGI(TAG, "HTTP_EVENT_HEADER_SENT");
    break;
  case HTTP_EVENT_ON_HEADER:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key,
             evt->header_value);
    break;
  case HTTP_EVENT_ON_DATA:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
    if (!esp_http_client_is_chunked_response(evt->client)) {
      printf("%.*s", evt->data_len, (char *)evt->data);
    }
    break;
  case HTTP_EVENT_ON_FINISH:
    ESP_LOGI(TAG, "HTTP_EVENT_ON_FINISH");
    break;
  case HTTP_EVENT_DISCONNECTED:
    ESP_LOGI(TAG, "HTTP_EVENT_DISCONNECTED");
    break;
  default:
    ESP_LOGI(TAG, "HTTP_EVENT_UNKNOWN, event=%d", evt->event_id);
    break;
  }
  return ESP_OK;
}

void create_client_and_test_website() {
  esp_http_client_config_t config = {
      .url = "https://example.com",
      .event_handler = _http_event_handler,
      .method = HTTP_METHOD_GET,
      .timeout_ms = 30000,
      .cert_pem = NULL,
      .crt_bundle_attach = esp_crt_bundle_attach,
  };

  esp_http_client_handle_t client = esp_http_client_init(&config);

  // Perform the HTTP GET request
[... 156 lines cut ...]
  ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G |
                                         WIFI_PROTOCOL_11N);

  ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
  vTaskDelay(1000 / portTICK_PERIOD_MS);
  ESP_ERROR_CHECK(esp_wifi_start());
  vTaskDelay(1000 / portTICK_PERIOD_MS);
  ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
}

void app_main(void) {
  esp_log_level_set("wifi", ESP_LOG_VERBOSE);
  esp_log_level_set("wifi_init", ESP_LOG_VERBOSE);

  esp_err_t ret = nvs_flash_init();
  if (ret == ESP_ERR_NVS_NO_FREE_PAGES ||
      ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
    ESP_ERROR_CHECK(nvs_flash_erase());
    ret = nvs_flash_init();
  }
  ESP_ERROR_CHECK(ret);
  wifi_init();
}
```