espressif/esp-idf #14241: Guru Meditation Error at  usb_dc_check_poll_for_interrupts() on esp32-S2

climateguard, 2024-07-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**IDF version.**
v4.4.4
**ESP version**
ESP32-S2

When use USB CDC as chanel for console output as i found. At random time.

`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40014a08  PS      : 0x00060d31  A0      : 0x80014ca8  A1      : 0x3ffca060  
A2      : 0xb6df0084  A3      : 0x00000000  A4      : 0x3ffffbac  A5      : 0x3ffffba4  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x3ffe791c  A9      : 0x000000fe  
A10     : 0x3ffffa74  A11     : 0x60080b10  A12     : 0x60080040  A13     : 0x60080000
A14     : 0x00000000  A15     : 0x3ffc9970  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0xb6df0086  LBEG    : 0x60080040  LEND    : 0x60080000  LCOUNT  : 0x40026a34  
0x40026a34: _xt_user_exc at C:/Users/subki/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:633

Core  0 was running in ISR context:
EPC1    : 0x400847af  EPC2    : 0x401592be  EPC3    : 0x00000000  EPC4    : 0x00000000
0x400847af: panic_print_str at C:/Users/subki/esp/esp-idf/components/esp_system/panic.c:132

0x401592be: esp_pm_impl_waiti at C:/Users/subki/esp/esp-idf/components/esp_pm/pm_impl.c:832

Backtrace: 0x40014a05:0x3ffca060 0x40014ca5:0x3ffca080 0x40014e3e:0x3ffca0d0 0x40014ea9:0x3ffca0f0 0x40013e45:0x3ffca120 0x40014989:0x3ffca140 0x40084b3e:0x3ffca160 0x40026b59:0x3ffca180 0x401592bb:0x3ffd8b10 0x400af7b6:0x3ffd8b30 0x4002e9b1:0x3ffd8b50 0x40030639:0x3ffd8b70
0x40084b3e: esp_usb_console_interrupt at C:/Users/subki/esp/esp-idf/components/esp_system/port/soc/esp32s2/usb_console.c:159

0x40026b59: _xt_lowint1 at C:/Users/subki/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x401592bb: cpu_ll_waiti at C:/Users/subki/esp/esp-idf/components/hal/esp32s2/include/hal/cpu_ll.h:202
 (inlined by) esp_pm_impl_waiti at C:/Users/subki/esp/esp-idf/components/esp_pm/pm_impl.c:830

0x400af7b6: esp_vApplicationIdleHook at C:/Users/subki/esp/esp-idf/components/esp_system/freertos_hooks.c:63

0x4002e9b1: prvIdleTask at C:/Users/subki/esp/esp-idf/components/freertos/tasks.c:3987 (discriminator 1)

0x40030639: vPortTaskWrapper at C:/Users/subki/esp/esp-idf/components/freertos/port/xtensa/port.c:142

ELF file SHA256: 631683cf67f343f6`

peter-marcisovsky, 2024-07-24:
Hi @climateguard could you give us some information about your workaround?

- When does this error occur?
- Are you using any of our USB examples?
- Do you have any USB device connected to your esp32s2 USB Host?
  - If yes, what kind of USB device is it?
- Which devkit are you using?
- How are you powering your board? Over USB does it have an external power source?
- Is there any way of reproducing the issue on our side?

Thank you.

climateguard, 2024-07-24:
Hi @peter-marcisovsky  I use only Uart 0 for sim7600 and Uart1 for sensors. No usb devices, usb only for logging because all uarts   are occupated. I use ESP32-S2-WROVER-N4R2 smd but with no spi ram active. Board are powered with external source and no 5v power on usb (only ground and data lines used). I did not found the way to reproduce. I faced issue in random period of board working but it does not exceed 90 minutes. If disable usb logging, everything works fine. For logs i use ESP_LOG and printf.

peter-marcisovsky, 2024-07-25:
We have noticed, you are using `IDF v4.4.4`, please note, that the `IDF v4.4` is nearly at [EOL](https://github.com/espressif/esp-idf?tab=readme-ov-file#esp-idf-release-support-schedule). Which means, that there will be no minor release of `IDF v4.4.x`, no bug fixes from our side for the `IDF v4.4.x`.

 - As a first thing, could you try updating your IDF to `IDF v4.4.8`, which was the final release for the `IDF v4.4`.
 - If that does not fix your issue, I would recommend you to updating your IDF to next stable release, which would be `IDF v5.x`.  
 - Updating from `IDF v4.x` to `IDF v5.x` is a breaking change, please check out our [migration guides](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/index.html)

climateguard, 2024-07-26:
at idf 4.4.8 same error but with additional info:
`Guru Meditation Error: Core  0 panic'ed (Memory protection fault). 
Write operation at address 0x800842a4 not permitted (INVALID_MEM_TYPE)

Core  0 register dump:
PC      : 0x40014a0b  PS      : 0x00060d34  A0      : 0x80014ca8  A1      : 0x3ffca060
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffffbac  A5      : 0x3ffffba4  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x3ffe791c  A9      : 0x000000fe  
A10     : 0x3ffffa74  A11     : 0x60080b10  A12     : 0x60080040  A13     : 0x60080000  
A14     : 0x00000000  A15     : 0x3ffc9970  SAR     : 0x0000001c  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x60080040  LEND    : 0x60080000  LCOUNT  : 0x40025461
0x40025461: xt_highint4 at C:/Users/subki/esp/esp-idf/components/esp_system/port/soc/esp32s2/highint_hdl.S:58

Core  0 was running in ISR context:
EPC1    : 0x400847b7  EPC2    : 0x4015952a  EPC3    : 0x00000000  EPC4    : 0x40014a0b
0x400847b7: panic_print_str at C:/Users/subki/esp/esp-idf/components/esp_system/panic.c:132

0x4015952a: esp_pm_impl_waiti at C:/Users/subki/esp/esp-idf/components/esp_pm/pm_impl.c:832

Backtrace: 0x40014a08:0x3ffca060 0x40014ca5:0x3ffca080 0x40014e3e:0x3ffca0d0 0x40014ea9:0x3ffca0f0 0x40013e45:0x3ffca120 0x40014989:0x3ffca140 0x40084b46:0x3ffca160 0x40026b59:0x3ffca180 0x40159527:0x3ffd8c30 0x400afa0a:0x3ffd8c50 0x4002e9b1:0x3ffd8c70 0x40030639:0x3ffd8c90
0x40084b46: esp_usb_console_interrupt at C:/Users/subki/esp/esp-idf/components/esp_system/port/soc/esp32s2/usb_console.c:159

0x40026b59: _xt_lowint1 at C:/Users/subki/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x40159527: cpu_ll_waiti at C:/Users/subki/esp/esp-idf/components/hal/esp32s2/include/hal/cpu_ll.h:202
 (inlined by) esp_pm_impl_waiti at C:/Users/subki/esp/esp-idf/components/esp_pm/pm_impl.c:830

0x400afa0a: esp_vApplicationIdleHook at C:/Users/subki/esp/esp-idf/components/esp_system/freertos_hooks.c:63

0x4002e9b1: prvIdleTask at C:/Users/subki/esp/esp-idf/components/freertos/tasks.c:3987 (discriminator 1)

0x40030639: vPortTaskWrapper at C:/Users/subki/esp/esp-idf/components/freertos/port/xtensa/port.c:142`

peter-marcisovsky, 2024-07-29:
Thank you for letting it know. Then, please follow the further instructions I have described in the previous message.

climateguard, 2024-08-16:
hello, have update of issue. Made migration to final 5 release and faces same issue
`Guru Meditation Error: Core  0 panic'ed (Memory protection fault). 
Write operation at address 0x80084308 not permitted (INVALID_MEM_TYPE)

Core  0 register dump:
PC      : 0x40014a08  PS      : 0x00060d34  A0      : 0x80014ca8  A1      : 0x3ffca500  
A2      : 0x4241aef5  A3      : 0x00000000  A4      : 0x3ffffbac  A5      : 0x3ffffba4  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x3ffe795c  A9      : 0x000000fe  
A10     : 0x3ffffa74  A11     : 0x60080b10  A12     : 0x60080040  A13     : 0x60080000  
A14     : 0x00000000  A15     : 0x3ffc9e08  SAR     : 0x0000001c  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x60080040  LEND    : 0x60080000  LCOUNT  : 0x4002548d
0x4002548d: xt_highint4 at C:/Users/sende/esp/v5/esp-idf/components/esp_system/port/soc/esp32s2/highint_hdl.S:58

Core  0 was running in ISR context:
EPC1    : 0x4008481b  EPC2    : 0x4015aa6a  EPC3    : 0x00000000  EPC4    : 0x40014a08
0x4008481b: panic_print_str at C:/Users/sende/esp/v5/esp-idf/components/esp_system/panic.c:132

0x4015aa6a: esp_pm_impl_waiti at C:/Users/sende/esp/v5/esp-idf/components/esp_pm/pm_impl.c:855

Backtrace: 0x40014a05:0x3ffca500 0x40014ca5:0x3ffca520 0x40014e3e:0x3ffca570 0x40014ea9:0x3ffca590 0x40013e45:0x3ffca5c0 0x40014989:0x3ffca5e0 0x40084baa:0x3ffca600 0x40026b8d:0x3ffca620 0x4015aa67:0x3ffd9480 0x400afe06:0x3ffd94a0 0x4002ecc9:0x3ffd94c0 0x40030941:0x3ffd94e0
0x40084baa: esp_usb_console_interrupt at C:/Users/sende/esp/v5/esp-idf/components/esp_system/port/soc/esp32s2/usb_console.c:159

0x40026b8d: _xt_lowint1 at C:/Users/sende/esp/v5/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1118

0x4015aa67: cpu_ll_waiti at C:/Users/sende/esp/v5/esp-idf/components/hal/esp32s2/include/hal/cpu_ll.h:202
 (inlined by) esp_pm_impl_waiti at C:/Users/sende/esp/v5/esp-idf/components/esp_pm/pm_impl.c:853

0x400afe06: esp_vApplicationIdleHook at C:/Users/sende/esp/v5/esp-idf/components/esp_system/freertos_hooks.c:63

0x4002ecc9: prvIdleTask at C:/Users/sende/esp/v5/esp-idf/components/freertos/tasks.c:4099 (discriminator 1)

0x40030941: vPortTaskWrapper at C:/Users/sende/esp/v5/esp-idf/components/freertos/port/xtensa/port.c:142`

climateguard, 2024-08-16:
esp_usb_console_interrupt uses for esp dfu mode or not?

climateguard, 2024-08-16:
if i switch off dfu logs in my realize  firmware, can it solve the problem?

peter-marcisovsky, 2024-08-27:
Hi @climateguard sorry for the delay. 
Could you please specify some steps to reproduce the issue, so we can test it on our side?
Which IDF version are you using now?
It would be very helpful for us. Thank you.

peter-marcisovsky, 2025-03-19:
Hi @climateguard any update on this issue?

peter-marcisovsky, 2025-03-27:
Closing the issue due inactivity. If your problem persists, please open a new issue.

closed 2025-03-27 by peter-marcisovsky