espressif/esp-idf #14244: USB HID Host - Some USB devices not detected on boot, working when attached afterwards

mrweaver, 2024-07-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

Run [`hid_host_example.c`](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c) and have the USB HID remote (G20s Pro) automatically detected and functional on boot.

### What is the actual behavior?

## Scenario 1 - Logitech Mouse [Success]
When a Logitech wireless mouse (MX Anywhere 2) is plugged into the USB port, then the example is run, the device is correctly detected immediately and inputs from the device are received as expected. 

## Scenario 2 - USB Remote [Failure]
When the same is performed as above, expect instead of the Logitech device, the G20s Pro remote ( Chinese air mouse), the device will not be detected on the initial boot. If the device is unplugged, then plugged back in, the device is correctly detected and functions as expected (see logs)

Not sure if this is expected for the device in question based on the descriptors, or if this is some unwanted behaviour. I have tried adding various delays in the code, but none seem to have any affect. 
@roma-jam - I was wondering if you'd have any input on this.

### Steps to reproduce.

1. Insert USB device 
    a. - Logitech
    b. Scenario 2 - USB Remote
2. Using VSCode, run unaltered example - [`hid_host_example.c`](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c)
3. Observe program outputs
    a.  Scenario 1 - Works
    b. Scenario 2 - Fails

### Debug Logs.

```plain
## Scenario 1 - Logitech Mouse [Success]
### `DEBUG` level output

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
[... 425 lines cut ...]
                wMaxPacketSize 16
                bInterval 10
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 3
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x3
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x2
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 10
I (21485) CLASS: Getting Manufacturer string descriptor
SG.Ltd
I (21485) CLASS: Getting Product string descriptor
SG Control Mic
```
```

peter-marcisovsky, 2024-07-25:
Hi @mrweaver thank you for reporting the issue and for providing all the logs. We will have a look and let you know.

roma-jam, 2024-07-29:
Hi @mrweaver ,

Thanks for the detailed logs.

>  I have tried adding various delays in the code

There are three configurable parameters available with the menuconfig. More details [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/usb_host.html#non-compliant-device-support): 

- `DEBOUNCE_DELAY`
- `RESET_HOLD_DELAY`
- `RESET_RECOVERY_DELAY` 

Have you tried to change them? Could you increase the `RESET_HOLD` and `RESET_RECOVERY` for, lets say, 500 ms (default values are 30 ms) and try to connect the device again? 

The problem should be the same with `hid` example and `usb_host_lib`, so you can use the last one to eliminate the hid logic from verification to simplify the output.  

Do you have any option to use USB analyzer to capture the transfer log with the device you are trying to use?

mrweaver, 2024-07-30:
Hey @roma-jam,

Thanks for getting back to me. I did see those parameters, but for some reason I (rather foolishly, in hindsight) assumed that the device should follow the standard USB timings and so I didn't change them.

Anyway, I recompiled it with some different values and found that for the case of this particular device, configuring the parameter `RESET_RECOVERY` to 80ms+ (while leaving the others as default) seems to do the trick and it resolves the issue - that is the device is discovered on boot.

Thanks so much for your help, I really appreciate it.

Cheers,
Michael.

closed 2024-07-30 by mrweaver