espressif/esp-idf #14246: Build problem on macOS M2 Pro, cffi architecture problem (have 'arm64', need 'x86_64'))

MacWyznawca, 2024-07-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3 release

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build project like on IDF 5.2.2. With ESP-IDF 5.2.2 all works great.

### What is the actual behavior?

Error when build: /_cffi_backend.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))

### Steps to reproduce.

Instal celan ESP-IDF 5.3 on macOS aarch64
Build project with signed app and bootloader

### Build or installation Logs.

```plain
% idf.py @profiles/dev build  
Running: idf.py -B /Users/jkk/LuON/LuONbaseDev -DSDKCONFIG_DEFAULTS=sdkconfig.dev build
Executing action: all (aliases: build)
Running ninja in directory /Users/jkk/LuON/LuONbaseDev
Executing "ninja all"...
[1/549] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle /Users/jkk/LuON/LuONbaseDev/esp-idf/mbedtls/x509_crt_bundle 
cd /Users/jkk/LuON/LuONbaseDev/esp-idf/mbedtls && /Users/jkk/.espressif/python_env/idf5.3_py3.9_env/bin/python /Users/jkk/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /Users/jkk/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /Users/jkk/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q
The cryptography package is not installed.Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Traceback (most recent call last):
  File "/Users/jkk/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 25, in <module>
    from cryptography import x509
  File "/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/cryptography/x509/__init__.py", line 7, in <module>
    from cryptography.x509 import certificate_transparency, verification
  File "/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/cryptography/x509/certificate_transparency.py", line 11, in <module>
    from cryptography.hazmat.bindings._rust import x509 as rust_x509
ImportError: dlopen(/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so, 0x0002): tried: '/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (no such file), '/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
[2/549] Performing build step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map /Users/jkk/LuON/LuONbaseDev/bootloader-prefix/src/bootloader-stamp/bootloader-build /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.elf /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.bin /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.map 
cd /Users/jkk/LuON/LuONbaseDev/bootloader && /usr/local/Cellar/cmake/3.22.2/bin/cmake --build .
[1/2] cd /Users/jkk/LuON/LuONbaseDev/bootloader/esp-idf/esptool_py && /Users/jkk/.espressif/py..._sizes.py --offset 0x9000 bootloader 0x0 /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.bin
Bootloader binary size 0x5920 bytes. 0x36e0 bytes (38%) free.
[2/2] Generated the signed Bootloader
FAILED: .signed_bin_timestamp /Users/jkk/LuON/LuONbaseDev/bootloader/.signed_bin_timestamp 
cd /Users/jkk/LuON/LuONbaseDev/bootloader && /usr/local/Cellar/cmake/3.22.2/bin/cmake -E copy /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.bin /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader-unsigned.bin && /Users/jkk/.espressif/python_env/idf5.3_py3.9_env/bin/python /Users/jkk/esp/esp-idf/components/esptool_py/esptool/espsecure.py sign_data --version 2 --keyfile /Volumes/LuOff/rsa_key/signing_key_1.pem -o /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.bin /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader-unsigned.bin && /usr/local/Cellar/cmake/3.22.2/bin/cmake -E echo "Generated signed binary image /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.bin" "from /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader-unsigned.bin" && /usr/local/Cellar/cmake/3.22.2/bin/cmake -E md5sum /Users/jkk/LuON/LuONbaseDev/bootloader/bootloader.bin > /Users/jkk/LuON/LuONbaseDev/bootloader/.signed_bin_timestamp
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py", line 147, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/espsecure/__init__.py", line 16, in <module>
    from cryptography import exceptions
  File "/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/cryptography/exceptions.py", line 9, in <module>
    from cryptography.hazmat.bindings._rust import exceptions as rust_exceptions
ImportError: dlopen(/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so, 0x0002): tried: '/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (no such file), '/Users/jkk/.espressif/python_env/idf5.3_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/jkk/LuON/LuONbaseDev/log/idf_py_stderr_output_92031 and /Users/jkk/LuON/LuONbaseDev/log/idf_py_stdout_output_92031
```

### More Information.

If I back to 5.2.2, all works great

dobairoland, 2024-07-25:
Please try the following: https://github.com/espressif/esp-idf/issues/12565#issuecomment-1807672658

Did it help?

MacWyznawca, 2024-07-25:
Thanks @dobairoland!

closed 2024-07-25 by MacWyznawca