espressif/esp-idf #14248: Add support for attributes in VFS

josesimoes, 2024-07-25:
### Is your feature request related to a problem?

Currently virtual file system doesn't provide support for attributes, which is something rather "basic" in a file system. 

### Describe the solution you'd like.

This doesn't need to be anything fancy, simple ones like hidden, read-only, system would be quite enough.
All file systems that VFS tapps into offer this: FatFS, littlefs, SPIFFS.

### Additional context.

Stumbled into this when making the migration from SPIFFs to littlefs for the flash storage in .NET nanoFramework nanoframework/nf-interpreter#2911.

Already discussed it with the author of littlefs for ESP extension. Please see issue here joltwallet/esp_littlefs#196.
It seems that there is path forward, still I believe it would be nice to have a couple of API to get and set attributes.

igrr, 2024-07-25:
@josesimoes what would be the equivalent API you would use on a POSIX platform for this purpose?

josesimoes, 2024-07-26:
I suppose the more suitable one would be `chmod`, right?
Still, hidden files would be missing because the current implementation lacks support for `.` started filenames, as stated in the docs.

I would suggest a more generic approach, such as providing `setattr` and `getattr` that accepts (or returns) an uint32 and then it's up to the file system to store and retrieve that number. It's up to the application layer to use as it sees fit.

josesimoes, 2024-12-20:
@haberturdeur here they are:

littlefs has [lfs_getattr](https://github.com/littlefs-project/littlefs/blob/630a0d87c23900c05ed5906db461ab91d3cc2c27/lfs.h#L540)

FatFS has [chmod](http://elm-chan.org/fsw/ff/doc/chmod.html)