espressif/esp-idf #14250: C++ build problem (already fixed June 21 but not included in 5.3)

dannybackx, 2024-07-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

No compile errors

### What is the actual behavior?

```
In file included from /home/danny/src/github/esp32/esp-idf-v5.3/components/esp_lcd/include/esp_lcd_panel_io.h:12,
                 from /home/danny/src/esp-idf/ZX3D95CD01S/minimal-gc9503/managed_components/espressif__esp_lcd_touch/include/esp_lcd_touch.h:18,
                 from /home/danny/src/esp-idf/ZX3D95CD01S/minimal-gc9503/managed_components/espressif__esp_lcd_touch_ft5x06/include/esp_lcd_touch_ft5x06.h:14,
                 from /home/danny/src/esp-idf/ZX3D95CD01S/minimal-gc9503/main/ft5x06.cpp:14:
/home/danny/src/github/esp32/esp-idf-v5.3/components/esp_lcd/include/esp_lcd_io_i2c.h:82:37: error: found ':' in nested-name-specifier, expected '::'
   82 |             i2c_master_bus_handle_t : esp_lcd_new_panel_io_i2c_v2, \
      |                                     ^
/home/danny/src/esp-idf/ZX3D95CD01S/minimal-gc9503/main/ft5x06.cpp:100:3: note: in expansion of macro 'esp_lcd_new_panel_io_i2c'
  100 |   esp_lcd_new_panel_io_i2c((esp_lcd_i2c_bus_handle_t)I2C_NUM_0, &io_config, &tp_io_handle);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/danny/src/github/esp32/esp-idf-v5.3/components/esp_lcd/include/esp_lcd_io_i2c.h:82:13: error: 'i2c_master_bus_handle_t' is not a class, namespace, or enumeration
   82 |             i2c_master_bus_handle_t : esp_lcd_new_panel_io_i2c_v2, \
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/home/danny/src/esp-idf/ZX3D95CD01S/minimal-gc9503/main/ft5x06.cpp:100:3: note: in expansion of macro 'esp_lcd_new_panel_io_i2c'
  100 |   esp_lcd_new_panel_io_i2c((esp_lcd_i2c_bus_handle_t)I2C_NUM_0, &io_config, &tp_io_handle);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/danny/src/github/esp32/esp-idf-v5.3/components/esp_lcd/include/esp_lcd_io_i2c.h:83:13: error: expected primary-expression before 'default'
   83 |             default : esp_lcd_new_panel_io_i2c_v1) (bus, io_config, ret_io) \
      |             ^~~~~~~
/home/danny/src/esp-idf/ZX3D95CD01S/minimal-gc9503/main/ft5x06.cpp:100:3: note: in expansion of macro 'esp_lcd_new_panel_io_i2c'
  100 |   esp_lcd_new_panel_io_i2c((esp_lcd_i2c_bus_handle_t)I2C_NUM_0, &io_config, &tp_io_handle);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/danny/src/github/esp32/esp-idf-v5.3/components/esp_lcd/include/esp_lcd_io_i2c.h:81:58: error: '_Generic' was not declared in this scope
   81 | #define esp_lcd_new_panel_io_i2c(bus, io_config, ret_io) _Generic((bus),  \
      |                                                          ^~~~~~
```

### Steps to reproduce.

Compile something that includes esp_lcd_new_panel_io_i2c with a C++ compiler

### Build or installation Logs.

```plain
see above
```

### More Information.

Fixed in https://github.com/espressif/esp-idf/issues/14037 but not included in 5.3, or so it would appear

suda-morris, 2024-07-26:
The backport has merged in our internal Gitlab server. It just needs some time to get synced to Github.

suda-morris, 2024-09-06:
Fixed in https://github.com/espressif/esp-idf/commit/0903bcca1bb1999b0431b68d9b496f5f406775ec

closed 2024-09-06 by suda-morris