espressif/esp-idf #14251: sdspi example error, it worked in Arduino IDE

aricagua, 2024-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The SD card should initialize successfully, allowing for filesystem mounting and operations.

### What is the actual behavior?

The SD card fails to initialize, resulting in a timeout error. The system is unable to mount the filesystem.

### Steps to reproduce.

1. Set up an ESP32-S3 with an SD (Adata 32GB) card using SDSPI.
2. Flash the example code (provided below) to the ESP32-S3.
3. Observe the serial output for initialization failure.

### Build or installation Logs.

```plain
Executing action: monitor
Running idf_monitor in directory /home/pepe/sd/sdspi
Executing "/home/pepe/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/pepe/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /home/pepe/sd/sdspi/build/sd_card.elf -m '/home/pepe/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/pepe/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
--- esp-idf-monitor 1.4.0 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�x��x���xx�x�xxx�x�xxx�x���x�x�x��x�xxxxxx�xx��x�xxxx����x�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (27) boot: compile time Jul 25 2024 23:18:02
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0ff54h ( 65364) map
I (111) esp_image: segment 1: paddr=0001ff7c vaddr=3fc94b00 size=0009ch (   156) load
I (112) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2dc30h (187440) map
I (151) esp_image: segment 3: paddr=0004dc58 vaddr=3fc94b9c size=02cdch ( 11484) load
I (154) esp_image: segment 4: paddr=0005093c vaddr=40374000 size=10a90h ( 68240) load
I (178) boot: Loaded app from partition at offset 0x10000
I (178) boot: Disabling RNG early entropy source...
I (190) cpu_start: Multicore app
I (199) cpu_start: Pro cpu start user code
I (199) cpu_start: cpu freq: 160000000 Hz
I (200) app_init: Application information:
I (202) app_init: Project name:     sd_card
I (207) app_init: App version:      1
I (212) app_init: Compile time:     Jul 25 2024 23:18:01
I (218) app_init: ELF file SHA256:  737f5f29a...
I (223) app_init: ESP-IDF:          v5.3-dirty
I (228) efuse_init: Min chip rev:     v0.0
I (233) efuse_init: Max chip rev:     v0.99 
I (238) efuse_init: Chip rev:         v0.1
I (243) heap_init: Initializing. RAM available for dynamic allocation:
I (250) heap_init: At 3FC981B0 len 00051560 (325 KiB): RAM
I (256) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (262) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (268) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (275) spi_flash: detected chip: gd
I (279) spi_flash: flash io: dio
W (283) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (296) sleep: Configure to isolate all GPIO pins in sleep state
I (303) sleep: Enable automatic switching of GPIO sleep configuration
I (310) main_task: Started on CPU0
I (330) main_task: Calling app_main()
I (330) example: Initializing SD card
I (330) example: Using SPI peripheral
I (330) example: Mounting filesystem
I (340) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (370) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
HINT: Please verify if there is an SD card inserted into the SD slot. Then, try rebooting the board.
I (370) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (370) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
HINT: Please refer ./README.md for details
I (380) main_task: Returned from app_main()
```

### More Information.

I've already tried adding pull-up resistors to the SD card lines.
I've attempted to slow down the SPI frequency.
The same SD card and ESP32-S3 with identical pin connections work correctly when using Arduino IDE.

aricagua, 2024-07-26:
I’m currently utilizing ESP-IDF version 5.3. I previously attempted to use version 5.2, but unfortunately, it didn’t yield the desired results. However, when I experimented with the Arduino ESP32 Core’s SD_test, it functioned flawlessly without any issues.

VojtechBartoska, 2024-07-29:
I see, understand.

Please give it a shot on 5.3 stable and report back if you still face some problem there: https://github.com/espressif/esp-idf/releases/tag/v5.3

aricagua, 2024-08-15:
It worked! Thanks

VojtechBartoska, 2024-08-21:
Perfect, closing the ticket.

closed 2024-08-21 by VojtechBartoska