espressif/esp-idf #14252: Add ESP_ERR_NOT_FINISHED to esp_now_send

dizcza, 2024-07-26:
### Is your feature request related to a problem?

I was getting ESP_ERR_ESPNOW_ARG and thinking that my input data format was incorrect. There was nothing wrong with the input data. The previous message hadn't been sent when I issued this command again getting ESP_ERR_ESPNOW_ARG instead of ESP_ERR_NOT_FINISHED.

### Describe the solution you'd like.

Output ESP_ERR_NOT_FINISHED when you call esp_now_send but the previous send command is not completed.

### Describe alternatives you've considered.

At least print a message in esp-wifi-lib that the esp-now is busy! Because it prints "Invalid argument" which is nonsense.

dizcza, 2024-07-26:
Well, I see in the docs

> Note that too short interval between sending two ESP-NOW data may lead to disorder of sending callback function. So, it is recommended that sending the next ESP-NOW data after the sending callback function of the previous sending has returned.

but it'd be nice to have a meaningful error in the code.

dizcza, 2024-07-26:
On second thought, maybe I was passing more than 250 bytes to `esp_now_send`... In either case, it's not clear:

Is `esp_now_send` a blocking operation? Does it return AFTER the data have been sent or does it queue the data to be sent later on?

zhangyanjiaoesp, 2024-11-07:
@dizcza 
1. Form this [branch](https://github.com/espressif/esp-idf/commit/ee74f278c89d47e95ba8bdd41e82d126f1329e0c) , the ESP-NOW supports longer data bytes (1490 bytes).
2. The `esp_now_send()` is not a blocking operation, it will return after post the data to wifi task.

dizcza, 2024-11-07:
So `ESP_ERR_NOT_FINISHED` doesn't make sense then since esp-now *always* says the data: it stacks in some buffer and does the job asynchronously for the user code.
I don't see any reason to limit the data size then...

dizcza, 2024-11-07:
I remember there was some sort of a callback indicating the success. Then I don't understand: if you accumulate the data and the user spits the data frequently to the esp-now, then indication of what packet would I get as success - the whole series is sent or?
If the callback is called after *all* packets / buffered data are sent, then it makes sense to add an indication of ESP_NOW_BUSY or whatever.

Anyhow. I no longer use esp-now. I opened the issue months ago...

You can close it.

Sherry616, 2024-11-08:
Thanks for reporting, we will close this issue. Feel free to reopen if have more requests. 
Thanks for using our Espressif product!

closed 2024-11-08 by Sherry616