espressif/esp-idf #14253: esp32s3 target does not support EXT_RAM_NOINIT_ATTR

pragulka, 2024-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

any

### Power Supply used.

USB

### What is the expected behavior?

Variables with EXT_RAM_NOINIT_ATTR attribute are placed in PSRAM

### What is the actual behavior?

Variables with EXT_RAM_NOINIT_ATTR attribute are placed in IRAM

### Steps to reproduce.

1. https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32s3/get-started/windows-setup.html
2. idf.py set-target esp32s3
3. idf.py menuconfig -> enable PSRAM
4. Add global variable in hello_world_main.c, eg:  EXT_RAM_NOINIT_ATTR volatile uint8_t _trace[1024*1024];
5. Add dummy read in app_main, eg: (void)_trace[1024*1024 - 1];
6. idf.py build
7. Observe: region `dram0_0_seg' overflowed by 763416 bytes

### More Information.

No  .noinit option under PSRAM config

Icarus113, 2024-07-29:
It's not supported yet, we will support it and let you know then

closed 2024-08-26 by espressif-bot

erkia, 2024-09-17:
Kudos! Always nice to see that the feature you are looking for was added just a few weeks ago :)