espressif/esp-idf #14255: V5.3使用usb_serial_jtag读取输入数据时，log打印不完整

johnzhy, 2024-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release v5.3

### Espressif SoC revision.

ESP32-S3FH4R2(QFN-56)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

我使用usb_serial_jtag作为日志输出，同时还想读取usb_serial_jtag输入的数据，在ESP-IDF V5.2.1版本下可以正常工作，输入数据和输出数据互相不影响

### What is the actual behavior?

升级到在ESP-IDF V5.2.2和V5.3，usb无法输出完整日志，日志中丢失大量字节

### Steps to reproduce.

1、usb_serial_jtag作为主要log输出
2、在main函数中启动读取usb数据的任务
`void usb_serial_jtag_proc(void *pvParameters)
{
    usb_serial_jtag_driver_config_t usb_serial_jtag_config;
    usb_serial_jtag_config.rx_buffer_size = 1024;
    usb_serial_jtag_config.tx_buffer_size = 1024;

    ESP_ERROR_CHECK(usb_serial_jtag_driver_install(&usb_serial_jtag_config));

    uint8_t * data = (uint8_t *)malloc(1024);
    while (1) {
        int len = usb_serial_jtag_read_bytes(data, 1023, 20/portTICK_PERIOD_MS);
        if(len)
        {
            data[len] = 0;
            ESP_LOGI("usb_serial_jtag", "read %d bytes: %s\n", len, data);
        }
    }
}

void usb_serial_jtag_proc_init(void)
{
    xTaskCreatePinnedToCore(&usb_serial_jtag_proc, "usb_serial_jtag", 4096*2, NULL, 20, NULL, 1);
}`
3、在main函数的while(1)中持续输出日志
`void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_AP");
    wifi_init_softap();
    extern void usb_serial_jtag_proc_init(void);
    usb_serial_jtag_proc_init();
    uint16_t i = 0;
    while(1)
    {
        ESP_LOGI(TAG, "sensor data: %d,%d,%d,%d,%d,%d,%d,end\r\n", i, i+1, i+2, i+3, i+4, i+5, i+6);
        i+= 1111;
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }
}`

### Debug Logs.

```plain
使用V5.2.1编译，使用串口工具打开串口，波特率设置2000000，间隔100ms发送数据01234567890123456789012345678901234567890123456789：
[08:21:07.816]收←◆[0;32mI (45215622) wifi softAP: sensor data: 7432,7433,7434,7435,7436,7437,7438,end
[0m

[08:21:07.821]发→◇01234567890123456789012345678901234567890123456789
□
[08:21:07.870]收←◆[0;32mI (45215692) usb_serial_jtag: read 52 bytes: 01234567890123456789012345678901234567890123456789

[0m

[08:21:07.908]收←◆[0;32mI (45215722) wifi softAP: sensor data: 8543,8544,8545,8546,8547,8548,8549,end
[0m

[08:21:07.914]发→◇01234567890123456789012345678901234567890123456789
□
[08:21:07.966]收←◆[0;32mI (45215782) usb_serial_jtag: read 52 bytes: 01234567890123456789012345678901234567890123456789

[0m

[08:21:08.003]收←◆[0;32mI (45215822) wifi softAP: sensor data: 9654,9655,9656,9657,9658,9659,9660,end
[0m

[08:21:08.022]发→◇01234567890123456789012345678901234567890123456789
□
[08:21:08.068]收←◆[0;32mI (45215892) usb_serial_jtag: read 52 bytes: 01234567890123456789012345678901234567890123456789

[0m

[08:21:08.113]收←◆[0;32mI (45215922) wifi softAP: sensor data: 10765,10766,10767,10768,10769,10770,10771,end
[0m

使用V5.3编译，不发送数据的情况下，日志也无法完整输出
[08:48:42.750]收←◆8,37
[08:48:42.787]收←◆819,end
[0m

[08:48:42.849]收←◆25,3
[08:48:42.886]收←◆896,38927,38928,38929,3830,end
[0m

[08:48:43.049]收←◆mI (
[08:48:43.086]收←◆21802) wifi softAP senso data:41146,41147,41148,4149,4150,41151,4152,end
[0m

[08:48:43.153]收←◆2257
[08:48:43.203]收←◆,42258,4259,42260,4226,4262,4263,end
[0m

[08:48:43.260]收←◆[0
[08:48:43.308]收←◆32mI (22002) wifi softAP:sesor data: 43368,43369,43370,43371,4332,43373,3374,end
[0m

[08:48:43.346]收←◆10
[08:48:43.389]收←◆2) wfi softAP sensor dat: 4479,44480,44481,44482,44483,44484,44485end
[0m

[08:48:43.546]收←◆0;
[08:48:43.590]收←◆32mI (22302) ifi softP: sensordata: 46701,46702,46703,4604,4675,46706,6707,end
[0m

[08:48:43.646]收←◆m
[08:48:43.690]收←◆ (2202) wii softAP: sensr data: 7812,4713,481,4815,4716,47817,47818,end
[0m

[08:48:43.748]收←◆en
[08:48:43.786]收←◆sor dta: 4823,48924,48925,48926,8927,4892,48929,ed
[0m

[08:48:43.849]收←◆[0
[08:48:43.892]收←◆;32mI 2202 wifi sotAP:sensor data:50034,5003,5036,5007,5038,50039,50040,end[m
[08:48:44.048]收←◆ (22802
[08:48:44.085]收←◆) wifi softP:sensor data:52256,52257,52258,5229,52260,52261,52262,en[0m

[08:48:44.150]收←◆32mI (
[08:48:44.186]收←◆2290) wifi oftAP: sesor ata:53367,5368,53369,5370,533715372,5337,end
[m

[08:48:44.249]收←◆78,5
[08:48:44.291]收←◆4479,54480,54481,5448254483,54484,end
[0m
```

peter-marcisovsky, 2024-07-29:
Hi @johnzhy, If I understood your problem correnctly, you are having problems with JTAG console on IDF 5.3, but the IDF 5.2 is working fine.
If that is the case, this problem is being solved right now. 
It has been already fixed in the master branch by [this](https://github.com/espressif/esp-idf/commit/7982ece4af9e6ed35be45efb63946b7ede7b8745) commit. 
Also has been backported to the IDF 5.2 by [this](https://github.com/espressif/esp-idf/commit/4af418c534246766325914c477a8c88c39541d21) commit. 
Unfortunatelly, the backport for the 5.3 is still open. 
Please keep using 5.2 until the backport is merged. 
We will keep the issue open for tracking, until the changes are merged to the 5.3

johnzhy, 2024-07-29:
> Hi @johnzhy, If I understood your problem correnctly, you are having problems with JTAG console on IDF 5.3, but the IDF 5.2 is working fine. If that is the case, this problem is being solved right [...]

Ok, thanks for your reply

peter-marcisovsky, 2024-07-30:
The backport to IDF 5.3 was just merged. Now it will take some time for the release branch to sync with GitHub.

peter-marcisovsky, 2024-08-02:
The branches have been synced and the changes are present in the IDF 5.3 release [here](https://github.com/espressif/esp-idf/commit/a0dbe28c9f30f54582a56f139fddf571d93062a4).
Feel free to close the issue if your problem has been resolved.

johnzhy, 2024-08-05:
The problem has been solved

closed 2024-08-05 by johnzhy