espressif/esp-idf #14256: BCT test fails when using a wired network

alikschen, 2024-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Tested 6 times using wifi mode, failed on the 2nd and 6th time, and succeeded on the other 4 times.
In the same code, when testing BCT in wired network mode, it fails once. Can you help analyze the reason?

The following is the information of WIFI mode test:
[BCT.pcap.zip](https://github.com/user-attachments/files/16398638/BCT.pcap.zip)
[log.txt](https://github.com/user-attachments/files/16398639/log.txt)
[uart-log (1).txt](https://github.com/user-attachments/files/16398641/uart-log.1.txt)

The following is the information of the wired network mode test:
[log_wire.txt](https://github.com/user-attachments/files/16398663/log_wire.txt)
[BCT 2.pcap.zip](https://github.com/user-attachments/files/16398666/BCT.2.pcap.zip)

shahpiyushv, 2024-08-02:
To add some more info, the HomeKit examples in [esp-homekit-sdk](https://github.com/espressif/esp-homekit-sdk) that use Wi-Fi pass the BCT tests most of the times. The intermittent failures are expected since BCT is very time sensitive. Howeer, with Ethernet transport, the hot-plugging tests are always failing. For Wi-Fi, to simulate hot plugging, we disconnect WI-Fi and reconnect after 10 seconds (Check code [here](https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/components/homekit/esp_hap_core/src/esp_hap_bct.c#L65)). For Ethernet, the cable is supposed to be removed and re-inserted after 10 seconds. SO, the mDNS behaviour could be slightly different (just a guess)

david-cermak, 2024-09-04:
@alikschen Could you please share the version of IDF and MDNS that you're using?
(I can see IDF version could be `v5.0.2-dirty`, but MDNS is an external dependency, cannot guess)

Also, if you could also attach the logs from board (uart-log.1.txt) in the wired test?

>  For Ethernet, the cable is supposed to be removed and re-inserted after 10 seconds.

Did you physically disconnected and reconnected the Ethernet cable?

alikschen, 2024-09-10:
espressif/mdns:
    component_hash: d96d277207c4fec77ec290b7a8b0ca4043e10f407df0ff6aa52e54a94f0c54a5
    version: 1.2.4

LinKvFM, 2024-09-12:
I also encountered a similar BCT failure when using a wired network. 
IDF version : v5.2.1
[debug.log](https://github.com/user-attachments/files/16975157/debug.log)
[Result-bridge-wired.txt](https://github.com/user-attachments/files/16975158/Result-bridge-wired.txt)
[BCT.zip](https://github.com/user-attachments/files/16975159/BCT.zip)

david-cermak, 2024-09-19:
@LinKvFM From the log it looks like the cable handling case passed in your case, but fails on `SUBSEQUENT CONFLICTS` ? Does it work with wireless network?
Is it repeatable?
Could you please your LWIP configuration?

Just trying to reproduce this and it passes with my BCT (v1.5), but probes only after 8-10-the attempt. Does it fail repeatably for you or passes sometimes?

alikschen, 2024-09-19:
Yes, most questions appear in SUBSEQUENT CONFLICTS，All will be retried, but 90% of the tests will fail

陈永喜   乐式科技
Tel: 400-1588-162 丨 Web: lelight.cn  丨 ***@***.***
总部：深圳市前海前湾一路1号A栋； 中山运营中心：中山市古镇镇东兴东路33号6栋乐式科技
 
From: david-cermak
Date: 2024-09-19 15:07
To: espressif/esp-idf
CC: alikschen; Mention
Subject: Re: [espressif/esp-idf] BCT test fails when using a wired network (IDFGH-13340) (Issue #14256)
@LinKvFM From the log it looks like the cable handling case passed in your case, but fails on SUBSEQUENT CONFLICTS ? Does it work with wireless network?
Is it repeatable?
Could you please your LWIP configuration?
Just trying to reproduce this and it passes with my BCT (v1.5), but probes only after 8-10-the attempt. Does it fail repeatably for you or passes sometimes?

david-cermak, 2024-09-19:
@alikschen Can you try to setup:
```
CONFIG_LWIP_AUTOIP_RATE_LIMIT_INTERVAL=10
```
(or any number between 10 and 60 would do)
You'd get some (more?) warning during initial probing, but `SUBSEQUENT CONFLICTS` should pass

LinKvFM, 2024-09-19:
@david-cermak This is my configuration file. [config.zip](https://github.com/user-attachments/files/17056130/config.zip)
Only test with wired, most failed on SUBSEQUENT CONFLICTS.

david-cermak, 2024-09-19:
@LinKvFM Thanks for sharing! The AUTOIP related configs are:
```
CONFIG_LWIP_AUTOIP=y
CONFIG_LWIP_AUTOIP_TRIES=2
CONFIG_LWIP_AUTOIP_MAX_CONFLICTS=9
CONFIG_LWIP_AUTOIP_RATE_LIMIT_INTERVAL=60
```

Could you please also try to decrease the rate limiting interval to say 10-20 seconds? (as mentioned in https://github.com/espressif/esp-idf/issues/14256#issuecomment-2360243536)

This should help at least pass the BCT with warnings (cleaner fix would be I believe in using the ACD module in the new lwip -- only available in v5.4)

alikschen, 2024-09-20:
After modification, the test was performed twice and the result failedCONFIG_LWIP_AUTOIP_RATE_LIMIT_INTERVAL=15

陈永喜   乐式科技
Tel: 400-1588-162 丨 Web: lelight.cn  丨 ***@***.***
总部：深圳市前海前湾一路1号A栋； 中山运营中心：中山市古镇镇东兴东路33号6栋乐式科技
 
From: david-cermak
Date: 2024-09-19 15:50
To: espressif/esp-idf
CC: alikschen; Mention
Subject: Re: [espressif/esp-idf] BCT test fails when using a wired network (IDFGH-13340) (Issue #14256)
@alikschen Can you try to setup:
CONFIG_LWIP_AUTOIP_RATE_LIMIT_INTERVAL=10

(or any number between 10 and 60 would do)
You'd get some (more?) warning during initial probing, but SUBSEQUENT CONFLICTS should pass

david-cermak, 2024-09-20:
> the test was performed twice and the result failed

@alikschen but the `"SUBSEQUENT CONFLICTS"` passed this time, correct? Could you please share some logs from the tests and the board?
Also noticed that in your original report, the Link allocation subtest passed as well, but failing in multicatst subtest and cable handling? (that's why I asked about the serial logs in my earlier post https://github.com/espressif/esp-idf/issues/14256#issuecomment-2329260560, as I'm interested in the Ethernet events while re-connecting the cable)

I'm sorry, it seems like we have two separate problems reported with two distinct root causes in this issue, one related to AUTOIP by @LinKvFM and another one with mDNS by @alikschen.

LinKvFM, 2024-09-20:
@david-cermak  is my new test result, it seems that there have been more failures.
[IPv4.zip](https://github.com/user-attachments/files/17070554/IPv4.zip)

alikschen, 2024-09-20:
CONFIG_LWIP_AUTOIP=y
CONFIG_LWIP_AUTOIP_TRIES=2
CONFIG_LWIP_AUTOIP_MAX_CONFLICTS=9
CONFIG_LWIP_AUTOIP_RATE_LIMIT_INTERVAL=60Now we need to use this configuration for testing. Do we need onboard logs and BCT logs?

陈永喜   乐式科技
Tel: 400-1588-162 丨 Web: lelight.cn  丨 ***@***.***
总部：深圳市前海前湾一路1号A栋； 中山运营中心：中山市古镇镇东兴东路33号6栋乐式科技
 
From: LinKvFM
Date: 2024-09-20 14:01
To: espressif/esp-idf
CC: alikschen; Mention
Subject: Re: [espressif/esp-idf] BCT test fails when using a wired network (IDFGH-13340) (Issue #14256)
@david-cermak is my new test result, it seems that there have been more failures.
IPv4.zip

david-cermak, 2024-09-20:
I can see that the link local tests passed in your case, too.
Let's focus on the multicast failures, now.

@LinKvFM Have you really updated the `CONFIG_LWIP_AUTOIP_RATE_LIMIT_INTERVAL` value? I don't see any warning in the initial probing in your logs?

LinKvFM, 2024-09-20:
@david-cermak Have not update it. Same configuration file, but different failure cases.

david-cermak, 2024-09-20:
@LinKvFM Could you please share the logs, then? You don't have to copy-paste it from your console, there's some utility features in IDF monitor, like Ctrl-T + Ctrl+L: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-monitor.html#keyboard-shortcuts

I'm sorry for asking *again* about the logs, but seeing that your test always fails differently (with the same setup) there must be something I am missing, perhaps related to esp-homekit? I believe I could find some pointers in logs (haven't found any in the wireshark capture, so-far)

LinKvFM, 2024-09-21:
@david-cermak I modified the configuration and tried again, the same result.
[BCT0921.zip](https://github.com/user-attachments/files/17082623/BCT0921.zip)

LinKvFM, 2024-09-23:
@david-cermak I have updated the version of [mDNS](https://components.espressif.com/components/espressif/mdns) to v1.4.0.
The result was also failed:
[BCT0921-2 (2).zip](https://github.com/user-attachments/files/17093372/BCT0921-2.2.zip)

david-cermak, 2024-09-24:
@LinKvFM Thanks for sharing more details. I'm sorry I still cannot reproduce your issues with my BCT v1.5. I'll try to test with v1.5.3 and a homekit project.

The only thing that I found strange in your logs is that you're supposed to disconnect and reconnect the network cable twice during the test, correct? First time for the link allocation test (AUTOIP) and second time for the hot plugging test (MDNS), but I see only one disconnecting in your logs. Could you explain?

Could you please rerun the test again with `CONFIG_MDNS_ENABLE_DEBUG_PRINTS=y` and also share the logs from the board?

david-cermak, 2024-09-24:
@LinKvFM still wasn't able to repeatable reproduce the issues you're having with the hot-plug tests, but had a suspicion there might be some trouble with IPv6 announcement upon acquiring IPv4 address through Ethernet.
To confirm this theory, I've prepared a test version of mDNS v1.4.2 in the staging area: https://components-staging.espressif.com/components/espressif/mdns/versions/1.4.2?language=en

could you please try with this version? Here's how you'd modify your `idf_component.yml`:
```yml
  espressif/mdns:
    version: "==1.4.2"
    service_url: "https://components-staging.espressif.com"
```

LinKvFM, 2024-09-24:
@david-cermak Yes, there was only one disconnection, and since the test was only running for about half an hour, I think it was because the test ended before the hot plugging test (mDNS) was run.
I will update the mDNS version to 1.4.2 and test again.

david-cermak, 2024-09-24:
> I will update the mDNS version to 1.4.2 and test again.

just set the original AUTOIP configuration with the staged mDNS, please
```
CONFIG_LWIP_AUTOIP=y
CONFIG_LWIP_AUTOIP_TRIES=2
CONFIG_LWIP_AUTOIP_MAX_CONFLICTS=9
CONFIG_LWIP_AUTOIP_RATE_LIMIT_INTERVAL=60
```
thanks!

LinKvFM, 2024-09-24:
@david-cermak A fatal error have occurred, but I have not run BCT yet. 
[idf_py_stdout.zip](https://github.com/user-attachments/files/17115278/idf_py_stdout.zip)

david-cermak, 2024-09-24:
> @david-cermak A fatal error have occurred, but I have not run BCT yet. [idf_py_stdout.zip](https://github.com/user-attachments/files/17115278/idf_py_stdout.zip)

This is weird, I've tested this again and no issue on my side. I've noticed though that you're printing the packet content (which is quite long). Could you please check whether the exception is related to those debug prints, disabling the `CONFIG_MDNS_ENABLE_DEBUG_PRINTS` ? Also it would be nice if you can decode the addresses from the backtrace with your elf file.

LinKvFM, 2024-09-25:
@david-cermak It also failed just like [this:](https://github.com/espressif/esp-idf/issues/14256#issuecomment-2364978053) 
[BCT0925.zip](https://github.com/user-attachments/files/17123247/BCT0925.zip)
mDNS version 1.4.2.

david-cermak, 2024-09-26:
@LinKvFM Okay, I'll try to reproduce the problems with the exact project and the same version of BCT you're using. Could you please share the entire project (with your modifications)? you can also share it via the technical support team.

In the meantime, could you please rerun the test with wireless network? AFAICT that is the supported option with homekit devices and works with BCT. This would help us rule out some weird issues cause by your test environment.

david-cermak, 2024-10-03:
@LinKvFM I've created a sample project from the `smart_outlet` example in https://github.com/espressif/esp-homekit-sdk/pull/143/files
Also updated my BCT to v1.5.4 and it seems to be mostly passing (some intermittent trouble with the premature probing).
Could you please check if the BCT works a little more predictable with this example?

Also, please share some more detail of your project (again, with the tech support if you prefer), specifically about your Ethernet chip and target.
Another theory what might have gone wrong in your scenario is that in your init code you'd set the interface keys to:
```cpp
        strcat(strcpy(if_key_str, "ETH_SPI_"), num_str);
        strcat(strcpy(if_desc_str, "eth"), num_str);
```
(assuming you're using SPI Ethernet)
But with the created example we always assume only one eth instance:

https://github.com/espressif/esp-homekit-sdk/blob/83fcfe6e4c534ec7773886fef64cf2a02a7f5f45/examples/smart_outlet/main/app_main.c#L163-L166

and that's why we could use the default interface key and mdns layer "knows" about it.

LinKvFM, 2024-10-10:
@david-cermak I created a sample project from the `smart_outlet` example according to your prompt in [https://github.com/LinKvFM/smart_outlet/tree/master](https://github.com/LinKvFM/smart_outlet/tree/master)

And the results:[BCT.zip](https://github.com/user-attachments/files/17323371/BCT.zip)

In addition to IPv4, it seems that IPv6 cannot be initialized. Do I need to do anything else to enable IPv6?

LinKvFM, 2024-10-10:
I forgot to tell you that the Ethernet chip we use is LAN8700. It is supported in idf.

LinKvFM, 2024-10-10:
@david-cermak Could you please share your test results, or your operation video or steps? I think there are many differences between our tests.

david-cermak, 2024-10-10:
Here's the last result I saved the other day; as mentioned before the premature probes fail in this instance, but all the multicast tests pass. I'll re-run and share more details later.

[bct_log.txt](https://github.com/user-attachments/files/17333216/bct_log.txt)

You can also try to run this project https://github.com/espressif/esp-idf/issues/14256#issuecomment-2390780448 in your environment (you can easily select the LAN8720 in menuconfig). I deliberately added the project in the PR to highlight the changes (70 lines total, which is effectively like 10 lines of code to support Ethernet).

Unlike your sample project with:
```
753 files changed  |  +108468 -0 lines changed
```
is much harder to follow for me, but I'll try...

LinKvFM, 2024-10-11:
@david-cermak I tried running your project [#14256(comment)](https://github.com/espressif/esp-idf/issues/14256#issuecomment-2390780448) more times, but it failed yet.
[bct-1011.zip](https://github.com/user-attachments/files/17336976/bct-1011.zip)
[sdkconfig.zip](https://github.com/user-attachments/files/17337021/sdkconfig.zip)

idf version:[release/v5.4](https://github.com/espressif/esp-idf/tree/release/v5.4)
espressif/ethernet_init: v0.3.1
router: AirPort-Extreme
target: esp32     PSRAM:2MB    flash: 16MB
Ethernet chip: LAN8720A

Before that, my idf version is [v5.2.1](https://github.com/espressif/esp-idf/tree/v5.2.1)
espressif/ethernet_init: v0.3.0

david-cermak, 2024-10-11:
@LinKvFM Please do not use IDF `v5.4`. There're still some known issues, mainly related to AUTOIP, so you won't pass link allocations. (both `v5.2` and `v5.3` are okay, ethernet_init version should be ~v0.3; I was on IDF `v5.3` when performing the tests)

> it seems that IPv6 cannot be initialized. Do I need to do anything else to enable IPv6?

I'll add IPv6 support to the sample project, need to call `esp_netif_create_ip6_linklocal()` after connecting. But let's focus on IPv4 first.

LinKvFM, 2024-10-11:
@david-cermak I changed idf version to 5.3 and tried again, nothing change.
[bct-ipv4-5.3.zip](https://github.com/user-attachments/files/17338564/bct-ipv4-5.3.zip)
Is there anything else I need to modify? Or is there something wrong with my steps?

david-cermak, 2024-10-11:
> I changed idf version to 5.3 and tried again, nothing change.

well, the results look very different to me, now the link allocations pass and the test's failing on mdns probes. why is that I cannot tell.

> Or is there something wrong with my steps?

I'd start with something that works, rather than looking for strange issues if the BCT fails differently every time you run it.

I'd suggest that I'd re-run it on my side multiple times and have the technical support team test it on their environment, as well. This way we'll (hopefully) have two separate instances of "golden samples". Then, the tech support team can walk you through the BCT process, step by step.

david-cermak, 2024-10-11:
> @david-cermak I changed idf version to 5.3 and tried again, nothing change. [bct-ipv4-5.3.zip](https://github.com/user-attachments/files/17338564/bct-ipv4-5.3.zip) Is there anything else I need to modify? Or is there something wrong with my steps?

Tip from @shahpiyushv Might be an issue with pairing. Could you please run the test again with the updated project:

https://github.com/espressif/esp-homekit-sdk/pull/143/commits/48b930bbb25065a54225bd720b2a86d7b76c3213

(I've commented out the pairing timer, so the mDNS would still work after these 10 minutes)

shahpiyushv, 2024-10-11:
@LinKvFM , if you recollect, we had mentioned separately that the HomeKit accessory needs to be paired with a controller, else the mDNS service is removed after 10 min. From your logs, I see sf=3 as TXT record, which means the accessory was not paired and as you can see, the tests have indeed failed after 10 minutes.

LinKvFM, 2024-10-11:
@shahpiyushv Yes, I remembered. @david-cermak Now I ran the test again after pairing the device. It seemed to complete more cases this time, but it still failed in the end. 
my project:  paired with a controller
[bct-ipv4.zip](https://github.com/user-attachments/files/17342171/bct-ipv4.zip)

smart_outlet example:  Remove pairing timer but not paired.
[bct-ipv4-smartOutlet.zip](https://github.com/user-attachments/files/17342246/bct-ipv4-smartOutlet.zip)

LinKvFM, 2024-10-14:
@david-cermak BCT passed when using WiFi. Can it be said that the failure is caused by Ethernet?

david-cermak, 2024-10-21:
@LinKvFM 

Could you please check again with the temporarily modified mdns library?
```bash
  espressif/mdns:
    version: "==1.4.2"
    service_url: "https://components-staging.espressif.com"
```
That's the only difference I could think of between WiFi and Ethernet (this adds another announcement on IPv4 whenever we get a new IPv6)

LinKvFM, 2024-10-22:
@david-cermak  Yes, I have been using version 1.4.2 and IPv6 has been enabled.

david-cermak, 2024-10-24:
I've repeated the test 10 times and it passed 10 times, have skipped some long running tests. like chattiness and link alloc and name change to focus on the problematic ones, like cable handling and hot swapping (also ran the whole test once, it passed too).

Could you please pull the latest update from the PR:
https://github.com/espressif/esp-homekit-sdk/pull/143/commits/4a17a28da92113fa76a1fbaf2861e23bbb56009c
(updated mdns to use custom netif, so the transitions are defined exactly just for your concrete network interface).
Here's the list of my component:
```
  espressif/ch390:
    component_hash: 5e0a9ee81b930ed04f9ad0242044454858194932acb521bd85cd2b9e92c2395d
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 0.2.0
  espressif/ethernet_init:
    component_hash: 3cd85b7a727f1d16dc4e46dff61f72b8cc2940b93f0a9bf0acb475faae44ea40
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 0.3.1
  espressif/lan867x:
    component_hash: 9f6f025a9de7a8146b6bc5ebb34242f4fdbf73f3d61f39fde6e7d85281d0ad6d
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.0.1
  espressif/libsodium:
    component_hash: f6e982479a2389cb6868e8fb761cf23aba6c355a8090b3e906299807775f58a3
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.0.20~1
  espressif/mdns:
    component_hash: af6306fe65d637a3683d1cf671508fcedd6b05f9ca029a8815abeab64001fb8d
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.4.0
```
please, use a clean project and do not set any other configuration, but those pre-defined in the `sdkconfig.default`

* start the simplified test first (skipping the long checks)
```
sudo BonjourConformanceTest -I enXX -L psrc -M n -N uct -D -F log.txt -Aip XXX -Amac YYY -V
```
and share the logs from the board.

LinKvFM, 2024-10-25:
@david-cermak Failed yet.
[ipv4.zip](https://github.com/user-attachments/files/17514934/ipv4.zip)
[default_2024-10-25_10_02_39.log](https://github.com/user-attachments/files/17514937/default_2024-10-25_10_02_39.log)

LinKvFM, 2024-10-25:
@david-cermak This case(`HOT-PLUGGING: WINNING SIMULTANEOUS PROBES`) has the most failures.
[ipv4-10.zip](https://github.com/user-attachments/files/17519797/ipv4-10.zip)
[log.smart_outlet.20241025165819.txt](https://github.com/user-attachments/files/17519806/log.smart_outlet.20241025165819.txt)
The list of my component:
```
dependencies:
  espressif/ch390:
    component_hash: 5e0a9ee81b930ed04f9ad0242044454858194932acb521bd85cd2b9e92c2395d
    dependencies:
    - name: idf
      require: private
      version: '>=5.1'
    source:
      registry_url: https://components.espressif.com
      type: service
    version: 0.2.0
  espressif/enc28j60:
    component_hash: ac9af63b0e9bc7dd9a8b252d0c58db448f2b57b01291aecd353dc1c4b8fe0a62
    dependencies:
    - name: idf
      require: private
      version: '>=4.4'
    source:
      registry_url: https://components.espressif.com
      type: service
    version: 1.0.0
  espressif/ethernet_init:
    component_hash: 4ecf9e57793e79df2c7b7bd62fbe189ac347a3bfed5378244776186898f71916
    dependencies:
    - name: espressif/ch390
      registry_url: https://components.espressif.com
      require: private
      version: '*'
    - name: espressif/enc28j60
      registry_url: https://components.espressif.com
      require: private
      version: '*'
    - name: espressif/lan867x
      registry_url: https://components.espressif.com
      require: private
      rules:
      - if: target in [esp32, esp32p4]
      version: '*'
    - name: idf
      require: private
      version: '>=5.1'
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 0.3.2
  espressif/lan867x:
    component_hash: 9f6f025a9de7a8146b6bc5ebb34242f4fdbf73f3d61f39fde6e7d85281d0ad6d
    dependencies:
    - name: idf
      require: private
      version: '>=5.3'
    source:
      registry_url: https://components.espressif.com
      type: service
    targets:
    - esp32
    - esp32p4
    version: 1.0.1
  espressif/libsodium:
    component_hash: f6e982479a2389cb6868e8fb761cf23aba6c355a8090b3e906299807775f58a3
    dependencies:
    - name: idf
      require: private
      version: '>=4.2'
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 1.0.20~1
  espressif/mdns:
    component_hash: af6306fe65d637a3683d1cf671508fcedd6b05f9ca029a8815abeab64001fb8d
    dependencies:
    - name: idf
      require: private
      version: '>=5.0'
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 1.4.0
  idf:
    source:
      type: idf
    version: 5.3.1
direct_dependencies:
- espressif/ethernet_init
- espressif/libsodium
- espressif/mdns
- idf
manifest_hash: 63bbbb22c4ee12a299db87c01508de4c2922c7ff6ea51e31992cc6c5228dab62
target: esp32
version: 2.0.0
```

david-cermak, 2024-10-25:
@LinKvFM Also tried to run the BCT with `LAN8720` and passed on my side as well. As suggested some time ago, let's run this via the tech support team and try to replicate the environment and the board that's passing on my developer's desk also on someone else's.
I cannot tell what could be wrong, but we should focus on the HW and environment. Is it possible for you to re-run the test with another router or another laptop?

What's weird in your last capture is that it seemed like it wasn't captured during the same test as the logs you shared. Have spent some time looking into the packet log and here's the strange part (starting from packet no 572):
```
No.     Time           Source                Destination           Protocol Length Info
    572 492.298555     169.254.201.226       224.0.0.251           MDNS     106    Standard query 0x0000 ANY MyHost-42.local, "QM" question A 168.254.252.69
    573 492.597371     169.254.252.69        224.0.0.251           MDNS     153    Standard query 0x0000 ANY Esp-Smart-Outlet-12._hap._tcp.local, "QM" question ANY MyHost-42.local, "QM" question ANY MyHost-42.local, "QM" question SRV 0 0 80 MyHost-42.local A 169.254.252.69
    574 492.598460     169.254.201.226       224.0.0.251           MDNS     106    Standard query 0x0000 ANY MyHost-42.local, "QM" question A 168.254.252.69
    575 492.600030     169.254.252.69        224.0.0.251           MDNS     85     Standard query response 0x0000 A, cache flush 169.254.252.69
    578 492.897564     169.254.252.69        224.0.0.251           MDNS     284    Standard query response 0x0000 PTR _hap._tcp.local PTR Esp-Smart-Outlet-12._hap._tcp.local SRV, cache flush 0 0 80 MyHost-42.local TXT, cache flush A, cache flush 169.254.252.69
```
while your test log says:
```
*** 17:06:39 Starting sub-test HOT-PLUGGING: WINNING SIMULTANEOUS PROBES
NOTICE  2024-10-25 17:06:39.608593+0800    winning_sim_probe 01160: sending conflicting probe for MyHost-42.local., tiebreaker in favor of device.
NOTICE  2024-10-25 17:06:39.908506+0800    winning_sim_probe 01160: sending conflicting probe for MyHost-42.local., tiebreaker in favor of device.
WARNING 2024-10-25 17:07:11.913081+0800    winning_sim_probe 01172: Did not get a response. Possible dropped packet. Retrying: 2 attempt
NOTICE  2024-10-25 17:07:11.913342+0800    winning_sim_probe 01160: sending conflicting probe for MyHost-42.local., tiebreaker in favor of device.
WARNING 2024-10-25 17:07:41.920411+0800    winning_sim_probe 01172: Did not get a response. Possible dropped packet. Retrying: 3 attempt
NOTICE  2024-10-25 17:07:41.920708+0800    winning_sim_probe 01160: sending conflicting probe for MyHost-42.local., tiebreaker in favor of device.
WARNING 2024-10-25 17:08:11.928597+0800    winning_sim_probe 01172: Did not get a response. Possible dropped packet. Retrying: 4 attempt
NOTICE  2024-10-25 17:08:11.928974+0800    winning_sim_probe 01160: sending conflicting probe for MyHost-42.local., tiebreaker in favor of device.
WARNING 2024-10-25 17:08:41.936949+0800    winning_sim_probe 01172: Did not get a response. Possible dropped packet. Retrying: 5 attempt
NOTICE  2024-10-25 17:08:41.937243+0800    winning_sim_probe 01160: sending conflicting probe for MyHost-42.local., tiebreaker in favor of device.
WARNING 2024-10-25 17:09:11.944652+0800    winning_sim_probe 01172: Did not get a response. Possible dropped packet. Retrying: 6 attempt
NOTICE  2024-10-25 17:09:56.949832+0800     mdns_recv_answer 00481: Timed out
ERROR   2024-10-25 17:09:56.949939+0800    recv_announcement 03440: Device did not send gratuitous announcement(s) for:
	MyHost-42.local.
FAILED (HOT-PLUGGING: WINNING SIMULTANEOUS PROBES)
```
i.e. the BCT's retrying to send a conflicting probe ~6 times, but we're seeing only 2 packets!

Please compare to my results:
```
No.     Time           Source                Destination           Protocol Length Info
    118 40.608933      169.254.125.0         224.0.0.251           MDNS     106    Standard query 0x0000 ANY MyHost-43.local, "QU" question A 168.254.176.38
    119 40.908246      169.254.176.38        224.0.0.251           MDNS     153    Standard query 0x0000 ANY Esp-Smart-Outlet-14._hap._tcp.local, "QM" question ANY MyHost-43.local, "QM" question ANY MyHost-43.local, "QM" question SRV 0 0 80 MyHost-43.local A 169.254.176.38
    120 40.908570      169.254.125.0         224.0.0.251           MDNS     106    Standard query 0x0000 ANY MyHost-43.local, "QU" question A 168.254.176.38
    121 41.208227      169.254.176.38        224.0.0.251           MDNS     153    Standard query 0x0000 ANY Esp-Smart-Outlet-14._hap._tcp.local, "QM" question ANY MyHost-43.local, "QM" question ANY MyHost-43.local, "QM" question SRV 0 0 80 MyHost-43.local A 169.254.176.38
    122 41.508310      169.254.176.38        224.0.0.251           MDNS     283    Standard query response 0x0000 PTR _hap._tcp.local PTR Esp-Smart-Outlet-14._hap._tcp.local SRV, cache flush 0 0 80 MyHost-43.local TXT, cache flush A, cache flush 169.254.176.38
```

and the corresponding BCT log
```
*** 18:27:56 Starting sub-test HOT-PLUGGING: WINNING SIMULTANEOUS PROBES
NOTICE  2024-10-25 18:27:56.276516+0200    winning_sim_probe 01160: sending conflicting probe for MyHost-43.local., tiebreaker in favor of device.
NOTICE  2024-10-25 18:27:56.576157+0200    winning_sim_probe 01160: sending conflicting probe for MyHost-43.local., tiebreaker in favor of device.
NOTICE  2024-10-25 18:27:57.176114+0200    recv_announcement 03418: Received announcement for MyHost-43.local.
PASSED (HOT-PLUGGING: WINNING SIMULTANEOUS PROBES)
```
seeing exactly 2 probes and one announcement (same as your packet capture, btw)

LinKvFM, 2024-10-28:
@david-cermak Could you please compare the hardware and configuration of your Ethernet chip `LAN8720`?
my sdkconfig:
```
...
#
# Ethernet Configuration
#
CONFIG_ENV_GPIO_RANGE_MIN=0
CONFIG_ENV_GPIO_RANGE_MAX=39
CONFIG_ENV_GPIO_IN_RANGE_MAX=39
CONFIG_ENV_GPIO_OUT_RANGE_MAX=33
CONFIG_ETHERNET_RX_TASK_STACK_SIZE=2048
CONFIG_ETHERNET_INTERNAL_SUPPORT=y
# CONFIG_ETHERNET_PHY_IP101 is not set
# CONFIG_ETHERNET_PHY_RTL8201 is not set
CONFIG_ETHERNET_PHY_LAN87XX=y
# CONFIG_ETHERNET_PHY_DP83848 is not set
# CONFIG_ETHERNET_PHY_KSZ80XX is not set
# CONFIG_ETHERNET_PHY_LAN867X is not set
CONFIG_ETHERNET_MDC_GPIO=23
CONFIG_ETHERNET_MDIO_GPIO=18
CONFIG_ETHERNET_PHY_RST_GPIO=-1
CONFIG_ETHERNET_PHY_ADDR=1
# CONFIG_ETHERNET_SPI_SUPPORT is not set
# end of Ethernet Configuration
...
```
```
### Using ESP32 internal MAC

* RMII PHY wiring is fixed and can not be changed through either IOMUX or GPIO Matrix. By default, they're connected as follows:

| GPIO   | RMII Signal | Notes        |
| ------ | ----------- | ------------ |
| GPIO21 | TX_EN       | EMAC_TX_EN   |
| GPIO19 | TX0         | EMAC_TXD0    |
| GPIO22 | TX1         | EMAC_TXD1    |
| GPIO25 | RX0         | EMAC_RXD0    |
| GPIO26 | RX1         | EMAC_RXD1    |
| GPIO27 | CRS_DV      | EMAC_RX_DRV  |

* One of the following GPIO pins can be used as RMII REF_CLK input/output:

| GPIO   | Function             | Notes        |
| ------ | -------------------- | ------------ |
| GPIO0  | EMAC_TX_CLK/CLK_OUT1 | input/output |
| GPIO16 | EMAC_CLK_OUT         | output       |
| GPIO17 | EMAC_CLK_180         | output       |

* SMI (Serial Management Interface) wiring is not fixed. You may need to changed it according to your board schematic. By default they're connected as follows:

| GPIO   | SMI Signal  | Notes         |
| ------ | ----------- | ------------- |
| GPIO23 | MDC         | Output to PHY |
| GPIO18 | MDIO        | Bidirectional |
```
By the way, I set RMII REF_CLK to output to GPIO17.

david-cermak, 2024-10-29:
My Ethernet config looks similar (only options different from defaults)
```
CONFIG_ETHERNET_PHY_LAN87XX=y
CONFIG_ETHERNET_PHY_ADDR=0
CONFIG_ETH_RMII_CLK_OUTPUT=y
```
But, I use an actual reset GPIO.
And I also use CLK output to GPIO 17. (unrelated to this issue, just a note about this [warning](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-reference/network/esp_eth.html#_CPPv4N22emac_rmii_clock_gpio_t22EMAC_APPL_CLK_OUT_GPIOE) in case you'd like to use WiFi in you application, as well)

Is it possible for you to get hold of the standard Ethernet kit: https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/hw-reference/esp32/get-started-ethernet-kit.html ?
It would be good if you can run one test with a standard board, just to rule out potential HW related issues.
(although from the logs you shared it looked more like an environmental issue, then HW related one)

LinKvFM, 2024-10-29:
> Is it possible for you to get hold of the standard Ethernet kit: https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/hw-reference/esp32/get-started-ethernet-kit.html ?

@david-cermak I don't have that kit. I'm using this one,(https://www.waveshare.com/lan8720-eth-board.htm) and ESP32-DevKitC(https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/hw-reference/esp32/get-started-devkitc.html#). 

In addition, I used another router (Netgear) and the same failed occurred. It may be that I set it up incorrectly, because I don't know how to set up the static bridge mode like Apple. I just turned off DHCP. The laptop is new, so I don't think it has anything to do with it.

LinKvFM, 2024-11-01:
@david-cermak Can you share some of the kits you used for BCT, such as chip, router or laptop, etc.

david-cermak, 2024-11-01:
> @david-cermak Can you share some of the kits you used for BCT, such as chip, router or laptop, etc.

laptop: MacBook Pro 2017 with macOS Ventura 13.7
router: AirPort Extreme
board: 
- standard EthernetKIt with IP101: https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/hw-reference/esp32/get-started-ethernet-kit.html 
- custom board with LAN8720

LinKvFM, 2024-11-05:
@david-cermak Great, ipv4 has been solved, now we can focus on ipv6. How to enable ipv6 base on [https://github.com/espressif/esp-idf/issues/14256#issuecomment-2436123974](https://github.com/espressif/esp-idf/issues/14256#issuecomment-2436123974)

david-cermak, 2024-11-05:
> Great, ipv4 has been solved

Just out of my curiosity, would you mind sharing what was the cause of the instability on your side?

> How to enable ipv6 base on https://github.com/espressif/esp-idf/issues/14256#issuecomment-2436123974

I'll push one more commit with IPv6 support in a minute and let you know.

LinKvFM, 2024-11-05:
> Just out of my curiosity, would you mind sharing what was the cause of the instability on your side?

@david-cermak Because I used the chip internal clock as the MAC reference clock. Now I have changed to use an external clock and share the clock source with the PHY chip

david-cermak, 2024-11-08:
@LinKvFM Thanks for sharing :+1: 

I've added IPv6 support to the test project in https://github.com/espressif/esp-homekit-sdk/pull/143/commits/04e9f4d14836f03675e86b37e6a6ae8af0af3920

LinKvFM, 2024-12-04:
A small question, my ipv6 test report shows: `Debug run finished, run a full test to get a Bonjour Conformance test PASS.` But other products show: `You successfully passed the Bonjour Conformance test - mDNS IPV6`. 
Other than that, everything else is the same, including commands and test items.

My report：
[ipv6-2.zip](https://github.com/user-attachments/files/18003520/ipv6-2.zip)

other：
[IPv6.zip](https://github.com/user-attachments/files/18003524/IPv6.zip)

david-cermak, 2025-01-10:
> my ipv6 test report shows: Debug run finished, run a full test to get a Bonjour Conformance test PASS.

You usually get this report when you run BCT with limited number of tests.

david-cermak, 2025-01-10:
Closing as resolved as in both cases (@LinKvFM and @alikschen ) the issue was caused by incorrect CLK on LAN8720