espressif/esp-idf #14257: Ethernet Link Down--ESP32S3 with W5500 to establish an Ethernet connection

Chillimeat, 2024-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am currently using ESP32S3 with W5500 to establish an Ethernet connection, using version V5.3. The current issue I'm facing is that the network card repeatedly restarts and shuts down. Even when it starts up normally, it fails to communicate with other hosts

The log information is as follows:

I (301) main_task: Started on CPU0
I (311) main_task: Calling app_main()
I (311) eth_example: GPIO ISR service installed
I (311) eth_example: esp-netif initialized
I (321) eth_example: Event loop created
I (321) eth_example: SPI bus initialized
E (431) w5500.mac: W5500 version mismatched, expected 0x04, got 0xa0
E (431) w5500.mac: emac_w5500_init(826): verify chip ID failed
I (431) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (441) esp_eth: esp_eth_driver_install(228): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x10a (ESP_ERR_INVALID_VERSION) at 0x4200a752
0x4200a752: app_main at /Users/11/Documents/code/tools/w5-s3/main/main.c:205 (discriminator 1)

file: "./main/main.c" line 205
func: app_main
expression: esp_eth_driver_install(&config, &handle)

abort() was called at PC 0x4037b4ef on core 0
0x4037b4ef: _esp_error_check_failed at /Users/11/esp/v5.3/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40375b16:0x3fc9c670 0x4037b4f9:0x3fc9c690 0x40382c1d:0x3fc9c6b0 0x4037b4ef:0x3fc9c720 0x4200a752:0x3fc9c750 0x42035973:0x3fc9c880 0x4037bf8d:0x3fc9c8b0
0x40375b16: panic_abort at /Users/11/esp/v5.3/esp-idf/components/esp_system/panic.c:463
0x4037b4f9: esp_system_abort at /Users/11/esp/v5.3/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x40382c1d: abort at /Users/11/esp/v5.3/esp-idf/components/newlib/abort.c:38
0x4037b4ef: _esp_error_check_failed at /Users/11/esp/v5.3/esp-idf/components/esp_system/esp_err.c:49
0x4200a752: app_main at /Users/11/Documents/code/tools/w5-s3/main/main.c:205 (discriminator 1)
0x42035973: main_task at /Users/11/esp/v5.3/esp-idf/components/freertos/app_startup.c:208
0x4037bf8d: vPortTaskWrapper at /Users/11/esp/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: d27500b09

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a50
0x40375a50: esp_restart_noos at /Users/11/esp/v5.3/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (26) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (26) boot: compile time Jul 27 2024 17:30:30
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
[... 326 lines cut ...]

    // Create instance of esp-netif for Ethernet
    esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_ETH();
    esp_netif_config_t cfg = {
        .base = &esp_netif_config,
        .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH
    };
    eth_netif = esp_netif_new(&cfg);
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(handle)));
    ESP_LOGI(TAG, "esp-netif instance created");
    esp_netif_ip_info_t ip_info;
    IP4_ADDR(&ip_info.ip, 192, 168, 4, 101);
    IP4_ADDR(&ip_info.gw, 192, 168, 4, 1);
    IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);

    ESP_ERROR_CHECK(esp_netif_dhcpc_stop(eth_netif));
    ESP_ERROR_CHECK(esp_netif_set_ip_info(eth_netif, &ip_info));

    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));

    ESP_ERROR_CHECK(esp_eth_start(handle));
    ESP_LOGI(TAG, "Ethernet driver started");
}
```

Chillimeat, 2024-07-27:
The device’s logs indicate that the Ethernet link repeatedly goes up and down, as shown below:
E (1776391) w5500.mac: emac_w5500_transmit(578): free size (0) < send length (42)
I (1776391) eth_example: Ethernet Link Up
I (1776391) eth_example: Ethernet HW Addr 00:00:00:00:00:00
I (1776401) esp_netif_handlers: eth ip: 192.168.4.101, mask: 255.255.255.0, gw: 192.168.4.1
I (1776411) eth_example: Ethernet Got IP Address
I (1776411) eth_example: ~~~~~~~~~~~
I (1776411) eth_example: ETHIP:192.168.4.101
I (1776421) eth_example: ETHMASK:255.255.255.0
I (1776421) eth_example: ETHGW:192.168.4.1
I (1776431) eth_example: ~~~~~~~~~~~
I (1778391) eth_example: Ethernet Link Down
E (1808391) w5500.mac: emac_w5500_transmit(578): free size (0) < send length (42)
I (1808391) eth_example: Ethernet Link Up
I (1808391) eth_example: Ethernet HW Addr 00:00:00:00:00:00
I (1808401) esp_netif_handlers: eth ip: 192.168.4.101, mask: 255.255.255.0, gw: 192.168.4.1
I (1808411) eth_example: Ethernet Got IP Address
I (1808411) eth_example: ~~~~~~~~~~~
I (1808411) eth_example: ETHIP:192.168.4.101
I (1808421) eth_example: ETHMASK:255.255.255.0
I (1808421) eth_example: ETHGW:192.168.4.1
I (1808431) eth_example: ~~~~~~~~~~~
I (1810391) eth_example: Ethernet Link Down

DarmorGamz, 2024-07-29:
Please try esp_base_mac_addr_set() and esp_netif_set_mac() instead of esp_eth_ioctl. 

I experienced this on v5.0 and the symptom was "Ethernet HW Addr 00:00:00:00:00:00"

I also experienced "w5500.mac: emac_w5500_transmit(n): free size (0) < send length (k)" when my pins were incorrectly connected. You'll need to verify tx & rx are correctly connected and your clock speed.

Change your clock speed to ".clock_speed_hz = SPI_MASTER_FREQ_8M;"

Add ".flags = SPICOMMON_BUSFLAG_MASTER | SPICOMMON_BUSFLAG_SCLK | SPICOMMON_BUSFLAG_MISO | SPICOMMON_BUSFLAG_MOSI" to the spi_bus_config_t.
Change transfer speed to ".max_transfer_sz = 0".

Remove the line "phy_config.phy_addr = 1;"

Chillimeat, 2024-07-30:
@DarmorGamz 
Thank you for the help provided, but I found that I have been stuck in a loop of obtaining versions.
```
I (321) eth_example: GPIO ISR service installed
I (331) gpio: GPIO[43]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (541) eth_example: W5500 reset done
I (541) gpio: GPIO[41]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (541) eth_example: Interrupt GPIO configured
I (541) eth_example: SPI bus initialized
E (651) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
E (651) w5500.mac: emac_w5500_init(826): verify chip ID failed
I (651) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (661) esp_eth: esp_eth_driver_install(228): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x10a (ESP_ERR_INVALID_VERSION) at 0x4200a7e5
file: "./main/main.c" line 110
func: example_eth_init
expression: esp_eth_driver_install(&config, &handle)

abort() was called at PC 0x4037b50b on core 0

Backtrace: 0x40375b16:0x3fc9c600 0x4037b515:0x3fc9c620 0x40382c39:0x3fc9c640 0x4037b50b:0x3fc9c6b0 0x4200a7e5:0x3fc9c6e0 0x4200a923:0x3fc9c810 0x42035b2f:0x3fc9c880 0x4037bfa9:0x3fc9c8b0

ELF file SHA256: 15b08351e

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a50
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
```

kostaond, 2024-08-08:
> Please try esp_base_mac_addr_set() and esp_netif_set_mac() instead of esp_eth_ioctl.

@DarmorGamz Could you please elaborate what is the issue? This exact scenario is [tested in CI](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/test_apps/main/esp_eth_test_apps.c#L71) for all supported Ethernet SPI modules.

kostaond, 2024-08-08:
@Chillimeat how is your W5500 connected to ESP32? Do you have custom board or is it connected just via jumper wires? What W5500 board do you use?

Chillimeat, 2024-08-09:
@kostaond 
Thank you very much for your attention. I am using DuPont wires for connection, and I have attached the relevant pictures.
[image]

Chillimeat, 2024-08-09:
> @DarmorGamz Could you please elaborate what is the issue? This exact scenario is [tested in [...]

I also used a separate 5V power supply for power.

kostaond, 2024-08-09:
It seems you have some issues with SPI communication. Double check (or triple check :)) SPI configuration, that the connection is correct and use wires as short as possible (max 10cm) with matching length. You can also try to lower the SPI frequency to 8 MHz.

kostaond, 2024-09-04:
@Chillimeat any update? Have you been able to resolve the issue?

kostaond, 2025-01-06:
Closing the issue due to lack of feedback from reporter.

closed 2025-01-06 by kostaond

MAE-91, 2025-01-08:
I am facing the same issue . Have you found the root cause?

kostaond, 2025-01-09:
@MAE-91 could you please be more specific? How is your W5500 connected to the ESP32? If it's module, what module do you use? What is behavior you observe? What IDF version? etc...

MAE-91, 2025-01-09:
I am currently benchmarking to replace my existing Ethernet solution, which uses 10 pins, with an SPI Ethernet solution to free up more GPIOs. The example works when I use the ESP devkit, but it fails on my custom board that I manually reworked. I am using a WIZ850io module connected to a breadboard, and it is wired to my board with manual soldering.

the error : 
w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
w5500.mac: emac_w5500_init(795): vefiry chip ID failed
esp_eth: esp_eth_driver_install(228): init mac failed

kostaond, 2025-01-10:
Seems like hardware issue... Triple check that your SPI MOSI/MISO/SCLK configuration align with the hardware and signal is present at both source and sink pins. Try to decrease SPI frequency to e.g. 8 MHz. Make sure the WIZ850io is functional or use another piece if you have such option.

MAE-91, 2025-01-10:
I have a question. is it possible to use W5500 as a PHY only? if not , Do you know any PHY chip that works with esp32 with SPI ?

kostaond, 2025-01-13:
It's not possible to have just PHY connected to MAC over SPI. MAC and PHY communicate over MII/RMII interface (ESP32 supports both but RMII is preffered and tested). See https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_eth.html#configure-mac-and-phy for more information.