espressif/esp-idf #14259: V5.3 ESP_HID_HOST example not recognizing BT Keyboard multimedia keys

Hamberthm, 2024-07-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.3

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKit v1

### Power Supply used.

USB

### What is the expected behavior?

Using a Logitech K380 keyboard and the ESP_HID_HOST example project, a key report should be seen in the serial console for all keys, including multimedia keys. Like when pressing key 'a', followed by 's':

> I (231072) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: KEYBOARD, MAP:  0, ID:   1, Len: 8, Data:
I (231082) ESP_HIDH_DEMO: 00 00 04 00 00 00 00 00 
I (231322) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: KEYBOARD, MAP:  0, ID:   1, Len: 8, Data:
I (231332) ESP_HIDH_DEMO: 00 00 04 16 00 00 00 00 

### What is the actual behavior?

When pressing multimedia keys, like MUTE, VOLUME UP/DOWN, or PLAY, the following message is seen instead of the packet content:

> E (430732) BT_HIDH: Not find report handle: 0 mode: REPORT

### Steps to reproduce.

1. Compile ESP_HID_HOST example in VS Code
2. Upload to ESP32, open serial monitor console
3. Pair the keyboard (Logitech K380)
4. Press keys and specially multimedia keys on the keyboard

### Debug Logs.

```plain
I (231782) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: KEYBOARD, MAP:  0, ID:   1, Len: 8, Data:
I (231782) ESP_HIDH_DEMO: 00 00 04 00 00 00 00 00 
I (231782) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: KEYBOARD, MAP:  0, ID:   1, Len: 8, Data:
I (231792) ESP_HIDH_DEMO: 00 00 00 00 00 00 00 00 
E (430602) BT_HIDH: Not find report handle: 0 mode: REPORT
E (430732) BT_HIDH: Not find report handle: 0 mode: REPORT
```

### More Information.

A similar bug was present in V4.4 but for all keys being pressed (not only multimedia keys). The issue report for this bug ( #8318 ) can be found here: https://github.com/espressif/esp-idf/issues/8318

Hamberthm, 2024-07-28:
This is the report map reported to the Logitech K380:

>   VENDOR  OUTPUT REPORT, ID:  17, Length:  19
>   VENDOR   INPUT REPORT, ID:  17, Length:  19
>   VENDOR  OUTPUT REPORT, ID:  16, Length:   6
>   VENDOR   INPUT REPORT, ID:  16, Length:   6
>   CCONTROL   INPUT REPORT, ID: 255, Length:   1
>    GENERIC   INPUT REPORT, ID:   4, Length:   1
>   CCONTROL   INPUT REPORT, ID:   3, Length:   7
>   CCONTROL   INPUT REPORT, ID:   2, Length:   2
>   KEYBOARD  OUTPUT   BOOT, ID:   1, Length:   1
>   KEYBOARD  OUTPUT REPORT, ID:   1, Length:   1
>   KEYBOARD   INPUT   BOOT, ID:   1, Length:   8
>   KEYBOARD   INPUT REPORT, ID:   1, Length:   8

Problem seems to be at line:

https://github.com/espressif/esp-idf/blob/bdb9f972c6a1f1c5ca50b1be2e7211ec7c24e881/components/esp_hid/src/esp_hidh.c#L658

I added some debug function calls and discovered that when a media key is pressed, K380 sends a packet with length 4, and first data byte 2. That should correspond to map index `>   CCONTROL   INPUT REPORT, ID:   2, Length:   2`, but as quoted above the condition `r->value_len == len - 1` will be false because 2 does not equal to 3 (4-1).

So the problem is the keyboard is sending a data packet of 4 when it should be 3 to be valid. I don't know if it's the keyboard's fault or what.

boblane1, 2024-07-29:
@Hamberthm 

Thanks for your reporting and analysis. Maybe you can also report the issue to Logitech engineers. If it is the keyboard's fault, you can work around this issue by loose the check condition.

Hamberthm, 2024-07-29:
@boblane1 thanks for responding.

I can confirm that, by changing the line in `esp_hidh.c` to:

`            if (r->report_id == *data && (r->report_type & ESP_HID_REPORT_TYPE_INPUT) &&`

Thus eliminating the condition `r->value_len == len - 1`, the K380 multimedia keys are now being accepted as per the following log:

> BDA:f4:73:35:86:e2:5a, Status: OK, Connected: YES, Handle: 0, Usage: KEYBOARD
> Name: , Manufacturer: , Serial Number: 
> PID: 0xb342, VID: 0x046d, VERSION: 0x4201
> Report Map Length: 270
>     VENDOR  OUTPUT REPORT, ID:  17, Length:  19
>     VENDOR   INPUT REPORT, ID:  17, Length:  19
>     VENDOR  OUTPUT REPORT, ID:  16, Length:   6
>     VENDOR   INPUT REPORT, ID:  16, Length:   6
>   CCONTROL   INPUT REPORT, ID: 255, Length:   1
>    GENERIC   INPUT REPORT, ID:   4, Length:   1
>   CCONTROL   INPUT REPORT, ID:   3, Length:   7
>   CCONTROL   INPUT REPORT, ID:   2, Length:   2
>   KEYBOARD  OUTPUT   BOOT, ID:   1, Length:   1
>   KEYBOARD  OUTPUT REPORT, ID:   1, Length:   1
>   KEYBOARD   INPUT   BOOT, ID:   1, Length:   8
>   KEYBOARD   INPUT REPORT, ID:   1, Length:   8
> W (13173) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 20, status 0x0
> I (13173) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:2
> I (16693) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP:  0, ID:   2, Len: 3, Data:
> I (16703) ESP_HIDH_DEMO: 01 00 00 
> I (16783) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP:  0, ID:   2, Len: 3, Data:
> I (16783) ESP_HIDH_DEMO: 00 00 00 
> I (16983) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP:  0, ID:   2, Len: 3, Data:
> I (16983) ESP_HIDH_DEMO: 20 00 00 
> I (17363) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP:  0, ID:   2, Len: 3, Data:
> I (17363) ESP_HIDH_DEMO: 00 00 00 
> I (17393) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP:  0, ID:   2, Len: 3, Data:
> I (17393) ESP_HIDH_DEMO: 40 00 00 
> I (17513) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP:  0, ID:   2, Len: 3, Data:
> I (17513) ESP_HIDH_DEMO: 00 00 00 

As you can see, the packet length is 3, but instead the keyboard is reporting it should be 2 as in the REPORT MAP.

I can confirm that pressing various multimedia keys at once will never populate the third byte, only the first two:

> I (442333) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP:  0, ID:   2, Len: 3, Data:
> I (442333) ESP_HIDH_DEMO: 80 03 00 

Then the third byte can be discarded and the REPORT MAP info is valid.

So, the question is, is it really necessary to check the report length when the keyboard is giving a report ID like in this case?

The multimedia keys work as they should in other OSs like Windows so I guess we should take the report MAP info and just ignore the third byte being sent.

boblane1, 2024-07-30:
@Hamberthm Thanks for your suggestion. Maybe we can change the condition to `r->value_len <= len - 1`. I think the input data length which is less than the length MAP info should be ignored.

Hamberthm, 2024-08-06:
> @Hamberthm Thanks for your suggestion. Maybe we can change the condition to `r->value_len <= len - 1`. I think the input data length which is less than the length MAP info should be ignored.

Hi @boblane1 , sorry for the late response.

I've testes your proposed solution and I can confirm it works to solve the issue. I think it's a nice workaround without loosing the check condition too much.

Final line would be as follows:

`            if (r->value_len <= len - 1 && r->report_id == *data && (r->report_type & ESP_HID_REPORT_TYPE_INPUT) &&`

I'm kind of new on Github, are you going to pull a request or should I do anything more?

Thank you so much

boblane1, 2024-08-07:
@Hamberthm 

I will fix this issue, and thanks for your reporting again.

LogiBrianLee, 2024-08-08:
@Hamberthm 

The K380 uses the bit map report descriptor, and the consumer(media) report uses the three bytes bit map. The bit map is a format which can define in the HID descriptor. As it is not the normal usage key code, you may need to change your condition checking if you want to get the report data.
Hope this can answer your question.

Hamberthm, 2024-08-08:
> [LogiBrianLee](/LogiBrianLee)

@LogiBrianLee hi! Thanks for coming by. I'm sorry but it seems you're not fully understanding what's the problem here.

I'm fully aware that the report the K380 sends is a bitmap or "Data, Variable" as the HID specification calls it. I even wrote a custom parser for it using the Report Descriptor.

The problem here, is that the keyboard declares a report size of 16 bits, as declared in the Report Descriptor the keyboard itself sends:

```
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x0D,        //   Report Count (13)
0x0A, 0x24, 0x02,  //   Usage (AC Back)
0x09, 0x40,        //   Usage (Menu)
0x0A, 0x23, 0x02,  //   Usage (AC Home)
0x0A, 0xAE, 0x01,  //   Usage (AL Keyboard Layout)
0x0A, 0x21, 0x02,  //   Usage (AC Search)
0x09, 0xB6,        //   Usage (Scan Previous Track)
0x09, 0xCD,        //   Usage (Play/Pause)
0x09, 0xB5,        //   Usage (Scan Next Track)
0x09, 0xE2,        //   Usage (Mute)
0x09, 0xEA,        //   Usage (Volume Decrement)
0x09, 0xE9,        //   Usage (Volume Increment)
0x09, 0x30,        //   Usage (Power)
0x09, 0x40,        //   Usage (Menu)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x03,        //   Report Size (3)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
```

Here you can see, the keyboard declares a report final size of 16 bits (13 Data + 3 Const bits).

The problem is, the keyboard ends up sending a 3 BYTE REPORT! (Thus, not compiling with what was declared in the Report Descriptor), and this is a HID specification violation!

You can see the keyboard sends a 3 byte report in the dump done by the ESP-IDF example project ESP_HID_DEMO:

```
I (442333) ESP_HIDH_DEMO: f4:73:35:86:e2:5a INPUT: CCONTROL, MAP: 0, ID: 2, Len: 3, Data:
I (442333) ESP_HIDH_DEMO: 80 03 00
```

Thats an example CCONTROL key pressed, and you can see the keyboard sends a trash THIRD BYTE, when it should be sending only TWO as the report descriptor says.

closed 2024-08-14 by espressif-bot