espressif/esp-idf #14260: Unit test example hangs after unity_run_menu() call

teasin951, 2024-07-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

nanoESP32-C6 v1.0

### Power Supply used.

USB

### What is the expected behavior?

After following the steps outlined in _README.md -> Build and flash -> 2._, the _Example output -> Test project output_, would be expected.

### What is the actual behavior?

The devices flashes, but monitoring does not start. After manually opening the monitoring, it prints out the test headers but does not perform any tests. Pressing Enter afterwards does not bring up menu, only prints: "Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication."

### Steps to reproduce.

1. Using the _New project wizard_ in ESP-IDF VSCodium extension create a new project with _unit_test_ template
2. Using the ESP-IDF terminal navigate to the _test_  directory and run `idf.py set-target ESP32C6` and `idf.py -p /dev/ttyACM0 flash monitor`
3. As the monitor does not open, run `ESP-IDF: Monitor device`

### Debug Logs.

```plain
--- esp-idf-monitor 1.4.0 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (134) boot: Loaded app from partition at offset 0x10000ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400294a6
0x400294a6: uart_hal_get_txfifo_count in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c110,len:0xe2c
load:0x4086e610,len:0x2e70
entry 0x4086c11a
I (23) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (23) boot: compile time Jul 28 2024 17:43:23
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=42018020 size=08808h ( 34824) map
I (99) esp_image: segment 1: paddr=00018830 vaddr=40800000 size=077e8h ( 30696) load
I (107) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=10c00h ( 68608) map
I (123) esp_image: segment 3: paddr=00030c28 vaddr=408077e8 size=03180h ( 12672) load
I (126) esp_image: segment 4: paddr=00033db0 vaddr=4080a970 size=01824h (  6180) load
I (133) boot: Loaded app from partition at offset 0x10000
I (134) boot: Disabling RNG early entropy source...
I (151) cpu_start: Unicore app
I (160) cpu_start: Pro cpu start user code
I (160) cpu_start: cpu freq: 160000000 Hz
I (161) app_init: Application information:
I (163) app_init: Project name:     unit_test_test
I (169) app_init: App version:      1
I (173) app_init: Compile time:     Jul 28 2024 17:43:19
I (179) app_init: ELF file SHA256:  6edde5a9c...
I (184) app_init: ESP-IDF:          v5.3-dirty
I (189) efuse_init: Min chip rev:     v0.0
I (194) efuse_init: Max chip rev:     v0.99 
I (199) efuse_init: Chip rev:         v0.0
I (204) heap_init: Initializing. RAM available for dynamic allocation:
I (211) heap_init: At 4080D270 len 0006F3A0 (444 KiB): RAM
I (217) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (223) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (230) spi_flash: detected chip: generic
I (234) spi_flash: flash io: dio
W (238) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (252) sleep: Configure to isolate all GPIO pins in sleep state
I (258) sleep: Enable automatic switching of GPIO sleep configuration
I (265) coexist: coex firmware version: dab85ae96
I (271) coexist: coexist rom version 5b8dcfa
I (276) main_task: Started on CPU0
I (276) main_task: Calling app_main()

#### Executing one test by its name #####

#### Running tests with [mean] tag #####

#### Running tests without [fails] tag #####

#### Running all the registered tests #####

#### Starting interactive test menu #####

(...)

Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
```

### More Information.

I have originally encountered this in my own project when trying to create some unit tests using ESP-IDF's extension command `ESP-IDF: Unit Test: Build and flash unit test app for testing`. After that I tried it with `idf.py` to no avail. The program seems to hang at unity_run_menu(). 

I have tried this on Linux with ESP-IDF v5.3.0, ESP-IDF v5.2.2 and on Windows with ESP-IDF v5.3.0, everything produced the same result. The main application works as expected, as does e.g. _blink_ example.

igrr, 2024-07-28:
> Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.

It looks like you might be using the USB port connected to USB_SERIAL_JTAG peripheral. By default, the application is configured to use UART port, instead. Please try enabling CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG option and see if that helps.

teasin951, 2024-07-29:
This indeed fixed it. Thank you very much for the swift response!

closed 2024-07-29 by teasin951