espressif/esp-idf #14261: ESP32 RMII Ethernet: emac_esp32_transmit(233): insufficient TX buffer size

imfatant, 2024-07-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

08:f9:e0:a7:9e:c8

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

WT32-ETH01

### Power Supply used.

External 3.3V

### What is the expected behavior?

Would it be reasonable to expect that the ESP32's RMII interface is capable of supporting a considerably higher bandwidth than its SPI host running at 33 MHz? Please correct me if I'm wrong.

So here's what's happening: When I'm watching a 1080p YouTube video stream on a laptop which is being supplied by an ESP32 using its RMII Ethernet + on-board LAN8720 chip (it's a WT32-ETH01 board) I'm getting considerable buffering, to the point that it's pretty much unwatchable. In addition to this, I'm getting repeated error messages over the serial port:
```
E (1493545) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1493545) example_wired_ethernet: Ethernet send packet failed
```

When I do the same, but now use an ESP32 with an external W5500 module over SPI running at 33 MHz, the video stream is dead smooth. And I get no insufficient TX buffer size error messages whatsoever.

I realise this issue has been raised before, both here (https://github.com/espressif/esp-idf/issues/13602) and here (https://esp32.com/viewtopic.php?t=24595) but could you please explain what might be going on?

### What is the actual behavior?

See above (I have combined these two fields.)

### Steps to reproduce.

Use the ESP-IDF v5.2.2. network/statoeth example.

### Debug Logs.

```plain
E (1586815) example_wired_ethernet: Ethernet send packet failed
E (1586825) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586835) example_wired_ethernet: Ethernet send packet failed
E (1586835) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586845) example_wired_ethernet: Ethernet send packet failed
E (1586855) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586865) example_wired_ethernet: Ethernet send packet failed
E (1586865) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586875) example_wired_ethernet: Ethernet send packet failed
E (1586885) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586885) example_wired_ethernet: Ethernet send packet failed
E (1586895) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586905) example_wired_ethernet: Ethernet send packet failed
E (1586915) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586915) example_wired_ethernet: Ethernet send packet failed
E (1586925) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
E (1586935) example_wired_ethernet: Ethernet send packet failed
E (1586935) esp.emac: emac_esp32_transmit(233): insufficient TX buffer size
```

kostaond, 2024-08-08:
It indicates that more data is tried to be transmitted over Ethernet than the hardware is able to handle. It doesn't necessarily mean that you utilized full 100mbps because there is limited number of Tx buffers (10 by default). So, if the frames are short but you queue more than 10 at a time, you get this error. Printing the error message consumes additional HW resources and bandwidth is even worst.

There are the following steps to try:
1) Analyze your traffic and properly configure Tx buffers. For example, if your app utilize mostly short frames, decrease size of the buffer but increase the number of the buffers to keep app's memory footprint same. If you don't care, just increase number of TX buffers. You can do it in Kconfig.
2) Just for check, try to use `master` version of IDF where `insufficient TX buffer size` is not printed (only `ESP_ERR_NO_MEM` is returned) to see if you get better performance.

You can also try to play with `iperf` and configure different packet lengths and different bandwidth limit options to understand the behavior.

KaeLL, 2024-08-08:
Should consider adding that to the docs, as it's a somewhat fairly common issue.

kostaond, 2024-08-14:
> Should consider adding that to the docs, as it's a somewhat fairly common issue.

That's actually very good idea since this is really not the first question on this topic.

Tarik2142, 2024-12-24:
I can confirm this. This occurs after a while and will cause the device to slow down and then completely freeze. This is a bug in the ethernet implementation. This problem never occurs with WiFi

kostaond, 2025-01-02:
> I can confirm this. This occurs after a while and will cause the device to slow down and then completely freeze. This is a bug in the ethernet implementation. This problem never occurs with WiFi

`insufficient TX buffer size` is not a bug. It's expected hardware limitation. Could you please provide more test scenario under which device "completely freezes"?

closed 2025-01-29 by kostaond