espressif/esp-idf #14262: Service changes in run time are not getting refreshed in ios BLE clients even upon implementing the refresh characteristics in server

BestBot7, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have implemented the refresh characteristics with service UUID 0x1801 and characteristic UUID 0x2a05 which are meant for indicating the service changes in the server to the ios BLE clients. I have implemented this on esp32c3. In nrf connect mobile app for android, we have a feature to refresh services in the run time with keeping the connection. But ios requires an indication to refresh the services and detect service change. I am not receiving any indication in my mobile app. Please help me in implementing this refresh characteristics properly and how to check the indication value on my ios mobile. Thanks in advance.

esp-zhp, 2024-08-01:
Are you using the NimBLE host or the Bluedroid host?

In the Bluedroid host, GAP and GATT service are already implemented, and the Service Change Characteristic is part of the GATT service.

[image]

When using the Service Change Characteristic, the mobile device needs to first write to the Client Characteristic Configuration of the server.

When the server adds or removes a service, it will notify or indicate the client, depending on what value was written to the Client Characteristic Configuration.

It's important to note that if the server adds or removes a characteristic, this will not trigger a notification or indication from the server.

ps:
BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 3, Part G page 1523

[image]
[image]

esp-zhp, 2024-08-01:
Additionally, you can proactively send a Service Change notification by using the API esp_ble_gatts_send_service_change_indication. (https://github.espressif.cn/espressif/esp-idf/-/blob/be47da19628967449635c9042306b861a2fe3ed8/components/bt/host/bluedroid/api/include/api/esp_gatts_api.h#L584)

BestBot7, 2024-08-07:
@esp-zhp, thanks for the reply! I am using nimBLE stack to implement this feature. I found that there is no direct implementation for runtime service changes. Can I know what should I do to indicate the client that services have changed. Also if the client does not trigger on notification or indication, how does it know to refresh and re-discover the services? Can you explain me in detail how should I achieve this in nimBLE. There is also a config setting called "Enable dynamic services", what is it used for, any idea? I enabled that and there is no difference. The client does not get any indication or refresh the services in runtime.

rahult-github, 2024-08-07:
Hi @BestBot7 , 

Apart from information shared on https://github.com/espressif/esp-nimble/issues/72  and other details already mentioned by Zhang Haipeng here, 

> Also if the client does not trigger on notification or indication, how does it know to refresh and re-discover the services? 

Client subscribes to server. When changes occur on server side, server will send notification to client and client then comes to know about some change on server side and then it decides to do service search or any other operation it has functionality for. 

 > There is also a config setting called "Enable dynamic services", what is it used for 
 
 This feature is related to adding / removing services runtime. You can run [this](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_dynamic_service) example for more details about this.  
 
  > The client does not get any indication or refresh the services in runtime. 
  
  Tried with nrfconnect app as scanner . Need to connect with the example, and subscribe. Then example will send indication periodically to inform of service change which causes nrfconnect app to discover service again. 
  
  
[Log 2024-08-07 15_52_12.txt](https://github.com/user-attachments/files/16521960/Log.2024-08-07.15_52_12.txt)

rahult-github, 2024-10-23:
Closing , based on inputs given here and discussion on esp forum.

closed 2024-10-23 by rahult-github