espressif/esp-idf #14263: Heap corruption after wifi_prov_mgr_deinit (in esp_bt_mem_release)

eug-ev, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

Chip is ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

USB

### What is the expected behavior?

I expect bluetooth provisioning library to init and deinit correctly

### What is the actual behavior?

I added support for BLE provisioning in the same manner as in wifi_prov_mgr
I suffered from heap corruption in wifi_prov_mgr_deinit.
Later I decided to open example project wifi_prov_mgr and it has the same problem.

### Steps to reproduce.

0. Create new wifi_prov_mgr example project
1. Enable debug log verbocity
2. Enable Comprehensive heap poisoning
3. Enable BLE provisioning
4. Add wifi_prov_mgr_deinit after provisioning completed
5. modify esp-idf/components/bt/controller/esp32/bt.c to highlight the problem [bt.c.txt](https://github.com/user-attachments/files/16407749/bt.c.txt) . Look for CHECK_HEAP macro in the file
6. run and watch for logs

### Debug Logs.

```plain
I (98424) app: wifi_prov_mgr_deinit
D (98424) wifi_prov_mgr: Provisioning not running
D (98434) wifi_prov_mgr: execute_event_cb : 6
D (98484) wifi_prov_mgr: check heap func wifi_prov_mgr_deinit line 1482
D (98534) wifi_prov_mgr: check heap func wifi_prov_mgr_deinit line 1482
D (98584) wifi_prov_mgr: check heap func wifi_prov_mgr_deinit line 1482
D (98634) wifi_prov_scheme_ble: check heap func wifi_prov_scheme_ble_event_cb_free_btdm line 213
D (98684) wifi_prov_scheme_ble: check heap func wifi_prov_scheme_ble_event_cb_free_btdm line 213
D (98734) wifi_prov_scheme_ble: check heap func wifi_prov_scheme_ble_event_cb_free_btdm line 213
D (98784) wifi_prov_scheme_ble: check heap func wifi_prov_scheme_ble_event_cb_free_btdm line 234
D (98834) wifi_prov_scheme_ble: check heap func wifi_prov_scheme_ble_event_cb_free_btdm line 234
D (98884) wifi_prov_scheme_ble: check heap func wifi_prov_scheme_ble_event_cb_free_btdm line 234
D (98884) BTDM_INIT: check heap func esp_bt_mem_release line 1452
[... 166 lines cut ...]
```
               0x3ffc0e39                        _coredump_dram_start = ABSOLUTE (.)
 *(.dram2.coredump .dram2.coredump.*)
                0x3ffc0e39                        _coredump_dram_end = ABSOLUTE (.)
 *libapp_trace.a:app_trace.*(.rodata .rodata.* .sdata2 .sdata2.* .srodata .srodata.*)
 *libapp_trace.a:app_trace_util.*(.rodata .rodata.* .sdata2 .sdata2.* .srodata .srodata.*)
 *libapp_trace.a:port_uart.*(.rodata .rodata.* .sdata2 .sdata2.* .srodata .srodata.*)
                0x3ffc0e3c                        . = ALIGN (0x4)
 *fill*         0x3ffc0e39        0x3 
                0x3ffc0e3c                        _bt_data_start = ABSOLUTE (.)
 *libbt.a:(.data .data.*)
 .data          0x3ffc0e3c     0x111c esp-idf/bt/libbt.a(hli_vectors.S.obj)
 .data.btdm_dram_available_region
                0x3ffc1f58       0x54 esp-idf/bt/libbt.a(bt.c.obj)
 .data.ble_svc_gap_name
                0x3ffc1fac       0x20 esp-idf/bt/libbt.a(ble_svc_gap.c.obj)
 .data.ble_hs_cfg
                0x3ffc1fcc       0x28 esp-idf/bt/libbt.a(ble_hs_cfg.c.obj)
                0x3ffc1fcc                ble_hs_cfg
 .data.ble_uuid_base
                0x3ffc1ff4       0x10 esp-idf/bt/libbt.a(ble_uuid.c.obj)
 .data.__compound_literal$6
                0x3ffc2004        0x4 esp-idf/bt/libbt.a(ble_gatts.c.obj)
 .data.__compound_literal$5
```

eug-ev, 2024-07-29:
It seems like it was introduced recently in https://github.com/espressif/esp-idf/commit/1d9559cc9dfc2d329eab446f82aca67b72a4eafa

wmy-espressif, 2024-07-30:
Hi @eug-ev thanks for reporting this issue. I am able to reproduce it and will provide you my feedback after further debugging.

wmy-espressif, 2024-07-30:
Hi @eug-ev , as you have pointed out, there is a bug in `esp_bt_mem_release` function on ESP32.

- Cause of the issue:

In function `esp_bt_mem_release`, it first releases the memory for Bluetooth controller, Bluetooth host and .data/.bss sections of global variables in bt.c, and then calls `esp_bt_controller_rom_mem_release` to release the memory reserved for Bluetooth ROM code.

However,  function `esp_bt_controller_rom_mem_release`  uses global struct `btdm_dram_available_region`, `btdm_dram_available_region` should be released in earlier steps. Thus invalid memory region is observed during comprehensive heap memory check.

- Proposed solution

Adjust the memory release order. First releases memory reserved for ROM code, then releases code for other regions. I have made some changes on bt.c based on your debugging version. Please refer to
[bt.c.txt](https://github.com/user-attachments/files/16424174/bt.c.txt)

wmy-espressif, 2024-07-30:
If you agree with the proposed solution, I will raise bugfix merge requests targeted on master branch and other release branches of esp-idf

eug-ev, 2024-07-30:
Im actually not an owner and I would be glad for solution merge, thanks!

closed 2024-08-14 by espressif-bot

eug-ev, 2024-08-14:
Thanks for the fix!
Another question - I saw that in some other bt.c files rom memory release is still after areas. 
I don't have hardware to test, just pointing if it could cause the same issue

wmy-espressif, 2024-08-15:
Hi @eug-ev Thanks for your reminder. For other chips like ESP32-C3, there is no such variable `btdm_dram_available_region`.  `esp_bt_controller_rom_mem_release` is safe to be executed after the other Bluetooth .bss/.data areas are released.

I am backporting the bugfix for this issue for other release branches, including release/v5.0, release/v5.1, release/v5.2, and release/v5.3. I will provide you information after all the backports are finished.