espressif/esp-idf #14264: Set command handler with context and without context

zlaazlaa, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I saw there is a guide say that "You can either use a command handler function which takes a context or a command handler function which does not take a context, not both." in [docs/en/api-reference/system/console.rst](https://github.com/espressif/esp-idf/blob/bdb9f972c6a1f1c5ca50b1be2e7211ec7c24e881/docs/en/api-reference/system/console.rst).
But in [components/console/commands.c](https://github.com/espressif/esp-idf/blob/master/components/console/commands.c) It seems like both are being executed.
```c
    if (cmd->func) {
        *cmd_ret = (*cmd->func)(argc, argv);
    }
    if (cmd->func_w_context) {
        *cmd_ret = (*cmd->func_w_context)(cmd->context, argc, argv);
    }
```
If we set two handlers while registering command, there may be a error. 
Is that a bug or something else? If it's a bug, I'd like to open a PR to fix it.

igrr, 2024-07-29:
I guess this line
https://github.com/espressif/esp-idf/blob/bdb9f972c6a1f1c5ca50b1be2e7211ec7c24e881/components/console/commands.c#L124
ensures that only one of the callbacks is actually registered.

zlaazlaa, 2024-07-29:
> I guess this line
> 
> https://github.com/espressif/esp-idf/blob/bdb9f972c6a1f1c5ca50b1be2e7211ec7c24e881/components/console/commands.c#L124
> 
> 
> ensures that only one of the callbacks is actually registered.

So the insurance is here, XD. Thanks for your answer, I'll close this issue.👍

closed 2024-07-29 by zlaazlaa