espressif/esp-idf #14265: error: invalid conversion from 'int' to 'emac_rmii_clock_gpio_t'

stkw0, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3 

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It builds correctly

### What is the actual behavior?

> /usr/share/esp-idf/components/esp_eth/include/esp_eth_mac_esp.h:205:35: error: invalid conversion from 'int' to 'emac_rmii_clock_gpio_t' [-fpermissive]
>   205 |     #define DEFAULT_RMII_CLK_GPIO CONFIG_ETH_RMII_CLK_IN_GPIO
>       |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |                                   |
>       |                                   int
> /usr/share/esp-idf/components/esp_eth/include/esp_eth_mac_esp.h:233:31: note: in expansion of macro 'DEFAULT_RMII_CLK_GPIO'
>   233 |                 .clock_gpio = DEFAULT_RMII_CLK_GPIO   \
>       |                               ^~~~~~~~~~~~~~~~~~~~~
> /home/droman/Develop/Work/Virgo/bsda/shared/common/fw/eth.cpp:139:49: note: in expansion of macro 'ETH_ESP32_EMAC_DEFAULT_CONFIG'
>   139 |     eth_esp32_emac_config_t esp32_emac_config = ETH_ESP32_EMAC_DEFAULT_CONFIG();
>       |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /home/droman/Develop/Work/Virgo/bsda/shared/common/fw/eth.cpp:140:23: warning: 'eth_esp32_emac_config_t::<unnamed union>::<unnamed struct>::smi_mdc_gpio_num' is deprecated: Please use smi_gpio instead [-Wdeprecated-declarations]
>   140 |     esp32_emac_config.smi_mdc_gpio_num = 23;

### Steps to reproduce.

Build a small project with the next C++ file:

```cpp
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_eth.h>

void app_main(void)
{
    eth_esp32_emac_config_t esp32_emac_config = ETH_ESP32_EMAC_DEFAULT_CONFIG();
}
```

closed 2024-08-14 by espressif-bot

BartVanHofwegen, 2024-09-15:
Issue is still present in 5.3.1

kostaond, 2024-09-16:
> Issue is still present in 5.3.1

It always be present in 5.3.1 since it didn't make it to this release... However, it's already there in v5.3 branch and it's waiting for next release https://github.com/espressif/esp-idf/commit/e9df36a2dfe9ee074f421aedaff8952a348b339d