espressif/esp-idf #14266: esp32 v5.3 httpd server reinitialization memory leak

LjubiTech-Maxko, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-D0WD (revision v3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Free memory should stay the same after each reinitialization, as we have observed in the previous version v5.2.1.

### What is the actual behavior?

There is constant drop of free memory after each reinitialization of the server.

### Steps to reproduce.

1. Create a new ESP-IDF project and paste the following code:

```C
#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "esp_event.h"
#include "esp_netif.h"
#include "esp_log.h"
#include "esp_http_server.h"
#include "esp_heap_trace.h"

#define NUM_RECORDS 200

static const char *TAG = "TEST";
static httpd_handle_t http_update_server_handle;
static heap_trace_record_t trace_record[NUM_RECORDS];

static esp_err_t http_server_init()
{
    ESP_LOGI(TAG, "%s: free mem before init: internal(%"PRIu32") - all(%"PRIu32")", __func__, esp_get_free_internal_heap_size(), esp_get_free_heap_size());
    httpd_config_t update_server_config = HTTPD_DEFAULT_CONFIG();
    update_server_config.stack_size = 16348;
[... 305 lines cut ...]

    16 bytes (@ 0x3f8014e0,    PSRAM) allocated CPU 0 ccount 0xe19a92e0 caller 0x40082342:0x400e0fb2:0x400e103a:0x400e1076:0x400e7198:0x400e720b:0x400e0fb2: mem_malloc at C:/esp-idf/components/lwip/lwip/src/core/mem.c:209
0x400e103a: do_memp_malloc_pool at C:/esp-idf/components/lwip/lwip/src/core/memp.c:254
0x400e1076: memp_malloc at C:/esp-idf/components/lwip/lwip/src/core/memp.c:350
0x400e7198: sys_timeout_abs at C:/esp-idf/components/lwip/lwip/src/core/timeouts.c:193
0x400e720b: lwip_cyclic_timer at C:/esp-idf/components/lwip/lwip/src/core/timeouts.c:265

0x400e73c4:0x400e0cb3:0x400886b0:
0x400e73c4: sys_check_timeouts at C:/esp-idf/components/lwip/lwip/src/core/timeouts.c:403
0x400e0cb3: tcpip_timeouts_mbox_fetch at C:/esp-idf/components/lwip/lwip/src/api/tcpip.c:100
 (inlined by) tcpip_thread at C:/esp-idf/components/lwip/lwip/src/api/tcpip.c:142
0x400886b0: vPortTaskWrapper at C:/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:136

====== Heap Trace Summary ======
Mode: Heap Trace Leaks
50112 bytes 'leaked' in trace (9 allocations)
records: 9 (200 capacity, 34 high water mark)
total allocations: 135
total frees: 135
================================
```

### More Information.

We have also added heap tracking to check for leaks, and see that there are leaks with v5.3. We have noticed the difference in `httpd_os_thread_create(...)` function which now has added memory caps for task creation, and we can see from the logs that the memory allocated for task stack was never released.

LjubiTech-Maxko, 2024-07-30:
Closing as we have seen that the memory leak is caused by `xTaskCreatePinnedToCoreWithCaps(...)` function which is used in `httpd_os_thread_create(...)`, and there is already closed issue #14222 which is related to that, with already available solution in https://github.com/espressif/esp-idf/commit/c3da2ace276be0a42c83c97535c701ea10a239a8

LjubiTech-Maxko, 2024-07-30:
Reopening since we have tested the mentioned solution and it doesn't work for http_server.
`assert failed: vPortAssertIfInISR port.c:452 (xPortInIsrContext())`

sudeep-mohanty, 2024-07-30:
HI @LjubiTech-Maxko,
The assert failure is a bug and has been fixed in the master branch - Ref: 7806aeb3728e7381d5cfe6e64275cb8cea09b1c1
The fix will also be backported to other release branches include v5.3 soon.

LjubiTech-Maxko, 2024-07-30:
Hi @sudeep-mohanty 
Thanks for the info.
We have tested again with both of the commits and can confirm that the httpd_server is not leaking memory anymore.
There are some reported leaks with the heap tracing but we assume those are not really leaks since there are always 3 reported no matter how many times we run the reinitialization loop. Posting 1 of them since all 3 are the same just for reference.

```C
    16 bytes (@ 0x3f8014bc,    PSRAM) allocated CPU 0 ccount 0x55d5dad0 caller 0x400824ca:0x400e3e96:0x400e3f0e:0x400e3f71:0x400ea9b4:0x400eaa45:0x400eab6c:0x400e3b3d:0x400e3bf9:0x4008b670:
0x400824ca: heap_caps_malloc_prefer at C:/esp-idf/components/heap/heap_caps.c:180
0x400e3e96: mem_malloc at C:/esp-idf/components/lwip/lwip/src/core/mem.c:209
0x400e3f0e: do_memp_malloc_pool at C:/esp-idf/components/lwip/lwip/src/core/memp.c:254
0x400e3f71: memp_malloc at C:/esp-idf/components/lwip/lwip/src/core/memp.c:350
0x400ea9b4: sys_timeout_abs at C:/esp-idf/components/lwip/lwip/src/core/timeouts.c:193
0x400eaa45: lwip_cyclic_timer at C:/esp-idf/components/lwip/lwip/src/core/timeouts.c:265
0x400eab6c: sys_check_timeouts at C:/esp-idf/components/lwip/lwip/src/core/timeouts.c:403
0x400e3b3d: tcpip_timeouts_mbox_fetch at C:/esp-idf/components/lwip/lwip/src/api/tcpip.c:111
0x400e3bf9: tcpip_thread at C:/esp-idf/components/lwip/lwip/src/api/tcpip.c:143
0x4008b670: vPortTaskWrapper at C:/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:136
```

Closing since initial issue si resolved in the mentioned commits.

closed 2024-07-30 by LjubiTech-Maxko