espressif/esp-idf #14267: newlib build paths included, causing loss of build reproducibility across build OSs

bryghtlabs-richard, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

When building same application with same sources and same tool versions, output binaries should match, even if the build OS is different.

### What is the actual behavior?

Some paths seem to be captured when building newlib, which will then be present in the application's .bin file.

### Steps to reproduce.

1. Enable CONFIG_APP_REPRODUCIBLE_BUILD
2. Confirm tool versions match across operating systems
3. Build same application on Windows 11 and WSL2 with `idf.py build`
4. Compare app .bin files from both builds
5. Identify most common difference: newlib build path

### More Information.

I saw the documentation stated "Versions of the build tools (CMake, Ninja) and the cross-compiler" may cause differences, but this seems unrelated to that. Here is the first difference in HXD:

[image]

bryghtlabs-richard, 2024-07-29:
No worries about a fix for v5.1.x, it's just the version I was using. Remapping those paths would be certainly be helpful, and may be the only thing left across different OSs with the same tool versions - I've only seen differences in those strings and differences in pointers that may have been offset due to those strings changing length.

closed 2024-09-14 by espressif-bot

bryghtlabs-richard, 2024-09-16:
I do not think this is complete, please re-evaluate.

igrr, 2024-09-16:
@bryghtlabs-richard The paths seem to be `/builds/idf/crosstool-NG/builds/` for all platforms now, could you please clarify, what is the issue you are seeing?

bryghtlabs-richard, 2024-09-16:
Sorry, I saw the Aug 15 update and misunderstood, I didn't see the update two days ago. Setting the path like that should fix it.