espressif/esp-idf #14268: Overwrite SDKConfig variable in cmakelist,txt

ioviedodev, 2024-07-29:
### Describe the solution you'd like.

It would be great in Cmakelist file could rewrite a SDKconfig variable, for example CONFIG_PARTITION_TABLE_CUSTOM_FILENAME.

SoucheSouche, 2024-07-30:
Hi @ioviedodev,

You can use sdkconfig.defaults file to place the config options you wish to modify from their default values
See the documentation [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html).

The sdkconfig file is used to generate `sdkconfig.h` containing macros of the config options. Trying to change their values in CMakelists would mean redefining the macros.

ioviedodev, 2024-07-30:
Hi @SoucheSouche , thank you for your reply so I have the following scenario.

I create a kconfig variable
CONFIG_OPTION // could have 0,1
This option i could see in Sdkconfig

and when I choose 
if OPTION=0
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME= "path1"
else 
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME= "path2"

SoucheSouche, 2024-07-31:
In that case I would just create 2 specific sdkconfig.defaults files (sdkconfig.defaults.option_0 and sdkconfig.defaults.option_1)
And then apply one or the other to the sdkconfig file before building the application.

DarmorGamz, 2024-08-02:
@ioviedodev 

Add the file Kconfig.projbuild to the same scope as the file with app_main().
```
menu "Ioviedodev App Configuration"

	choice BUILD_TYPE
		prompt "Build Type"
		default BUILD_DEVELOPMENT

        config BUILD_DEVELOPMENT
            bool "Development"
        config BUILD_PRODUCTION
            bool "Production"
	endchoice

endmenu
```

In the root CMakeLists.txt add the below before calling project(whatever_app_name)
```
find_package(Python3 REQUIRED COMPONENTS Interpreter)
set(python ${Python3_EXECUTABLE})
execute_process(COMMAND ${python} "${CMAKE_CURRENT_SOURCE_DIR}/.tools/prebuild.py")
```

Create a directory in root ./tools and create the file prebuild.py.

In prebuild.py:
```
import sys
import os

def copy_and_replace(src_file):
    with open(src_file, 'r') as file:
        data = file.read()
    with open("../sdkconfig.defaults", 'w') as file:
        file.write(data)

def isDev():
	if not os.path.exists("../sdkconfig"):
		print ('sdkconfig not found. Defaulting build type!')
		print ('Build Type: Development')
		copy_and_replace('../sdkconfig.defaults-dev')
	else:
		with open("../sdkconfig", 'r') as f:
			if 'CONFIG_BUILD_DEVELOPMENT=y' in f.read():
				print ('Build Type: Development')
				copy_and_replace('../sdkconfig.defaults-dev')
			else:
				print ('Build Type: Production')
				copy_and_replace('../sdkconfig.defaults-prod')
isDev()
```

Now create the files sdkconfig.defaults-dev, sdkconfig.defaults-prod, and if you haven't already, create sdkconfig.defaults.

In sdkconfig.defaults-dev:
```
if OPTION=0
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME= "path1"
```

In sdkconfig.defaults-prod:
```
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME= "path2"
```

SoucheSouche, 2024-08-09:
Hello @ioviedodev,
should I consider this issue resolved or is there anything else we can help you with?

ESP-Marius, 2024-08-15:
Closing this one then, @ioviedodev feel free to reopen if you still think the suggested solution do not meet your requirements.

closed 2024-08-15 by ESP-Marius

ioviedodev, 2024-08-22:
Hi, thanks a lot. Your help me so much, @DarmorGamz and @SoucheSouche. I resolved it.