espressif/esp-idf #14269: wifi_mesh: BSSID(MAC) of parent and root have 1 bit added in comparison to real one

nicoDHX, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

that i see the same mac from parent, as i see the mac on parent.

### What is the actual behavior?

macs are different by 1 bit.

### Steps to reproduce.

program with wifi-mesh example 2 devices and print parent macs.

### Debug Logs.

```plain
device 1 has mac: 54:32:04:07:34:50

log from dev2: 
(4916703) mesh_main: ******ROUTE_TABLE*****[0m
(4916703) mesh_main: ADDR: 54:32:04:07:32:30 , IP: 84.50.4.7 , PORT: 12338
(4916703) mesh_main: ADDR: 54:32:04:07:38:a4 , IP: 84.50.4.7 , PORT: 42040
(4916703) mesh_main: ******END ROUTE_TABLE*****
*********RSSI: -80
(4916713) mesh_main: Parent BSSID: 54:32:04:07:34:51
(4916723) mesh_main: ROOT BSSID: 54:32:04:07:34:51 , IP: 84.50.4.7 , PORT: 20788
(4917093) mesh_main: [#RX:3775/3736][L:2] parent:54:32:04:07:34:51, receive from 54:32:04:07:34:50, size:1460, heap:253924, flag:0[err:0x0, proto:0, tos:0]

log from device 3:
(18464) mesh_main: ******ROUTE_TABLE*****
(18464) mesh_main: ADDR: 54:32:04:07:38:a4 , IP: 84.50.4.7 , PORT: 42040
(18464) mesh_main: ******END ROUTE_TABLE*****
*********RSSI: -30
(18464) mesh_main: Parent BSSID: 54:32:04:07:32:31 , IP: 84.50.4.7 , PORT: 12594
(18474) mesh_main: ROOT BSSID: 54:32:04:07:34:51 , IP: 84.50.4.7 , PORT: 20788
```

zhangyanjiaoesp, 2024-07-30:
@nicoDHX 
For an  ESP device, the mac of the STA interface is `base_mac`, and the mac of the softAP interface is `base_mac+1`. Please refer to this [doc] (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#mac-address).

In the mesh route table, it saves the STA's mac address, and the parent's mac is the softAP's mac address.

nicoDHX, 2024-07-30:
perfect, didn't found related information when i searched. Thank you.

closed 2024-07-30 by nicoDHX