espressif/esp-idf #14270: Wifi mesh, obtain rssi of neighbors and parent

nicoDHX, 2024-07-29:
### Is your feature request related to a problem?

Problem: Need a way to monitor mesh rssi parameters so i can send them to an app and process them to do things like monitoring network rssi during the day, select best spot for devices, etc.

### Describe the solution you'd like.

i would like a function that would give me a list of all devices in the mesh network that are in sight and the rssi with them.

nopnop2002, 2025-04-09:
Here is an example.
Leaf nodes only know the RSSI of their parent node.
https://github.com/espressif/esp-mesh-lite/tree/master/examples/mesh_local_control

zhangyanjiaoesp, 2025-05-08:
@nicoDHX 
do you still  have the problem?

closed 2025-06-10 by Alvin1Zhang