espressif/esp-idf #14271: ESP32-C6 not reporting L-LTF data in WiFi CSI callback and wrong subcarrier order for HT-LTF

benjamin-kl, 2024-07-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Scenario: receiving a csi_rx_cb callback for a HT20 PPDU.
Expected behavior:
Receive 256 bytes in the wifi_csi_info_t buffer, which is 64 (subcarriers) * 2 bytes (Im/Re) L-LTF CSI data + 64 (subcarriers) * 2 bytes (Im/Re) HT-LTF CSI data. The order of the subcarriers is 0~31,-32~-1 in both cases, as described in the WiFi API guide (https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-guides/wifi.html#wi-fi-channel-state-information)

### What is the actual behavior?

Actual behavior (esp32s3 DevKitC-1U):
same as expected behavior

Actual behavior (esp32):
same as expected behavior

Actual behavior (esp32c6 DevKitC-1): 
Receive 128 bytes in the callback. L-LTF data is missing. The order of the subcarriers is different from what is described in the WiFi API guide (null subcarriers/guard band in front?)

### Steps to reproduce.

Minimal reproducible example (receive):
```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include <sstream>

void _wifi_csi_cb(void *ctx, wifi_csi_info_t *csi_i) {
    std::stringstream ss;
    ss << "num values: " << csi_i->len << " [ ";
    for (int i = 0; i < csi_i->len; i++) {
        ss << static_cast<int>(csi_i->buf[i]) << " ";
    }
    ss << "]\n";
    printf(ss.str().c_str());
    fflush(stdout);
}

extern "C" void app_main(void) {
    ESP_ERROR_CHECK(nvs_flash_init());
    wifi_init_config_t wifi_cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&wifi_cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20));
    ESP_ERROR_CHECK(esp_wifi_start());
    const wifi_promiscuous_filter_t filt = {
            .filter_mask = WIFI_PROMIS_FILTER_MASK_DATA
    };
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&filt));
    ESP_ERROR_CHECK(esp_wifi_set_channel(13, WIFI_SECOND_CHAN_NONE));
    ESP_ERROR_CHECK(esp_wifi_set_csi(true));
    wifi_csi_config_t configuration_csi;
#ifdef CONFIG_SOC_WIFI_HE_SUPPORT
    configuration_csi.enable = 1;
    configuration_csi.acquire_csi_legacy = 1; //should give L-LTF CSI, but doesn't
    configuration_csi.acquire_csi_ht20 = 1; //should give HT-LTF CSI, but gives wrong subcarrier order
    configuration_csi.acquire_csi_ht40 = 0;
    configuration_csi.acquire_csi_su = 0;
    configuration_csi.acquire_csi_mu = 0;
    configuration_csi.acquire_csi_dcm = 0;
    configuration_csi.acquire_csi_beamformed = 0;
    configuration_csi.acquire_csi_he_stbc = 0;
    configuration_csi.val_scale_cfg = 0;
#else
    configuration_csi.lltf_en = 1; //gives L-LTF CSI
    configuration_csi.htltf_en = 1; //gives HT-LTF CSI
    configuration_csi.stbc_htltf2_en = 0;
    configuration_csi.ltf_merge_en = 0;
    configuration_csi.channel_filter_en = 0;
    configuration_csi.manu_scale = 0;
    configuration_csi.shift = 0;
#endif
    ESP_ERROR_CHECK(esp_wifi_set_csi_config(&configuration_csi));
    ESP_ERROR_CHECK(esp_wifi_set_csi_rx_cb(&_wifi_csi_cb, NULL));

    while(true) {
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
}
```

### Debug Logs.

```plain
Example CSI from esp32c6 (with code above):
num values: 128 [ 0 0 0 0 0 0 0 0 -1 -24 0 -24 2 -24 4 -23 6 -25 7 -24 9 -24 10 -23 11 -23 12 -23 12 -21 13 -22 11 -22 11 -21 11 -21 10 -20 10 -20 10 -22 9 -21 9 -22 9 -22 9 -23 9 -23 8 -23 9 -25 9 -26 9 -25 9 -26 0 0 10 -25 9 -26 9 -25 9 -26 9 -25 105 29 59 -99 12 69 -35 10 -67 -83 79 94 -75 84 114 33 -3 -123 39 102 94 -2 -32 95 67 -105 109 99 -13 -87 25 -57 70 -16 52 -39 -25 118 83 127 -128 109 -78 54 83 -77 49 37 -93 22 -58 -116 ]

Example CSI from esp32s3 (with code above):
num values: 256 [ 0 0 -11 -6 -11 -9 -12 -8 -10 -8 -11 -7 -11 -6 -11 -6 -11 -5 -12 -5 -11 -5 -12 -4 -12 -4 -12 -4 -13 -3 -13 -3 -13 -3 -13 -3 -14 -3 -14 -4 -13 -4 -13 -5 -13 -5 -12 -6 -12 -6 -11 -7 -11 -7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -7 -10 -6 -10 -6 -11 -6 -11 -5 -11 -5 -12 -5 -12 -5 -12 -5 -11 -5 -11 -5 -11 -5 -11 -5 -11 -5 -10 -5 -10 -6 -10 -7 -10 -7 -10 -8 -10 -8 -9 -9 -9 -9 -9 -10 -8 -10 -8 -9 -8 -7 -9 0 0 -51 -26 -51 -26 -45 -19 -52 -24 -48 -17 -47 -18 -49 -16 -47 -16 -51 -13 -51 -13 -49 -9 -51 -6 -50 -5 -51 -3 -56 -6 -51 -1 -56 -7 -54 -3 -58 -10 -58 -11 -55 -12 -54 -15 -53 -19 -49 -18 -48 -20 -44 -21 -42 -29 -37 -29 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -41 -26 -38 -26 -36 -31 -35 -31 -37 -39 -33 -33 -32 -40 -33 -37 -33 -42 -28 -44 -31 -43 -26 -43 -27 -40 -28 -40 -28 -39 -32 -37 -30 -32 -33 -39 -33 -30 -34 -33 -37 -28 -40 -28 -40 -30 -43 -30 -46 -31 -46 -25 -45 -22 -46 -19 ]
```

QingzhaoYin, 2024-08-28:
@benjamin-kl Actually, the CSI of the ESP32C6 differs from that of the ESP32C3. In the ESP32C6, the HT-LTF overrides the L-LTF, with a subcarrier index range of (-32 to -1, 0 to 32) and an effective subcarrier index range of (-28 to -1, 1 to 28). The current ESP32C6 documentation contains inaccuracies, but we will update it promptly.

benjamin-kl, 2024-09-03:
@QingzhaoYin Thanks for taking a look at this. However I think this is not only an issue with the documentation, but there is a bug in the software.

If the subcarrier order is as you said it is, I would expect to see zero-values at the beginning and zero-values at the end of the CSI data. If you take a look at the CSI I provided (recorded with C6), the CSI values for the first four subcarriers are indeed zero. However, the last three are not. 

Even worse, the C6 seems to occasionally report completely wrong CSI data. Three examples:

```
128;[0, 0, 0, 0, 0, 0, 0, 0, -3, 1, 0, 0, 8, -4, 7, -6, 4, -10, 2, -15, 0, -20, -1, -25, -1, -29, 0, -32, -1, -34, -1, -35, -1, -35, -1, -33, -2, -30, -1, -23, 0, -17, -1, -11, -2, -5, -2, 3, 0, 14, 1, 23, -2, 31, -5, 40, -6, 47, -7, 52, -7, 57, -4, 58, 0, 0, -5, 47, -5, 42, -5, 35, -5, 42, -5, 35, -1, -1, -1, 0, -4, 0, -2, 2, -1, 2, -1, 2, -2, 2, -2, 4, -3, 2, -3, 3, 1, 0, 4, -1, 2, 1, 0, 2, 2, 2, 3, 3, 3, 3, 3, 3, 4, 4, 3, 5, 2, 6, 3, -1, 13, 7, 0, 0, 0, 0, 0, 0]

128;[0, 0, 0, 0, 0, 0, 0, 0, -1, 1, 2, 1, 10, 0, 6, -4, 2, -10, 0, -14, 3, -17, 3, -21, 3, -25, 4, -28, 6, -30, 8, -32, 8, -32, 6, -31, 3, -28, 1, -23, 2, -17, 3, -12, 1, -5, -1, 4, -4, 13, -7, 22, -11, 30, -13, 37, -15, 42, -17, 46, -20, 48, -22, 47, 0, 0, -22, 40, -19, 35, -17, 27, -19, 35, -17, 27, -25, -19, -24, -18, -25, -19, -24, -18, 91, -5, 83, -89, 36, -24, -17, -76, -71, -24, 59, 33, 114, -80, -83, -15, 113, 14, -50, -43, -44, -85, 2, 114, -41, 46, -54, 48, -46, -2, -15, -94, -53, 3, 60, 5, -127, -62, 46, -66, -103, -82, 66, 0]

128;[-119, 48, -65, 117, 118, 32, -27, -85, -56, 124, 41, 72, -90, -60, -8, -84, 113, 39, 45, -99, -11, -101, -95, 47, -122, 27, 90, -98, -31, -13, -73, -12, 69, -22, 63, 78, -93, 28, 81, 19, -33, 94, -12, -26, 36, -126, -23, 5, 59, -70, 88, 122, -91, 75, 20, 122, 42, 23, -45, 12, 63, 22, 59, -71, -44, 6, 83, 71, 63, 63, 66, 76, -106, 104, -106, 111, -53, -81, 69, 66, 57, -119, 102, 8, 0, 81, 87, -4, 69, 1, -86, -58, 99, -73, 57, 14, -73, 90, -94, 84, 82, -41, 91, 47, -19, -61, 16, -94, -42, -82, -94, -103, -50, 59, 47, -105, -66, -57, 19, 107, -22, -58, -66, -115, 0, -46, -81, -29]
```

All three samples were received _from the same transmitter_ during a very short timeframe.

The first sample is as we expect.
The second sample does not have zero-values at the end
The third sample is completely malformed

QingzhaoYin, 2024-09-04:
@benjamin-kl 
Actually, invalid CSI data may be read in wifi_csi_rx_cb The "rx_channel_estimate_info_vld" field in wifi_pkt_rx_ctrl_t indicates whether the channel information is valid. From the data you provided, I don’t see the information for wifi_pkt_rx_ctrl_t. In my opinion, you may need to use this bit as a filter.

omalleyjp, 2024-12-17:
> @benjamin-kl Actually, invalid CSI data may be read in wifi_csi_rx_cb . The "rx_channel_estimate_info_vld" field in wifi_pkt_rx_ctrl_t indicates whether the channel information is valid. If you do not use this bit as a filter, it's possible to get the data as you provided.

I could not find rx_channel_estimate_info_vld in any documentation. Could provide a reference?

QingzhaoYin, 2024-12-17:
@omalleyjp This field is not yet included in the current documentation. We are in the process of updating the CSI-related documentation. In the meantime, you can refer to the esp_wifi_rxctrl_t structure in esp_wifi_he_types.h, where this field is located.
[image]

zxfrq, 2025-03-31:
|                    ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:69:34: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'cwb'
   69 |             rx_ctrl->mcs, rx_ctrl->cwb, rx_ctrl->smoothing, rx_ctrl->not_sounding,
      |                                  ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:69:48: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'smoothing'
   69 |             rx_ctrl->mcs, rx_ctrl->cwb, rx_ctrl->smoothing, rx_ctrl->not_sounding,
      |                                                ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:69:68: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'not_sounding'
   69 |             rx_ctrl->mcs, rx_ctrl->cwb, rx_ctrl->smoothing, rx_ctrl->not_sounding,
      |                                                                    ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:70:20: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'aggregation'
   70 |             rx_ctrl->aggregation, rx_ctrl->stbc, rx_ctrl->fec_coding, rx_ctrl->sgi,
      |                    ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:70:42: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'stbc'
   70 |             rx_ctrl->aggregation, rx_ctrl->stbc, rx_ctrl->fec_coding, rx_ctrl->sgi,
      |                                          ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:70:57: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'fec_coding'
   70 |             rx_ctrl->aggregation, rx_ctrl->stbc, rx_ctrl->fec_coding, rx_ctrl->sgi,
      |                                                         ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:70:78: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'sgi'
   70 |             rx_ctrl->aggregation, rx_ctrl->stbc, rx_ctrl->fec_coding, rx_ctrl->sgi,
      |                                                                              ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:71:42: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'ampdu_cnt'
   71 |             rx_ctrl->noise_floor, rx_ctrl->ampdu_cnt, rx_ctrl->channel, rx_ctrl->secondary_channel,
      |                                          ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:71:80: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'secondary_channel'
   71 |             rx_ctrl->noise_floor, rx_ctrl->ampdu_cnt, rx_ctrl->channel, rx_ctrl->secondary_channel,
      |                                                                                ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:72:40: error: 'wifi_pkt_rx_ctrl_t' {aka 'const esp_wifi_rxctrl_t'} has no member named 'ant'
   72 |             rx_ctrl->timestamp, rx_ctrl->ant, rx_ctrl->sig_len, rx_ctrl->rx_state);
      |                                        ^~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c: In function 'wifi_csi_init':
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:90:10: error: 'wifi_csi_config_t' {aka 'wifi_csi_acquire_config_t'} has no member named 'lltf_en'
   90 |         .lltf_en           = true,
      |          ^~~~~~~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:91:10: error: 'wifi_csi_config_t' {aka 'wifi_csi_acquire_config_t'} has no member named 'htltf_en'
   91 |         .htltf_en          = true,
      |          ^~~~~~~~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:92:10: error: 'wifi_csi_config_t' {aka 'wifi_csi_acquire_config_t'} has no member named 'stbc_htltf2_en'
   92 |         .stbc_htltf2_en    = true,
      |          ^~~~~~~~~~~~~~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:93:10: error: 'wifi_csi_config_t' {aka 'wifi_csi_acquire_config_t'} has no member named 'ltf_merge_en'
   93 |         .ltf_merge_en      = true,
      |          ^~~~~~~~~~~~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:94:10: error: 'wifi_csi_config_t' {aka 'wifi_csi_acquire_config_t'} has no member named 'channel_filter_en'
   94 |         .channel_filter_en = true,
      |          ^~~~~~~~~~~~~~~~~
C:/Users/22565/Desktop/esp-csi-master/esp-csi-master/examples/get-started/csi_recv/main/app_main.c:95:10: error: 'wifi_csi_config_t' {aka 'wifi_csi_acquire_config_t'} has no member named 'manu_scale'
   95 |         .manu_scale        = false,
      |          ^~~~~~~~~~
[... cut ...]
[998/1009] Linking C static library esp-idf\cmock\libcmock.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\22565\Desktop\esp-csi-master\esp-csi-master\examples\get-started\csi_recv\build\log\idf_py_stderr_output_7596 and C:\Users\22565\Desktop\esp-csi-master\esp-csi-master\examples\get-started\csi_recv\build\log\idf_py_stdout_output_7596

When I use official code to flash ESP32 C6,the above error occurs and I don't know how to fix it, can you give some advice, thanks!

QingzhaoYin, 2025-03-31:
@zxfrq 
The code version you are using may be too old, and some struct variables are no longer used in ESP32C6. The official code has not been updated yet, but it will be soon.

In the meantime, you can resolve this issue as follows: 
On ESP32C6, the` CONFIG_SOC_WIFI_HE_SUPPORT` macro is enabled, 
```
/**
  * @brief Channel state information(CSI) configuration type
  *
  */
#if CONFIG_SOC_WIFI_HE_SUPPORT
typedef wifi_csi_acquire_config_t wifi_csi_config_t;
#else
typedef struct {
    bool lltf_en;           /**< enable to receive legacy long training field(lltf) data. Default enabled */
    bool htltf_en;          /**< enable to receive HT long training field(htltf) data. Default enabled */
    bool stbc_htltf2_en;    /**< enable to receive space time block code HT long training field(stbc-htltf2) data. Default enabled */
    bool ltf_merge_en;      /**< enable to generate htlft data by averaging lltf and ht_ltf data when receiving HT packet. Otherwise, use ht_ltf data directly. Default enabled */
    bool channel_filter_en; /**< enable to turn on channel filter to smooth adjacent sub-carrier. Disable it to keep independence of adjacent sub-carrier. Default enabled */
    bool manu_scale;        /**< manually scale the CSI data by left shifting or automatically scale the CSI data. If set true, please set the shift bits. false: automatically. true: manually. Default false */
    uint8_t shift;          /**< manually left shift bits of the scale of the CSI data. The range of the left shift bits is 0~15 */
    bool dump_ack_en;       /**< enable to dump 802.11 ACK frame, default disabled */
} wifi_csi_config_t;
#endif // !CONFIG_SOC_WIFI_HE_SUPPORT
```
so you should use wifi_csi_acquire_config_t to configure CSI.(MAC_VERSION_NUM of ESP32C6 is 3)
```
#if CONFIG_SOC_WIFI_MAC_VERSION_NUM == 3
typedef struct {
    uint32_t enable                 : 1;    /**< enable to acquire CSI */
    uint32_t acquire_csi_legacy     : 1;    /**< enable to acquire L-LTF */
    uint32_t acquire_csi_force_lltf : 1;    /**< enable to acquire L-LTF */
    uint32_t acquire_csi_ht20       : 1;    /**< enable to acquire HT-LTF when receiving an HT20 PPDU */
    uint32_t acquire_csi_ht40       : 1;    /**< enable to acquire HT-LTF when receiving an HT40 PPDU */
    uint32_t acquire_csi_vht        : 1;    /**< enable to acquire VHT-LTF when receiving an VHT20 PPDU */
    uint32_t acquire_csi_su         : 1;    /**< enable to acquire HE-LTF when receiving an HE20 SU PPDU */
    uint32_t acquire_csi_mu         : 1;    /**< enable to acquire HE-LTF when receiving an HE20 MU PPDU */
    uint32_t acquire_csi_dcm        : 1;    /**< enable to acquire HE-LTF when receiving an HE20 DCM applied PPDU */
    uint32_t acquire_csi_beamformed : 1;    /**< enable to acquire HE-LTF when receiving an HE20 Beamformed applied PPDU */
    uint32_t acquire_csi_he_stbc_mode: 2;   /**< when receiving an STBC applied HE PPDU,
                                                 0- acquire the complete HE-LTF1
                                                 1- acquire the complete HE-LTF2
                                                 2- sample evenly among the HE-LTF1 and HE-LTF2 */

    uint32_t val_scale_cfg           : 4;   /**< value 0-8 */
    uint32_t dump_ack_en             : 1;   /**< enable to dump 802.11 ACK frame, default disabled */
    uint32_t reserved                : 15;  /**< reserved */
} wifi_csi_acquire_config_t;

```
That is same to wifi_pkt_rx_ctrl_t
```
#if CONFIG_SOC_WIFI_HE_SUPPORT
typedef esp_wifi_rxctrl_t wifi_pkt_rx_ctrl_t;
#else
/** @brief Received packet radio metadata header, this is the common header at the beginning of all promiscuous mode RX callback buffers */
typedef struct {
    signed rssi: 8;               /**< Received Signal Strength Indicator(RSSI) of packet. unit: dBm */
    unsigned rate: 5;             /**< PHY rate encoding of the packet. Only valid for non HT(11bg) packet */
    unsigned : 1;                 /**< reserved */
    unsigned sig_mode: 2;         /**< Protocol of the received packet, 0: non HT(11bg) packet; 1: HT(11n) packet; 3: VHT(11ac) packet */
    unsigned : 16;                /**< reserved */
    unsigned mcs: 7;              /**< Modulation Coding Scheme. If is HT(11n) packet, shows the modulation, range from 0 to 76(MSC0 ~ MCS76) */
    unsigned cwb: 1;              /**< Channel Bandwidth of the packet. 0: 20MHz; 1: 40MHz */
    unsigned : 16;                /**< reserved */
    unsigned smoothing: 1;         /**< Set to 1 indicates that channel estimate smoothing is recommended.
                                       Set to 0 indicates that only per-carrierindependent (unsmoothed) channel estimate is recommended. */
    unsigned not_sounding: 1;      /**< Set to 0 indicates that PPDU is a sounding PPDU. Set to 1indicates that the PPDU is not a sounding PPDU.
                                       sounding PPDU is used for channel estimation by the request receiver */
    unsigned : 1;                 /**< reserved */
[... 13 lines cut ...]
    unsigned timestamp: 32;       /**< timestamp. The local time when this packet is received. It is precise only if modem sleep or light sleep is not enabled. unit: microsecond */
    unsigned : 32;                /**< reserved */
#if CONFIG_IDF_TARGET_ESP32S2
    unsigned : 32;                /**< reserved */
#elif CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32C2
[... cut ...]
#elif CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32C2
    unsigned : 32;                /**< reserved */
    unsigned : 32;                /**< reserved */
    unsigned : 32;                /**< reserved */
#endif
    unsigned sig_len: 12;         /**< length of packet including Frame Check Sequence(FCS) */
    unsigned : 12;                /**< reserved */
    unsigned rx_state: 8;         /**< state of the packet. 0: no error; others: error numbers which are not public */
} wifi_pkt_rx_ctrl_t;
#endif
```