espressif/esp-idf #14272: arduino-as-component: I can't instantiate the class BluetoothSerial

fjmsouza, 2024-07-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Board**
ESP32

Device Description
ESP32 WROOM 32

**IDE Name**
VSCode

**Operating System**
Linux

**Description**
When compiling the code including the `BluetoothSerial SerialBT;` , the console reported undefined references.

**Sketch**

```
#include <Arduino.h>
#include <WiFi.h>
#include <BluetoothSerial.h>

const char *WIFI_SSID = "";      // Put here your Wi-Fi SSID
const char *WIFI_PASS = "";      // Put here your Wi-Fi password

BluetoothSerial SerialBT;

void connectionSetup()
{
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  Serial.println("Waiting connection to WiFi..");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
  }  
  Serial.println("Connected!");
}

void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  Serial.begin(115200);
  pinMode(2, OUTPUT);
  Serial.println(WiFi.macAddress());
  connectionSetup();

}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(2, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(2000);                      // wait for a second
  digitalWrite(2, LOW);   // turn the LED off by making the voltage LOW
  delay(2000);  
  Serial.println("ui!");                    // wait for a second
}
```

**Debug Message**

 *  Executando tarefa: ninja  

```
[... 40 lines cut ...]
-- Adding linker script /home/flaviojose/esp/v5.1.4/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- DEBUG: Use esp-modbus component folder: /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp-modbus.
ESP Insights Project commit: f2628c4-dirty
ESP RainMaker Project commit: f2628c4-dirty
-- Components: app_trace app_update arduino bootloader bootloader_support bt chmorgan__esp-libhelix-mp3 cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cbor espressif__esp-dsp espressif__esp-modbus espressif__esp-zboss-lib espressif__esp-zigbee-lib espressif__esp_diag_data_store espressif__esp_diagnostics espressif__esp_insights espressif__esp_modem espressif__esp_rainmaker espressif__esp_schedule espressif__esp_secure_cert_mgr espressif__jsmn espressif__json_generator espressif__json_parser espressif__libsodium espressif__mdns espressif__qrcode espressif__rmaker_common esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 joltwallet__littlefs json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/flaviojose/esp/v5.1.4/esp-idf/components/app_trace /home/flaviojose/esp/v5.1.4/esp-idf/components/app_update /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/components/arduino /home/flaviojose/esp/v5.1.4/esp-idf/components/bootloader /home/flaviojose/esp/v5.1.4/esp-idf/components/bootloader_support /home/flaviojose/esp/v5.1.4/esp-idf/components/bt /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/chmorgan__esp-libhelix-mp3 /home/flaviojose/esp/v5.1.4/esp-idf/components/cmock /home/flaviojose/esp/v5.1.4/esp-idf/components/console /home/flaviojose/esp/v5.1.4/esp-idf/components/cxx /home/flaviojose/esp/v5.1.4/esp-idf/components/driver /home/flaviojose/esp/v5.1.4/esp-idf/components/efuse /home/flaviojose/esp/v5.1.4/esp-idf/components/esp-tls /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_adc /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_app_format /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_coex /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_common /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_eth /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_event /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_gdbstub /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_hid /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_http_client /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_http_server /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_https_ota /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_https_server /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_hw_support /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_lcd /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_local_ctrl /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_mm /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_netif /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_netif_stack /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_partition /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_phy /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_pm /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_psram /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_ringbuf /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_rom /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_system /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_timer /home/flaviojose/esp/v5.1.4/esp-idf/components/esp_wifi /home/flaviojose/esp/v5.1.4/esp-idf/components/espcoredump /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__cbor /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp-dsp /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp-modbus /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp-zboss-lib /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp-zigbee-lib /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp_diag_data_store /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp_diagnostics /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp_insights /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp_modem /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp_rainmaker /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp_schedule /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__esp_secure_cert_mgr /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__jsmn /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__json_generator /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__json_parser /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__libsodium /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__mdns /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__qrcode /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/espressif__rmaker_common /home/flaviojose/esp/v5.1.4/esp-idf/components/esptool_py /home/flaviojose/esp/v5.1.4/esp-idf/components/fatfs /home/flaviojose/esp/v5.1.4/esp-idf/components/freertos /home/flaviojose/esp/v5.1.4/esp-idf/components/hal /home/flaviojose/esp/v5.1.4/esp-idf/components/heap /home/flaviojose/esp/v5.1.4/esp-idf/components/http_parser /home/flaviojose/esp/v5.1.4/esp-idf/components/idf_test /home/flaviojose/esp/v5.1.4/esp-idf/components/ieee802154 /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/managed_components/jo [... cut ...]
/home/flaviojose/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/flaviojose/DeskTop/GitHub/LIVE/WiFi_BT_IDF_ARD/WiFiDirect_BT_Client/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:745: undefined reference to `esp_spp_deinit'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  O processo de terminal "ninja" foi terminado com o código de saída: 1. 
 ```

VojtechBartoska, 2024-07-30:
Hello @fjmsouza, what version of ESP-IDF do you use? Looks like a mismatch of compatible versions between ESP-IDF and Arduino.

fjmsouza, 2024-07-30:
Hello @VojtechBartoska , thank you for your reply!
ESP-IDF v5.1.4 (edited in the top)

**Observation 1:** While installing the Arduino component, I encountered issues generating the `sdkconfig` file. After reviewing the output, I noted the requirement: `5.1 < ESP-IDF version <= 5.2`, which I followed. Now, generating the configuration works fine for simple examples like the blink program, and too, connecting to Wi-Fi functions correctly. However, when adding `BluetoothSerial.h` and invoking a `BluetoothSerial` class, I encountered related problems.

**Observation 2:** While waiting for tips, I exploring the `CMakeLists.txt` file. And, in the process, looking to `esp_spp_api.c`, where are `esp_spp_disconnect()` and `esp_spp_deinit()`, they are not being invoked, possibly because `BTC_SPP_INCLUDED` is set to `false`. This issue is illustrated in the attached figure.

[image]

fjmsouza, 2024-07-30:
[image]

Ops, I don't what is the ESP-IDF version... 1.8.0 or 5.1.4???

fjmsouza, 2024-07-30:
Now, compiling!

Enabling SPP (Serial Port Profile) and some handling in Partition Table, seems ok.
I will test and return with results.

[image]

VojtechBartoska, 2024-07-31:
version of ESP-IDF is 5.1.4, it is compatible with version of Arduino ESP32 core 3.0.X.

This documentation should help you: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

fjmsouza, 2024-07-31:
Thank you, @VojtechBartoska, I'll have a look.
For now, struggling a lot with the SDK Configuration Editor(BT, partition table, decreasing IRAM usage...), leaving it similar to the previous Arduino sdkconfig, a .ino serving as a reference to me.

But as sad, compiling with this fine tunning

closed 2024-12-18 by Alvin1Zhang