espressif/esp-idf #14273: Incorrect return value from math.h tgammaf(-inf)

projectgus, 2024-07-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1832-g23e4823f17, v5.0.4 and v5.2.2

### Espressif SoC revision.

ESP32 and ESP32-C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC, others

### Power Supply used.

USB

### What is the expected behavior?

Expect the result of `tgammaf(-inf)` to be NAN. (See https://pubs.opengroup.org/onlinepubs/9799919799/functions/tgammal.html and the C standard.)

This is a very similar issue to https://github.com/espressif/esp-idf/issues/7282 but applies to the single precision `tgammaf()` function.

### What is the actual behavior?

Result of `tgammaf(-inf)` is `inf`

### Steps to reproduce.

1. Add these two lines to a program such as the hello_world example:

```c
printf("tgamma(-inf) = %f\n", tgamma(-INFINITY));
printf("tgammaf(-inf) = %f\n", tgammaf(-INFINITY));
```

Note the first line is the double-precision function that was fixed in #7282, second line is single-precision function.

2. Flash and execute.

### Debug Logs.

```plain
tgamma(-inf) = nan
tgammaf(-inf) = inf
```
```

### More Information.

Thanks for your time and attention. :grin:

igrr, 2024-07-30:
Hi @projectgus!
This has been reported in upstream last year (https://sourceware.org/pipermail/newlib/2023/020171.html), but no fix there so far. We'll see if we can come up with a fix for this.

gerekon, 2024-07-31:
Hi @projectgus!
We will try to fix this in the next toolchain release.

projectgus, 2024-08-01:
Thanks Ivan and Alexey. It's not particularly urgent from MicroPython perspective as we have a workaround, but I figured good to pass it upstream (so you can decide if you want to pass it to your upstream!)

closed 2024-09-14 by espressif-bot