espressif/esp-idf #14274: The TCP transport layer is unstable

kanjiviroja1991, 2024-07-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

If there is consistent internet service, the websocket connection ought to be reliable and never break. We shouldn't be observe websocket disconnect issue because of tcp driver

### What is the actual behavior?

Websocket would regularly disconnect because of a transpot layer issue, even though we were absolutely sure there was no internet connection issue.

Please refer transport layer component from ESP IDF 5.2.1 

### Steps to reproduce.

1. Step
2. Step
3. Step
...
1. Connect Internet service (Ethernet, WiFi or GSM)
2. Connect Websocket  and verify connection
3. Put it in idle state more than 24 hours and capture serial logs 
4. We got this error from below list of API:
29-07-2024 22:08:36,[WS] Error read data
    **29-07-2024 22:08:36,[WS] Error receive data
    29-07-2024 22:20:00,[WS] ************** ERR_TCP_TRANSPORT_CONNECTION_FAILED ****************

==> rlen = esp_transport_read(client->transport, client->rx_buffer, client->buffer_size, client->config->network_timeout_ms);
This API frequently thrown ERR_TCP_TRANSPORT_CONNECTION_FAILED error due to we lost our websocket stability on field.

### Debug Logs.

```plain
29-07-2024 14:13:39,[WS] ************** ERR_TCP_TRANSPORT_CONNECTION_FAILED **************
29-07-2024 14:13:39,[WS] Error read data
29-07-2024 14:13:39,[WS] Error receive data
```

### More Information.

[esp_websocket_client.zip](https://github.com/user-attachments/files/16422250/esp_websocket_client.zip)
[tcp_transport.zip](https://github.com/user-attachments/files/16422255/tcp_transport.zip)

euripedesrocha, 2024-10-03:
@kanjiviroja1991 thanks for reporting, could you please provide more details on your setup? If possible to add a reproducer as well. 
esp_websocket_client version information will be useful also. 

From the logs shared, it looks like that there was some instability in the network and the client failed to receive, hence the disconnection.

kanjiviroja1991, 2024-10-03:
@euripedesrocha , We checked both wifi and ethernet, and we are also keeping an eye on the connection status on our laptop to see if the network is unstable, so there isn't a problem with the internet. However, we found no problem in network connection but ESP32 somehow detect it network issue.

closed 2025-04-07 by euripedesrocha