espressif/esp-idf #14275: How to import a static library as an IDF component

TxRxTxRx, 2024-07-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

see more information

### What is the actual behavior?

see more information

### Steps to reproduce.

see more information

### More Information.

I have read the relevant documentation, including the examples in "esp-idf\examples\build_system\cmake", but there are no examples of importing a static library as an IDF component. I referred to the "import_lib" example to try to add an IDF component, and below is my build project.

The directory of my project is as follows

```cmake
my_project/
├── components/
│   ├── mylibA/
│   │   ├── CMakeLists.txt
│   │   ├── include/
│   │   │   └── myliba.h
│   │   └── lib/
│   │       └── libmyliba.a
│   └── mylibB/
│       ├── CMakeLists.txt
│       ├── include/
│       │   └── mylibb.h
│       └── lib/
│           └── libmylibb.a
├── CMakeLists.txt
└── ...
```

In components/mylibA/CMakeLists.txt, the code is as follows:

```cmake
idf_component_register()
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/lib/libmyliba.a")
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include")
set(private_requires    mylibB)
add_prebuilt_library(lib_${COMPONENT_NAME} ${lib_path}
                     PRIV_REQUIRES ${private_requires}
                    )
set_target_properties(lib_${COMPONENT_NAME} PROPERTIES INTERFACE_INCLUDE_DIRECTORIES ${include_path})

target_link_libraries(${COMPONENT_LIB} INTERFACE lib_${COMPONENT_NAME})
set_target_properties(${COMPONENT_LIB} PROPERTIES IMPORTED_LINK_OPTIONS "-Wl,--whole-archive ${lib_path} -Wl,--no-whole-archive")

```

In components/mylibB/CMakeLists.txt, the code is as follows:

```cmake
idf_component_register()
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/lib/libmylibb.a")
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include")

add_prebuilt_library(lib_${COMPONENT_NAME} ${lib_path})
set_target_properties(lib_${COMPONENT_NAME} PROPERTIES INTERFACE_INCLUDE_DIRECTORIES ${include_path})

target_link_libraries(${COMPONENT_LIB} INTERFACE lib_${COMPONENT_NAME})
set_target_properties(${COMPONENT_LIB} PROPERTIES IMPORTED_LINK_OPTIONS "-Wl,--whole-archive ${lib_path} -Wl,--no-whole-archive")

```

Is this way of writing CMake correct? Currently, testing this method shows that the unused symbols in the two static libraries do not appear in the project's map file, as if the WHOLE_ARCHIVE parameter has not taken effect.

SoucheSouche, 2024-07-30:
Hi @TxRxTxRx,
We will take a look at it shortly.

TxRxTxRx, 2024-07-31:
> Hi @TxRxTxRx, We will take a look at it shortly.

Thank you very much for your reply. I will supplement the relevant information.

In my project,  "myliba" depends on "mylibb".
If there is only the mylibb component, the code in components/mylibB/CMakeLists.txt is modified as follows:

```cmake
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/lib/libmylibb.a")
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include")
idf_component_register()

add_prebuilt_library(lib_${COMPONENT_NAME} ${lib_path})

target_include_directories(lib_${COMPONENT_NAME} INTERFACE ${include_path})
target_include_directories(${COMPONENT_LIB} INTERFACE ${include_path})

target_link_libraries(${COMPONENT_LIB} 
    INTERFACE 
    -Wl,--whole-archive lib_${COMPONENT_NAME} -Wl,--no-whole-archive)

```

The WHOLE_ARCHIVE attribute is effective, and the weak symbols in the mylibb component can be overridden by strong symbols.

However, when I write the CMake for myliba in the same way, WHOLE_ARCHIVE does not work, and the weak symbols in both myliba and mylibb are not correctly overridden. The code in components/mylibA/CMakeLists.txt is as follows:

```cmake
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/lib/libmyliba.a")
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include")
set(private_requires    mylibB)

idf_component_register()

add_prebuilt_library(lib_${COMPONENT_NAME} ${lib_path}
                     PRIV_REQUIRES ${private_requires}
                    )

target_include_directories(lib_${COMPONENT_NAME} INTERFACE ${include_path})
target_include_directories(${COMPONENT_LIB} INTERFACE ${include_path})

target_link_libraries(${COMPONENT_LIB} 
    INTERFACE 
    -Wl,--whole-archive lib_${COMPONENT_NAME} -Wl,--no-whole-archive)

```

I would like to know the correct way to do this. After many tests, I still can't correctly add WHOLE_ARCHIVE for multiple static libraries with dependencies.

SoucheSouche, 2024-08-05:
Hi @TxRxTxRx,

Had your project not included this dependency between libA and libB, I would have suggested to refer to https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_prebuilt as the basic example on how to import prebuilt static libraries in the ESP-IDF build system.

Now, since you have a dependency between your static libs, it is probably not as straightforward as if those library where independent.
Could you maybe provide the CMakeLists of the main and base directory of your project if you are able to do so?

TxRxTxRx, 2024-08-05:
@SoucheSouche 

Of course, here is the directory structure of my project:
```c
my_project/
├── components/
│ ├── demo/
│ │ ├── CMakeLists.txt
│ │ ├── include/
│ │ ├── dummy.c                       // This is a dummy C file
│ │ └── libdemo.a
│ └── platform/
│   ├── CMakeLists.txt
│   ├── include/
│   ├── dummy.c                       // This is a dummy C file
│   └── libplatform.a
├── main
│ ├── main.c
│ └── CMakeLists.txt
└── CMakeLists.txt
```
The code in my_project/components/demo/CMakeLists.txt is:
```cmake
# Set the static library file path
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/libdemo.a" )

# Set the header file path
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include" )

# Set public dependencies
set(public_requires     platform)

# Set private dependencies 
set(private_requires    log )

idf_component_register(SRCS dummy.c)

add_prebuilt_library(lib_${COMPONENT_NAME} ${lib_path}
                     REQUIRES ${public_requires}
                     PRIV_REQUIRES ${private_requires}
                    )

target_include_directories(lib_${COMPONENT_NAME} INTERFACE ${include_path})

# target_link_options(lib_${COMPONENT_NAME} INTERFACE -Wl,--whole-archive ${lib_path} -Wl,--no-whole-archive)

# Link the imported library to the current component.
target_link_libraries(${COMPONENT_LIB} INTERFACE -Wl,--whole-archive lib_${COMPONENT_NAME} -Wl,--no-whole-archive )
```

The code in my_project/components/platform/CMakeLists.txt is:
```cmake
# Set the static library file path
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/libplatform.a" )

# Set the header file path
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include" 
                        "${IDF_PATH}/components/driver/gptimer/"
                        "${IDF_PATH}/components/esp_netif/include"
                        "${IDF_PATH}/components/esp_event/include"
                        "${IDF_PATH}/examples/system/console/advanced/components/cmd_nvs"
                        "${IDF_PATH}/examples/system/console/advanced/components/cmd_system"
                        "${IDF_PATH}/examples/system/console/advanced/components/cmd_wifi" )

# Set public dependencies
set(public_requires    console driver esp_adc nvs_flash unity vfs fatfs cmd_nvs cmd_system )

# Set private dependencies 
set(private_requires    app_update espressif__esp_encrypted_img espressif__esp-dsp )

idf_component_register(SRCS dummy.c)

add_prebuilt_library(lib_${COMPONENT_NAME} ${lib_path}
                     REQUIRES ${public_requires}
                     PRIV_REQUIRES ${private_requires}
                    )

target_include_directories(lib_${COMPONENT_NAME} INTERFACE ${include_path})

# target_link_options(lib_${COMPONENT_NAME} INTERFACE -Wl,--whole-archive ${lib_path} -Wl,--no-whole-archive)

# Link the imported library to the current component.
target_link_libraries(${COMPONENT_LIB} INTERFACE -Wl,--whole-archive lib_${COMPONENT_NAME} -Wl,--no-whole-archive )

```

The code in my_project/main/CMakeLists.txt is:
```cmake
idf_component_register(SRCS "main.c"
                    INCLUDE_DIRS "."
                    )
```

The code in my_project/CMakeLists.txt is:
```cmake
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(main)
```

I look forward to your reply.

SoucheSouche, 2024-08-05:
@TxRxTxRx, thanks a lot for sharing. I will take a look at it now.

TxRxTxRx, 2024-08-05:
@fhrbata Thank you for your suggestion

Using this method, it seems that the main component cannot find the header files of the mylibA and mylibB components. If we add
```cmake
set_target_properties(${COMPONENT_NAME} PROPERTIES INTERFACE_INCLUDE_DIRECTORIES ${include_path})
```
the REQUIRES and PRIV_REQUIRES in add_prebuilt_library will become invalid, and the compiler will show undefined reference errors.

fhrbata, 2024-08-06:
Hello @TxRxTxRx ,

> Using this method, it seems that the main component cannot find the header files of the mylibA and mylibB components.

Yeah, the `set_target_properties` seems to just overwrite everything in `INTERFACE_INCLUDE_DIRECTORIES`, including what was added e.g. by `target_link_libraries`, so the info about  directory includes is lost. This can be seen by printing the `INTERFACE_LINK_LIBRARIES` property before and after calling `set_target_properties`.

```bash
BEFORE set_target_properties: __idf_cxx;__idf_newlib;__idf_freertos;__idf_esp_hw_support;__idf_heap;__idf_log;__idf_soc;__idf_hal;__idf_esp_rom;__idf_esp_common;__idf_esp_system;__idf_xtensa;__idf_cxx;__idf_newlib;__idf_freertos;__idf_esp_hw_support;__idf_heap;__idf_log;__idf_soc;__idf_hal;__idf_esp_rom;__idf_esp_common;__idf_esp_system;__idf_xtensa;lib_mylibA
AFTER set_target_properties: -Wl,--whole-archive /home/fhrbata/work/esp-idf/examples/get-started/hello_world/components/mylibA/lib/libmyliba.a -Wl,--no-whole-archive
```

I thought that appending to `INTERFACE_LINK_LIBRARIES` should solve this.
```cmake
set_property(TARGET ${COMPONENT_LIB} APPEND PROPERTY INTERFACE_LINK_LIBRARIES "-Wl,--whole-archive ${lib_path} -Wl,--no-whole-archive")
```

but there is another problem with the
```cmake
target_link_libraries(${COMPONENT_LIB} INTERFACE lib_${COMPONENT_NAME})
```
because cmake will add the library into the link twice, first without `--whole-archive` as a result of the  `target_link_libraries`  and second time with it as result of setting ` INTERFACE_LINK_LIBRARIES` property , leading to  multiple definition of function within the archive. I haven't been able to find out how to tell cmake to link it just once with `--whole-archive` while using `target_link_libraries`.

Appending the prebuild lib directly into `INTERFACE_LINK_LIBRARIES` and adding its interface to the component seems to be working.  Here is what I ended up with

`components/mylibA/CMakeLists.txt`
```cmake
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include")
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/lib/libmyliba.a")
idf_component_register(
    INCLUDE_DIRS ${include_path} 
    SRCS dummy.c
    REQUIRES mylibB
)
set_property(TARGET ${COMPONENT_LIB} APPEND PROPERTY INTERFACE_LINK_LIBRARIES "-Wl,--whole-archive ${lib_path} -Wl,--no-whole-archive")
```

`components/mylibB/CMakeLists.txt`
```cmake
set(include_path        "${CMAKE_CURRENT_LIST_DIR}/include")
set(lib_path            "${CMAKE_CURRENT_LIST_DIR}/lib/libmylibb.a")
idf_component_register(
    INCLUDE_DIRS ${include_path} 
    SRCS dummy.c
)
set_property(TARGET ${COMPONENT_LIB} APPEND PROPERTY INTERFACE_LINK_LIBRARIES "-Wl,--whole-archive ${lib_path} -Wl,--no-whole-archive")
```
`main/CMakeLists.txt`
```cmake
idf_component_register(SRCS "hello_world_main.c"
    PRIV_REQUIRES spi_flash mylibA
                    WHOLE_ARCHIVE
                    INCLUDE_DIRS "") 
```

I'm not sure if this is what you are looking for. Maybe others will have other/better idea how this should be handled, because I cmake knowledge is pretty limited.

Thank you

TxRxTxRx, 2024-08-07:
@fhrbata Thank you for your suggestion. I tried your method and indeed found that the pre-built libraries on build.ninja all had the -Wl,--whole-archive attribute added. However, I still couldn't pass the compilation. The compiler gave an "undefined reference" error, indicating that it couldn't find the symbols of the IDF component. I have checked and confirmed that the components with missing symbols were added to REQUIRES.

SoucheSouche, 2024-08-07:
> @fhrbata Thank you for your suggestion. I tried your method and indeed found that the pre-built libraries on build.ninja all had the -Wl,--whole-archive attribute added. However, I still couldn't [...]

hi @TxRxTxRx, even after adding the PRIV_REQUIRES to the component registration in your static libs, you still get the undefined references to esp-idf components ?

TxRxTxRx, 2024-08-07:
@SoucheSouche 
Yes, I can find the components and their static library files that indicate undefined reference errors in build/esp-idf, and confirm adding them to REQUIRES or PRIV_REQUIRES in the component registration interface of the static library.

SoucheSouche, 2024-08-08:
Hi @TxRxTxRx,

I tried to create a dummy project locally with 2 static libraries ( one of them depending on the other ).
Running `idf.py build` in "lib_comp" will compile the dummy project to generate the `.a` files and place them in "main" directory together with the include files from both static libs.
Running `idf.py build` in "my_project" will build the main application.

`CmakeLists.txt` files are named according to their place in the structure so they are easy to identify later in the message.

The project is inspired from https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_prebuilt as I was mentioning in a previous message.

```
my_project/
├── lib_comp/
│ ├── main/             // this is a dummy main to compile the libraries
│ │ ├── CMakeListsDummy.txt
│ │ └── main.c 
│ ├── components/
| | ├── lib_a/
| | │ ├── CMakeListsLibA.txt
| | │ ├── lib_a.c 
| | | └── include/
| | └── lib_b/
| |   ├── CMakeListsLibB.txt
| |   ├── lib_b.c 
| |   └── include/
| └── CMakeListsLibComp.txt
├── main
│ ├── main.c
│ └── CMakeListsMain.txt
└── CMakeListsProject.txt
```

CMakeListsDummy:
```cmake
    idf_component_register(SRCS "main.c"
                        INCLUDE_DIRS "")
```

CMakeListsLibA:
```cmake
set(include ${CMAKE_CURRENT_LIST_DIR}/include)

    idf_component_register(SRCS lib_a.c
                        INCLUDE_DIRS ${include}
                        PRIV_REQUIRES app_update spi_flash log)

    # After build, copy the archive file and header file to parent example directory's main component
    add_custom_command(TARGET ${COMPONENT_LIB}
                    POST_BUILD
                    COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_FILE:${COMPONENT_LIB}> ${CMAKE_SOURCE_DIR}/../main
                    COMMAND ${CMAKE_COMMAND} -E copy ${include}/lib_a.h ${CMAKE_SOURCE_DIR}/../main
                    COMMENT "Copying built archive file and header to parent example directory...")
```

CMakeListsLibB:
```cmake
    set(include ${CMAKE_CURRENT_LIST_DIR}/include)

    idf_component_register(SRCS lib_b.c
                        INCLUDE_DIRS ${include}
                        PRIV_REQUIRES app_update spi_flash log lib_a)

    # After build, copy the archive file and header file to parent example directory's main component
    add_custom_command(TARGET ${COMPONENT_LIB}
                    POST_BUILD
                    COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_FILE:${COMPONENT_LIB}> ${CMAKE_SOURCE_DIR}/../main
                    COMMAND ${CMAKE_COMMAND} -E copy ${include}/lib_b.h ${CMAKE_SOURCE_DIR}/../main
                    COMMENT "Copying built archive file and header to parent example directory...")
```

CMakeListsLibComp:
```cmake
    cmake_minimum_required(VERSION 3.16)

    # Since we're only interested in the prebuilt libraries, trim
    # the build
    set(COMPONENTS lib_a lib_b main esptool_py)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)

    project(lib_comp C)
```

CMakeListsMain:
```cmake
    idf_component_register(SRCS "main.c"
                        INCLUDE_DIRS ".")

    # Import the libraries, specifying target names and the libraries paths.
    add_prebuilt_library(my_lib_a "liblib_a.a")
    add_prebuilt_library(my_lib_b "liblib_b.a")

    # `main` calls a function from the libraries, so link them to `main`
    target_link_libraries(${COMPONENT_LIB} PRIVATE my_lib_a)
    target_link_libraries(${COMPONENT_LIB} PRIVATE my_lib_b)
```

CMakeListsProject:
```cmake
    cmake_minimum_required(VERSION 3.16)

    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
    project(import_prebuilt)
```

SoucheSouche, 2024-08-21:
Hi @TxRxTxRx, I will close this issue now since I didn't get any update after my last suggestion. Feel free to reopen if you still can't solve your problem with the insights provided so far.

closed 2024-08-21 by SoucheSouche

diplfranzhoepfinger, 2025-10-01:
Confirm this https://github.com/espressif/esp-idf/issues/14275#issuecomment-2270964551 worked for me.
However a more modern Syntax like

`target_link_libraries(forte PUBLIC $<LINK_LIBRARY:WHOLE_ARCHIVE,forte-com-CAN>)`

Would be nice