espressif/esp-idf #14279: ESP-IDF v5.3中的警告：BLE_MESH: No outbound bearer found, inbound bearer 1

aboyuanna, 2024-07-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我在ESP-IDF v5.1.3中使用以下代码向BLE MESH中的所有节点发送数据，工作正常。但是同样的代码切换到v5.3时就出现了标题中的警告，而且节点没有收到数据。

esp_ble_mesh_msg_ctx_t ctx = {0};
    esp_err_t err;
    ctx.net_idx = 0x0000;
    ctx.app_idx = 0x0000;
    ctx.addr = 0xFFFF;
    ctx.send_ttl = DEFAULE_MSG_TTL;
    esp_ble_mesh_client_model_send_msg(vendor_client.model, &ctx, ESP_BLE_MESH_VND_MODEL_OP_CMD_NO_ACK, length, data, 0, false, ROLE_NODE);

forx157, 2024-07-31:
Hi, @aboyuanna 
Warning log: `BLE_MESH: No outbound bearer found, inbound bearer 1` indicates that a node is giving up relaying a local element that needs to receive and process a message. This warning only indicates that the relay behavior is affected and will not impact the normal message advertising. Therefore, the inability of the reported node to receive messages should be caused by other factors.

To better assist you in resolving the issue, could you kindly provide a brief overview of your application scenario, testing methods, network environment(such as the number of nodes, node distances, etc.)? Additionally, we would appreciate it if you could provide more logs, preferably adjusting the log level to info. You can adjust the log level in menuconfig following this path: `(Top) → Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL → BLE_MESH_STACK`.

Furthermore, if the receiving end displays a warning log `Incomplete timer expired`, you can try updating IDF first and then test if the issue persists.

aboyuanna, 2024-07-31:
我简化了应用场景，问题仍然存在。
网络中有两个节点，节点的单播地址为0x0002和0x0003，两个节点相距20厘米。
现在0x0002通过两种方式向0x0003发送数据。

当ctx.addr = 0x0003;时，发送成功。
### 以下是节点0x0002的log：

I (742993) BLE_MESH: send, app_idx 0x0000 src 0x0002 dst 0x0003
I (742999) BLE_MESH: send, len 57: c1a10a01020304050607080910111213141516171819203132333435363738394041424344454647484950515253545556575859603c3b24ec
I (743014) BLE_MESH: Seg, send_tag 0x01, send_szmic 0, aszmic 0
I (743021) BLE_MESH: Settings store, waiting 0 seconds
I (743032) BLE_MESH: Use security credentials 0x00, proxy 0
I (743040) BLE_MESH: Network PDU, count 2, interval 20
I (743047) BLE_MESH: Settings store, waiting 0 seconds
I (743083) BLE_MESH: Use security credentials 0x00, proxy 0
I (743085) BLE_MESH: Network PDU, count 2, interval 20
I (743090) BLE_MESH: Settings store, waiting 0 seconds
I (743101) BLE_MESH: Use security credentials 0x00, proxy 0
I (743102) BLE_MESH: Network PDU, count 2, interval 20
I (743108) BLE_MESH: Settings store, waiting 0 seconds
I (743119) BLE_MESH: Use security credentials 0x00, proxy 0
I (743120) BLE_MESH: Network PDU, count 2, interval 20
I (743126) BLE_MESH: Settings store, waiting 0 seconds
I (743138) BLE_MESH: Use security credentials 0x00, proxy 0
I (743138) BLE_MESH: Network PDU, count 2, interval 20
I (743144) BLE_MESH: Settings store, waiting 0 seconds
I (743155) BLE_MESH: Use security credentials 0x00, proxy 0
I (743156) BLE_MESH: Network PDU, count 2, interval 20
I (744136) BLE_MESH: Attempts: 3
I (744136) BLE_MESH: Resending 0/5, cred 0x00
I (744137) BLE_MESH: Settings store, waiting 0 seconds
I (744145) BLE_MESH: Resending 1/5, cred 0x00
I (744146) BLE_MESH: Settings store, waiting 0 seconds
I (744156) BLE_MESH: Resending 2/5, cred 0x00
I (744156) BLE_MESH: Settings store, waiting 0 seconds
I (744167) BLE_MESH: Resending 3/5, cred 0x00
I (744168) BLE_MESH: Settings store, waiting 0 seconds
I (744178) BLE_MESH: Resending 4/5, cred 0x00
I (744178) BLE_MESH: Settings store, waiting 0 seconds
I (744189) BLE_MESH: Resending 5/5, cred 0x00
I (744189) BLE_MESH: Settings store, waiting 0 seconds
I (744308) BLE_MESH: Settings store, waiting 0 seconds
I (744315) BLE_MESH: Seg 0/5 acked
I (744316) BLE_MESH: Seg 1/5 acked
I (744316) BLE_MESH: Seg 4/5 acked
I (744317) BLE_MESH: Attempts: 2
I (745111) BLE_MESH: Settings store, waiting 0 seconds
I (745118) BLE_MESH: Seg 2/5 acked
I (745118) BLE_MESH: Seg 3/5 acked
I (745118) BLE_MESH: Attempts: 1
I (745119) BLE_MESH: Resending 5/5, cred 0x00
I (745124) BLE_MESH: Settings store, waiting 0 seconds

### 以下是节点0x0003的log：
I (740297) BLE_MESH: Seg 1/5 received
I (740606) BLE_MESH: Seg 4/5 received
I (741325) BLE_MESH: Seg 0/5 received
I (741398) BLE_MESH: Settings store, waiting 0 seconds
I (741403) BLE_MESH: Use security credentials 0x00, proxy 0
I (741403) BLE_MESH: Network PDU, count 2, interval 20
I (741532) BLE_MESH: Seg 2/5 received
I (741633) BLE_MESH: Seg 3/5 received
I (742205) BLE_MESH: Settings store, waiting 0 seconds
I (742209) BLE_MESH: Use security credentials 0x00, proxy 0
I (742210) BLE_MESH: Network PDU, count 2, interval 20
I (742349) BLE_MESH: Seg 5/5 received
I (742349) BLE_MESH: Settings store, waiting 0 seconds
I (742354) BLE_MESH: Settings store, waiting 0 seconds
I (742359) BLE_MESH: Use security credentials 0x00, proxy 0
I (742360) BLE_MESH: Network PDU, count 2, interval 20
[... 56 lines cut ...]

### 以下是节点0x0003的log：
I (406571) BLE_MESH: Seg 1/5 received
I (406579) BLE_MESH: Seg 2/5 received
I (406675) BLE_MESH: Seg 3/5 received
I (406783) BLE_MESH: Seg 4/5 received
I (407671) BLE_MESH: Not sending ack for non-unicast address
I (408371) BLE_MESH: Not sending ack for non-unicast address
I (409071) BLE_MESH: Not sending ack for non-unicast address
I (409772) BLE_MESH: Not sending ack for non-unicast address
I (410472) BLE_MESH: Not sending ack for non-unicast address
I (411172) BLE_MESH: Not sending ack for non-unicast address
I (411873) BLE_MESH: Not sending ack for non-unicast address
I (412573) BLE_MESH: Not sending ack for non-unicast address
I (413273) BLE_MESH: Not sending ack for non-unicast address
I (413974) BLE_MESH: Not sending ack for non-unicast address
I (414674) BLE_MESH: Not sending ack for non-unicast address
I (415374) BLE_MESH: Not sending ack for non-unicast address
I (416075) BLE_MESH: Not sending ack for non-unicast address
I (416775) BLE_MESH: Not sending ack for non-unicast address
I (417475) BLE_MESH: Not sending ack for non-unicast address
I (418176) BLE_MESH: Not sending ack for non-unicast address
I (418876) BLE_MESH: Not sending ack for non-unicast address
I (419576) BLE_MESH: Not sending ack for non-unicast address
W (420276) BLE_MESH: Incomplete timer expired

forx157, 2024-08-05:
Hi, @aboyuanna 
Is the issue still present?

aboyuanna, 2024-08-05:
> Hi, @aboyuanna Is the issue still present?

Hi, I am currently busy with other things and don't have time to test.

closed 2024-08-20 by aboyuanna

UncleBigLu, 2024-10-15:
> indicates that the relay behavior is affected and will not impact the normal message advertising. Therefore, t

Hi. I found that if there's a vendor server and vendor client on the same ESP32, and the vendor client send message to `0xFFFF`, this warning will show up.
According to your answer this warning is related to the relay feature, and I found the code that emit this warning is in `esp-idf/components/bt/esp_ble_mesh/core/net.c`, line 1475. But in my case I **disabled relay** and set the message ttl to **0** to disable message forwarding, this warning still exists.
My mesh config is in the following

```
  76 static esp_ble_mesh_cfg_srv_t config_server = {
  77     /* 3 transmissions with 20ms interval */
  78     .net_transmit = ESP_BLE_MESH_TRANSMIT(2, 20),
  79     .relay = ESP_BLE_MESH_RELAY_DISABLED,
  80     .beacon = ESP_BLE_MESH_BEACON_ENABLED,
  81 #if defined(CONFIG_BLE_MESH_GATT_PROXY_SERVER)
  82     .gatt_proxy = ESP_BLE_MESH_GATT_PROXY_ENABLED,
  83 #else
  84     .gatt_proxy = ESP_BLE_MESH_GATT_PROXY_NOT_SUPPORTED,
  85 #endif
  86 #if defined(CONFIG_BLE_MESH_FRIEND)
  87     .friend_state = ESP_BLE_MESH_FRIEND_ENABLED,
  88 #else
  89     .friend_state = ESP_BLE_MESH_FRIEND_NOT_SUPPORTED,
  90 #endif
  91     .default_ttl = 0,
  92 };
```
According to the code in `esp-idf/components/bt/esp_ble_mesh/core/net.c` if the ttl == 0 or ttl == 1 function 'bt_mesh_net_relay` should return in line 1448 or line 1441. So is this a bug..?